Qucs/qucs #517: Compiling bug?

maikell, 2016-06-02:
struct appindex * idx = gperfapphash::get (key, strlen (key));
                             ^
equation.cpp:811:48: error: invalid use of incomplete type 'struct qucs::eqn::application::evalTypeFast()::appindex

Commit version: 9d924caa3879fbca978987a9fe964eadbc5f0fc4

in3otd, 2016-06-18:
we will need more details to understand where this comes from...
the `struct appindex

in3otd, 2016-06-18:
oops, pressed the wrong button, I deleted the previous incomplete comment...

we will need more details to understand where this comes from; the `struct appindex` is defined in `gperfapphash.cpp`, which is generated using `gperf`. So this sound a bit like #295, #492 or maybe even #512.

Which platform are you using? Which versions of `gperf` ?

in3otd, 2016-07-02:
likely due to #532, fixed in #534 .
Please open a new issue if compiling still does not work there.

closed 2016-07-02 by in3otd