Qucs/qucs #519: QUCS Installation Error

mohani99, 2016-06-04:
I am trying to install **qucs**  on **Ubuntu 16.04 LTS**
I tried to install it from the repository:
```
sudo apt-add-repository ppa:qucs/qucs
sudo apt-get update
sudo apt-get install qucs
```
But without any success. So I tried to download **qucs-0.0.19-rc1.tar.gz** from [here](https://sourceforge.net/projects/qucs/files/qucs/0.0.19-snapshots/). 

After installation, I ran this code:

```
tar xvzf qucs-0.0.19.tar.gz
cd qucs-0.0.19
./configure
```
But this error appeared on the terminal:
```
checking Checking if "admsXml" is working...  no  
configure: error: need admsXml, but not found. consider --with-mkadms=<alternative admsXml> 
configure: error: ./configure failed for qucs-core
```
Is this something that I am missing?

felix-salfelder, 2016-06-04:
both, qucs (the gui) and qucs-core (a simulator) still depend on admsXml.

admsXml is part of ADMS. just install it. cf. `README.md`.

mohani99, 2016-06-04:
I've found another solution. Just install **qucs-0.0.19-150523-git-0046130** from the same like that I provided before and follow exactly the steps in the `README` file. It'll do the trick. Cheers!

closed 2016-06-04 by mohani99