Qucs/qucs #522: Exported graph image has the markers cut

in3otd, 2016-06-12:
a little thing, that I find somewhat annoying:
select a graph with markers, then export it as image (`File` -> `Export as image` then check `Export selected only`): the exported image is sized to fit the graph only, but the marker lines are visible.

Either we should:
-  not show the markers at all (not nice), or
- size the exported image to include the graphs (good), or
- allow the user to choose (even better).

Example:
- exported image with cropped markers (current behaviour)
[image]

- exported image as it should be
[image]

ra3xdh, 2016-06-13:
I confirm this issue. There also exists similar issue with wire labels. I will add fix soon. It's strange that I never met it before. I often use export of diagrams and schematics to images for articles preparation.

ra3xdh, 2016-06-13:
@in3otd, See #523 for fix.

guitorri, 2016-11-02:
Fixed with #523.

closed 2016-11-02 by guitorri