Qucs/qucs #531: qucs compilation crash with following error: logging.h:37:8: error: ‘FILE’ does not name a type

borisov-r, 2016-06-24:
/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/components -I../src/interface -I.   -O2 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT nodelist.lo -MD -MP -MF .deps/nodelist.Tpo -c -o nodelist.lo nodelist.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/components -I../src/interface -I. -O2 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT nodelist.lo -MD -MP -MF .deps/nodelist.Tpo -c nodelist.cpp  -fPIC -DPIC -o .libs/nodelist.o
In file included from nodelist.cpp:32:0:
logging.h:37:8: error: ‘FILE’ does not name a type
 extern FILE * file_status;
        ^~~~
logging.h:38:8: error: ‘FILE’ does not name a type
 extern FILE * file_error;
        ^~~~
make[5]: *** [Makefile:859: nodelist.lo] Error 1
make[5]: Leaving directory '/home/radoslav/work/qucs/qucs-core/src'
make[4]: *** [Makefile:905: all-recursive] Error 1
make[4]: Leaving directory '/home/radoslav/work/qucs/qucs-core/src'
make[3]: *** [Makefile:597: all] Error 2
make[3]: Leaving directory '/home/radoslav/work/qucs/qucs-core/src'
make[2]: *** [Makefile:768: all-recursive] Error 1
make[2]: Leaving directory '/home/radoslav/work/qucs/qucs-core'
make[1]: *** [Makefile:655: all] Error 2
make[1]: Leaving directory '/home/radoslav/work/qucs/qucs-core'
make: *** [Makefile:380: all-recursive] Error 1

andresmmera, 2016-06-25:
Hello, 
This morning, I've installed a fresh copy of Fedora 24 and, for my surprise, I faced exactly the same error than you. The issue was solved by moving "#include <stdio.h>" from logging.c to logging.h

Hope it helps

borisov-r, 2016-06-25:
Hi, thanks for the replay.
I'm using arch linux with gcc version 6.1.1 20160602 (GCC).
Tried to move all #include directives from logging.c to logging.h, but got many different erros.
Finally I was able to compile qucs just by commenting line 32 in file "qucs-core/src/nodelist.cpp":
//#include "logging.h"
I think the issue is related to the multiple time definition of the logging.h header into nodelist.cpp.
Don't really know how but it worked for me.

in3otd, 2016-06-26:
I'm reopening as I think we should understand a bit better where this issue is coming from, since also @andresmmera was able to reproduce it.

in3otd, 2016-06-26:
installed Fedora 24 in a VM, indeed the error is there...

it can be fixed by adding `#include <stdio.h>` [here](https://github.com/Qucs/qucs/blob/master/qucs-core/src/logging.h#L32) . ( @andresmmera , you answer had the actual file to include cut, due to a "feature" of the GitHub markdown, so at first I didn't understand which `#include` should be moved and I have the impression neither the OP did - but thanks for the explanation via IRC :grin:)

Not sure why it did not fail before... Fedora 24 uses `gcc` 6.1.1 so maybe something has changed there (note that also `clang` 3.8.0 fails)

felix-salfelder, 2016-06-26:
i don't see, why the file pointers should be exported at all. file_status and file_error conceptually look very private in the scope of the logging implementation (logging.c).

iow, including stdio.h from logging.h is a bad idea, because it's an implementation detail.

in3otd, 2016-06-26:
> i don't see, why the file pointers should be exported at all. file_status and file_error conceptually look very private in the scope of the logging implementation (logging.c).

you mean we should have some nice method to set ` file_status` and `file_error` instead of the nice raw C style here?

> iow, including stdio.h from logging.h is a bad idea, because it's an implementation detail.

but in the current implementation shouldn't `stdio.h` be included there to tell what `FILE` is, instead of relying on having `stdio.h` included before `logging.h` by every file using it??

felix-salfelder, 2016-06-26:
> you mean we should have some nice method to set

a nice method with a name that *explains the purpose of doing so*. `redirect_error_to_stdout()` (wild guess!) does not even take an argument...

> shouldn't stdio.h be included there

no, there should be no use of the `FILE` pointer outside of logging.c, because it's internal and an implementation detail.

currenly nothing outside logging.c uses the pointers (as it seems), and cleaning up *now* seems the natural thing to do.

in3otd, 2016-07-08:
fixed in #536

closed 2016-07-08 by in3otd