Qucs/qucs #532: Need to check if gperf is installed.

in3otd, 2016-06-26:
While checking #531 I realized that there is no check whether  `gperf` is installed and working (didn't we discuss this already??). 
If it's not installed there is just an error message on the console, like
```
make[2]: Entering directory `/home/qucs/qucs-git/qucs-core/src'
I -m 8 gperfapphash.gph | sed -e 's/{""},/{"",0},/g' > gperfapphash.cpp
/bin/sh: I: command not found
```
which can go unnoticed as it quickly scrolls away due to the other compilation messages, then the compilation continues, until an error like the ones in #492 and #517 appears.
Worse yet, at this stage an empty `gperfapphash.cpp` file has been created, so if the user then installs `gperf` now the compilation will still fail, as `gperfapphash.cpp` is already present and will not be recreated.

(I did not check what happens using `cmake`)

felix-salfelder, 2016-06-26:
here, `qucs-core/configure.ac` contains

    dnl Check for additional programs in PATH
    AC_PATH_PROG([GPERF],[gperf],[gperf])
    AS_IF([test -z "$GPERF"],
         [AC_MSG_ERROR([gperf required in the PATH])])

so maybe the check is just broken?

in3otd, 2016-06-26:
oh, I didn't even check if a test for `gperf` was already there... apparently it got broken in https://github.com/Qucs/qucs/commit/3077b51b8d11373fa2e32cdc6849ce734af1507b, `git blame` says.
There is a similar (non-) issue with the `git` [check] (https://github.com/Qucs/qucs/blob/master/qucs-core/configure.ac#L919) also.

in3otd, 2016-07-02:
fixed by #534

closed 2016-07-02 by in3otd