Qucs/qucs #533: another gperfappgen weirdness

felix-salfelder, 2016-06-26:
trying to fix #531, i am not getting beyong this point (haven't built qucsator for some time).

    /bin/sh ../libtool  --tag=CXX   --mode=link clang++  -O2 -std=gnu++11 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__   -o gperfappgen gperfappgen.o  -lm 
    ./gperfappgen  > gperfapphash.gph #1
    /bin/sh: ./gperfappgen: No such file or directory

digging further, i am getting this

    qucs-core_builddir/src $ make gperfappgen
    /bin/sh ../libtool  --tag=CXX   --mode=link clang++  -O2 -std=gnu++11 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__   -o gperfappgen gperfappgen.o  -lm 
    qucs-core_builddir/src $ ls gperfappgen
    (nothing!)

really strange... or is it just me?

in3otd, 2016-06-26:
uhm, never seen this before, but it seems you are building out of tree... I don't think I ever tried that with the autotools in qucs :grimacing:

felix-salfelder, 2016-06-27:
i think you should, it's much less messy and more flexible. also, that's what travis is doing (successfully).

(i do not think, this issue is related to "out of tree", but who knows...)

in3otd, 2016-06-27:
ok, tried to build out of tree, `configure` told me I should `make distclean` in the source tree before, then it worked fine here, also just (re)making only `gperfappgen`...

felix-salfelder, 2016-06-29:
it seems the problem was a hosed llvm. maybe.
i wonder why configure did not catch this. and i cannot reproduce. `:|`

sorry for the noise...

closed 2016-06-29 by felix-salfelder

in3otd, 2016-07-03:
I'm just playing with a brand new Slackware 14.2... here now I get the same error ! :confused: 
This is with the "old" work directory. Cloning to a clean directory, everything works as expected.
Not the first time I see `libtool` doing strange things...

felix-salfelder, 2016-07-04:
> Not the first time I see `libtool` doing strange things...

not entirely sure, if libtool is to blame here. some broken (dist)clean target maybe?
anyway, changing the titie, for the record...

in3otd, 2016-07-04:
it turns out that `libtool` in `qucs-core`, which apparently should be a script automatically generated by `configure`, is largely incomplete (only the last part, between `# ### BEGIN LIBTOOL TAG CONFIG: CXX` and `# ### END LIBTOOL TAG CONFIG: CXX` is present).
So when `libtool` is called it actually does nothing, as seen. Why it was not autogenerated correctly I do not know. I guess some interactions between old autogenerated files (`qucs-core` was previously compiled by another system in the same directory) and a newer autotools version(s).

in3otd, 2016-07-04:
just for the record, what I did is reusing the `qucs` source code directory from another Linux system (i.e. the directory is mounted from another disk). There I did the usual `./configure; make` stuff and got the problem above. I did not explicitly do a `make distclean` before. I should have done, probably but I didn't think it was necessary.

I blamed `libtool` just because I already saw strange errors (andresmmera also reported similar issues) appearing when `libtool` is called, but it can well be that the `libtool` script is not (re)generated correctly, for some unclear reason