Qucs/qucs #538: "File -> Example" does nothing

pidgeon777, 2016-07-11:
I suppose that by clicking on:

File -> Examples

it would open a window to browse the examples provided within the software package. Unfortunately, nothing happens by doing that.

[image]

I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine.

in3otd, 2016-07-11:
did you have previously another (older) version of Qucs installed?
If it was not properly uninstalled, you might still have a spurious `QUCSDIR` environment variable pointing to the old installation directory; if you open a command window and type `echo %QUCSDIR%` do you see a path to the old Qucs ?

pidgeon777, 2016-07-12:
I never installed Qucs previously. I ran the command you suggested, and here is the result:

[image]

andresmmera, 2016-07-12:
I can reproduce this bug in Fedora 24. It seems the examples folder is missing:

> vfs-open:

 /usr/local/share/qucs/docs/examples/: error opening location: Error when getting information for file '/usr/local/share/qucs/docs/examples': No such file or directory

[andresmmera@localhost qucs]$ ls /usr/local/share/qucs/docs/
cs  de  en  es  fr  pt  ru  uk

Then, I went to qucs/qucs/examples folder and I did:
./autogen.sh
./configure
sudo make install

At this point, the examples folder was created at the aforementioned folder, but... the thing is that there's only a text file (README) which contains the following:

>  This distribution contains no examples, please copy them manually from the qucs git repository:
git clone git://git.code.sf.net/p/qucs/git qucs

felix-salfelder, 2016-07-12:
there are examples in `$toplevel/examples` plus a `Makefile.am` with my name in it. maybe i once tried?

there are compressed files involved. we should unpack them first. also, i can't find any license statements.

in3otd, 2016-07-12:
yep, this has nothing to do with `QUCSDIR`, at first I thought it was due to the path problems we saw recently...

The `examples` directory is "simply" missing from the binary distribution; apparently the examples were usually "manually" added by the [release script] (https://github.com/Qucs/qucs/blob/master/contrib/qucs-release.sh).

pidgeon777, 2016-07-12:
Are you sure? In the portable distribution I downloaded, there is a folder containing some alleged examples:

[image]

in3otd, 2016-07-12:
oh, so they ended up there... I think they should be in `C:/qucs-win64-bin/share/qucs/docs/examples/` instead.

felix-salfelder, 2016-07-12:
the examples collection is not part of qucs-doc. this makes sense, because none of them is documentation for qucs. perhaps it was my idea to use `${docdir}`, sorry about the confusion.

on second thought... the schematics should likely go to `${pkgdatadir}/examples`  which defaults to `${prefix}/share/qucs/examples`. any objections?

in3otd, 2016-07-24:
> there are examples in `$toplevel/examples` plus a `Makefile.am` with my name in it. maybe i once tried?

BTW, do you/someone know/remember why we have also `$toplevel/qucs/examples`?

felix-salfelder, 2016-07-24:
maybe someone sensed the need for splitting off qucs-gui into a seperate project?
then, all the *sch files should certainly be part of that qucs-gui project.

(i totally agree with this, move the files and i will merge it)

in3otd, 2016-07-24:
don't know, looking the the files history, both the example schematics in `examples/` and the (almost empty) directory `qucs/examples/` were added by @fransschreuder about 3 years ago , almost at the same time.

Did I understand correctly that your proposal is to put everything in `qucs/examples/`? Either directory is fine for me. Today the examples are of course just for the Qucs GUI + `qucsator`; if the Qucs GUI will ever support more simulators the examples may need to be moved again. (Where, to a simulator-specific directory in the GUI repo?)

felix-salfelder, 2016-07-24:
> to put everything in qucs/examples/?

yes (beacuse the examples are for qucs "the gui"). i also suggest to turn this into an external repo, because some examples are really big.

> the examples may need to be moved again.  ... simulator-specific

 if the example is an .sch file, it can be opened/viewed/edited with qucs-gui and thus is an example for qucs-gui. if it does not work, it should probably not be included (YMMV). if it's simulator specific it should probably be part of a different project, such as qucs-$simulator. there are example .sch files in  gnucap-qucs (or was it qucs-gnucap?).

guitorri, 2016-11-24:
Should be fixed with #610

closed 2017-01-20 by guitorri