Qucs/qucs #539: Bad alignment with certain components

pidgeon777, 2016-07-11:
In this example I'm placing a voltage-controlled switch, S4. You can see how sometimes it is almost impossible to connect the components with a wire, because in this case the connector of S4 is not aligned to the grid:

[image]

This problem is annoying, because it won't help in making the proper connections between the components.

I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine.

andresmmera, 2016-07-11:
Hello, 

You can align your component on the grid by doing left click ->Set on Grid (CTRL+U). It should work :-)

pidgeon777, 2016-07-11:
Thank you, I didn't know about that feature. I originally thought that when moving/placing/rotating components, they would automatically get aligned to the grid. Maybe this issue could be turned to a feature request (automatic grid alignment) ;).

At the moment the best solution seems to do CTRL+A followed by the alignment (CTRL+U).

in3otd, 2016-07-11:
yes, the problem is known since some time and there were some fixes proposed. These fixes are not yet included partly because it's slightly trickier than it seems (some "components", like the Paintings should not be automatically grid-aligned) and mainly because of a lack of manpower to review all the code changes proposals...

pidgeon777, 2016-07-12:
Thank you, hopefully that feature will be added in the future. For now I suppose it would be enough to just stick to the CTRL+A -> CTRL+U workaround.