Qucs/qucs #540: Strange behavior of the cursor when in "Wire placement" mode

pidgeon777, 2016-07-12:
This problem arises when I select the "Wire placement" mode (CTRL+E), zooming in and out. Let's suppose I start with the whole schematic displayed (press "0"):

[image]

then I decide to zoom in:

[image]

zoom again:

[image]

and again:

[image]

You see how the two lines (vertical and horizontal) which marks the cursor, doesn't cross anymore, after a certain amount of zoom. I tried this in the "Wire placement" mode (CTRL+E). When zooming out, the two lines will eventually cross again.

I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine, resolution is 1920x1080.

in3otd, 2016-07-12:
this appears to be the issues solved by #504... seems ready to be merged...

guitorri, 2016-11-24:
Should be fixed with #504

closed 2016-11-24 by guitorri