Qucs/qucs #541: "Move Up/Down" buttons greyed out in the equation property window.

pidgeon777, 2016-07-12:
I created a list of equations:

[image]

It is impossible to modify the order of the displayed equations, because the "Move Up/Down" buttons are greyed out in the equation property window:

[image]

I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine.

in3otd, 2016-07-12:
Thanks for finding this bug! If you switch the application language to English then it will work as expected ...

The problem comes from [here](https://github.com/Qucs/qucs/blob/master/qucs/qucs/components/componentdialog.cpp#L588), the comparison should be with `tr("equation")` (But I think a translator could mess up this anyway, proposing a different translation here and in the Equation component description, or am I wrong?).

pidgeon777, 2016-07-12:
Probably it's as you're suggesting, maybe the translator has some effect even on the description of the equation... after the translation, the test in the code won't work anymore ("equation" get translated to "equazione" when using Italian language).

guitorri, 2016-10-30:
Fixed with #548.

closed 2016-10-30 by guitorri