Qucs/qucs #542: component library can't load lib files on windows

aviadb, 2016-07-12:
when launching component library
an error message is displayed

`
Cannot open "/c/qucs-win64-bin/share/qucs/library/.lib
`

The component library then opened empty without any libraries loaded

pidgeon777, 2016-07-12:
I confirm the same problem:

[image]

[image]

[image]

I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine.

pidgeon777, 2016-07-15:
This is weird. I've set Qucs language to Italian, and the software rightly starts in Italian when opened directly in Windows (i.e. double click on "qucs.exe"). Anyway, if I open Qucs through the command window as you suggested, the program runs in English.

About your suggestion, trying with:

`set QUCSDIR=c:\qucs-win64-bin\bin`

leads to this, when launching component library:

[image]

It seems that the correct command would be:

`set QUCSDIR=c:\qucs-win64-bin`

leading to the proper result:

[image]

in3otd, 2016-07-16:
thanks; I didn't actually try and had the feeling it may not have been fully right.

Regarding the language issue you mentioned, on the command line it uses English regardless if you set the `QUCSDIR` variable or not? I think that if Qucs cannot find the language translation files it defaults to English.

We had quite some issues with the various components paths in the past, #501 mentions it also fixed some problems with the translations.

pidgeon777, 2016-07-24:
Hi in3otd, what do you exactly mean with checking if on the command line it uses English? This is what I did:

1) Start a cmd session.
2) cd C:\Users\XXX\Desktop\qucs-win64-bin\bin
3) qucs

Anyway, I just tried again, and this time Qucs started in Italian. Then I decided to set it in English, and I started it again through the cmd window. It ran in English. Then I set it in Italian one last time, and it run in Italian. So, it seems that currently the problem doesn't arise anymore.

in3otd, 2016-07-24:
I meant to say/write "starting it from the command line does it use English regardless if you set the `QUCSDIR` variable or not?", so you did it correctly.

 I'll try to see if I can reproduce the issue here...

felix-salfelder, 2020-09-26:
> So, it seems that currently the problem doesn't arise anymore.

closing. please reopen as needed.

closed 2020-09-26 by felix-salfelder