Qucs/qucs #545: Can you "make -j" qucs-core ?

in3otd, 2016-07-15:
Recently a couple of users reported that they were not able to compile Qucs using "parallel make" (`make -j`), which also appears to be the default on some distributions (Gentoo, maybe?).

I also saw this issue sometimes; the quick "fix" I have done is to put `.NOTPARALLEL:` in `qucs-core/src/components/verilog/Makefile.am`.

I was wondering if others see this issue...

guitorri, 2016-10-30:
Just tried `make -j8`. I fails also on OSX. I think it is caused because `admsXml` has not yet dropped the headers when another parallel execution tries to flatten the module.

See a portion of the make log below. It is out of order but you get the idea.
(weird that the admsXml warning message is getting mangled)
```
admsXml potentiometer.va  -e ./analogfunction.xml -e ./qucsVersion.xml -e ./qucsMODULEcore.xml -e ./qucsMODULEdefs.xml -o potentiometer
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
[warning] [./nigbt.va:14]: standard [warning] va[m./MESFET.va:13s] :f[warning]  isltea ncdraeradt ev[da ./EKV26MOS.va:18m(]ns:o  tsf tifaloneud nacdrr dei anvt ae-mdIs  (p nafotithl )ef  o.cu.rn.ed a t'iendisciplines.vamsd ' -
I( npoatt hf)o u.n.d.  i'ndisciplines.vams '-
I path) [warning] ..[../log_amp.va:17 ]':disciplines.vams 's
tandard vams file created (not found in -I path) ... 'disciplines.vams'
[warning] [./mod_amp.va:12]: standard vams file created (not found in -I path) ... 'constants.vams'
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
[fatal..] [source:error] there is no nature with access I, missing discipline.h file?
make[5]: *** [mod_amp.cpp] Error 1
make[5]: *** Waiting for unfinished jobs....
[info...] admsXml-2.3.4 (a508cf6) Aug  7 2016 12:01:24
```

As Felix pointed, installing the headers might solve 2 issues at once: https://github.com/Qucs/ADMS/issues/49

in3otd, 2016-10-30:
thanks; do you think we should in the meantime add the `.NOTPARALLEL:` in `qucs-core/src/components/verilog/Makefile.am`, until ADMS is fixed? BTW, does that "fix" the issue also on your side? We will then have one potential build issue less and it should not harm the non-parallel make.

guitorri, 2016-10-30:
Yes, `.NOTPARRALEL:` works. I think we should add this to avoid build issues. I will send a PR in a minute. We keep this issue open till we fix the Verilog-A header inclusion.

Back to source of the problem. `admsXml` is called just once for a each `.va` file.  Contrary to what I said above, there are no multiple passes of `admsXml`. It looks for the headers in the current directory and creates them if not found:

```
admsXml nigbt.va  -e ./analogfunction.xml -e ./qucsVersion.xml -e ./qucsMODULEcore.xml -e ./qucsMODULEdefs.xml -o nigbt
``` 

I don't know for sure, but my guess is that parallel executions of `admsXml` are simultaneously trying to write the header file in the `components\verilog\` directory. When comes the time  load the header, it is either still open or corrupted by the parallel execution. 

In any case, the sane solution is to have the headers static in on place and include them from there.

felix-salfelder, 2016-10-30:
in the long run, there should be no admsXml invocation during qucsator build. so any workaround is appropriate here.

i hope we will be able to amend this with 0.0.20...

background: models are simulator *input*, not *part of* a simulator.

guitorri, 2017-01-29:
Short note. Sometimes running `make -j8` it gives:
```
scan_spice.lpp:51:10: fatal error: 'parse_spice.hpp' file not found
#include "parse_spice.hpp"
         ^
updating parse_spice.hpp
```
Doing `make` again lets it continue.

guitorri, 2017-07-12:
Since ADMS 2.3.6 the `.vams` headers are installed.
See : https://github.com/Qucs/ADMS/commit/e220505df2c24f9c3054926e9edc226bc3f83e38

I will move this one to the next milestone as it is low priority.

felix-salfelder, 2020-09-01:
also (still) getting `parse_spice.hpp not found`. looks unrelated to adms or vams (?)