Qucs/qucs #550: Bad components preview in "Qucs Library Tool 0.0.19"

pidgeon777, 2016-07-24:
When selecting a component in the Qucs Library Tool window, sometimes the preview of the component is wrong.
Steps to reproduce the problem:

1) Tools -> Component Library (CTRL + 5).

[image]

2) Digit "741" (with no quotes) in the search box, and select the first result.

[image]

3) Press arrow down in the keyboard.

[image]

4) Press arrow down in the keyboard.

[image]

5) Press arrow down in the keyboard.

[image]

6) Press arrow down in the keyboard.

[image]

7) Press arrow down in the keyboard.

[image]

8) Press arrow down in the keyboard.

[image]

9) Press arrow down in the keyboard.

[image]

10) Press arrow down in the keyboard.

[image]

11) Press arrow down in the keyboard.

[image]

You can notice how, in many cases, the preview of the selected component is wrong.
I'm using Qucs v0.0.19 (qucs-win64-bin portable version), running it on a Windows 7 x64 machine.

in3otd, 2016-07-24:
this is almost surely related to #543... I did port the search code from the GUI to the Library Tool...

andresmmera, 2016-08-15:
I reckon that #556 solves the problem. Just for checking, could you test it?

guitorri, 2016-11-04:
Fix merged with #580.

closed 2016-11-04 by guitorri