Qucs/qucs #551: Simple DC sweep from tutorial creates strange plots.

JohnCC330, 2016-07-29:
I'm a new user to QUCS, and after the simple voltage divider, I hit the wall with the second tutorial (DC sweep of transistor parameters). I tried to enter the components as in the tutorial, but get strange results, as can be seen in the screenshot attached (also attached the circuit file, and a screenshot of same).
I find the plots rather intriguing - No idea where those pulses come from!
(My apologies for posting this as an issue, I couldn't find how to do attachments on Sourceforge)

BTW, this is QUCS 0.0.18 on Linux. I get *loads* of warnings in the log, but 

[image]
[image]
[transistor_dc.sch.gz](https://github.com/Qucs/qucs/files/389881/transistor_dc.sch.gz)

JohnCC330, 2016-07-30:
Ok... Not my best moment - Ground was missing. Still strange these signals appeared, isn't it?

closed 2016-07-30 by JohnCC330

andresmmera, 2016-07-31:
Hello John, 

Maybe too late,... but I usually employ [this schematic](https://dl.dropboxusercontent.com/u/21294824/BJT_StaticIV.sch) as BJT curve tracer.

Hope it helps :-)

JohnCC330, 2016-08-01:
Thanks Andrés. Yes, I had solved the problem, as I said in my previous mail, by adding a Gnd connection to the circuit. Your circuit works great, but I rather prefer to change Vce logarithmically - it gives much more information in the lower range, where the non-linear events are. Gracias!
[image]