Qucs/qucs #553: Search lib components tool doesn't place the proper component

andresmmera, 2016-08-10:
Hello, 

I've found a problem with the component library tool... It doesn't place the right part after using the searchbox. 

Steps to reproduce (OS: Fedora 24):
i) Open the component library tool (CTRL+5)
ii) Search a part. For example, select the NMOSFET tab and type IRF510 in the search line.
iii) Select the part on the list and then press the "Copy to clipboard" button.
iv) Paste the component into a schematic. Then go to the component properties (Double click) and you'll see that the name of the component is 2N3796.

I'll try to take a look to the code.

[image]
[image]

in3otd, 2016-08-12:
may be the same as #550 ...

andresmmera, 2016-08-12:
Yes,... I didn't check previous issues :S

closed 2016-08-12 by andresmmera