Qucs/qucs #557: Only numbers must be allowed for subcircuit ports

in3otd, 2016-09-07:
from https://sourceforge.net/p/qucs/discussion/311050/ : subcircuits ports should only use numbers but currently the user can enter an arbitrary text. This messes up loading back the schematic, especially if there are spaces in the strings (likely due to some compatibility stuff, cfr. the code)

In principle we could allow arbitrary text strings, but for the moment it will be easier to simply check that the port `Num` field contains a number and not text.

felix-salfelder, 2016-09-08:
`Num` does not sound much like `PortName`. i agree that it does not make much sense to use it for port names. (i suspect Num is used to "specify" the port order only.)

yet, i don't think there should be restrictions on strings a user puts into a parameter, because a sane parameter implementation will not inspect the parameter names.

what should happen, if there is a non-number and when? proper behaviour (imo) would be to catch the problem where it arises, i.e. the netlister should complain if it cannot figure out the port order.

ChrisOfBristol, 2017-12-13:
This is relevant https://sourceforge.net/p/qucs/discussion/311050/thread/1ce0d259/?limit=25#35a9
I posted this https://sourceforge.net/p/qucs/bugs/203/ as I wasn't aware of this bug report.

The port number can be entered as a digit but seems to be dealt with as if it was in string format and that will also fail to load. The same problem applies to resistors and may apply to other components but I've not checked. It makes the program almost unusable as I have to save after every few changes to names just in case they don't work.

in3otd, 2017-12-13:
um I need to check this again - here it seems to work also using strings as port names. And BTW how did you enter Port names with spaces, here I'm not able to enter a string with spaces as name, only chars and numbers.
Same for resistors, I'm not able to enter a Resistor name with spaces and just alphanumeric characters work fine.

ChrisOfBristol, 2017-12-14:
In the attached example, change the extension from txt to sch, load it then click on 'Positief', then in the field called 'Edit Properties Num, Number of the port within the...', change '1' to '1 a'. Save this and it won't reload, it gives the error "Format Error: Wrong 'painting' line format."

[Fan Thermostat 2 transistor.txt](https://github.com/Qucs/qucs/files/1560403/Fan.Thermostat.2.transistor.txt)

in3otd, 2017-12-14:
ok, thanks, the problem is with the Num field, not with the Name field I tried above. The strange thing is that I can change the Ports Num field to a string with spaces - and cause the loading error - but I cannot add any space in the Name field of the resistors or other components. Typing spaces in the Name field has no effect here. Looking at the code, entering spaces in the Name field should not be possible, there is a check on the characters entered.
Do you have the Qucs language set to something other than English? I'm wondering if that could make a difference.

ChrisOfBristol, 2017-12-14:
I have found 'Language Set after Reload' which is set to 'System Language'. 
I'm on Ubuntu Budgie 17.10 and the system has these settings:
Language: English (United Kingdom)
Formats: United Kingdom
Input Sources English (UK)

Don't be mislead by the foreign language words in the sample circuit, that was for posting on a non-English language site.

in3otd, 2017-12-14:
yep, I was thinking/hoping the language could make a difference, as said here I cannot enter any space for the Name field.
The check in the code uses a regular expression whose behavior I think may depend on some system setting; are you using a `locale` with UTF-8?

ChrisOfBristol, 2017-12-14:
```
chris@Dell-DXP051:~$ locale
LANG=en_GB.UTF-8
LANGUAGE=en_GB:en
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
chris@Dell-DXP051:~$
```

in3otd, 2017-12-17:
I've installed Ubuntu Budgie 17.10 in a Virtualbox with the same system and local settings reported above and tried Qucs there; I'm not able to replicate the behaviour you see, I cannot enter any space for the Name field of any component. Quite strange, don't know what could cause that issue.

ChrisOfBristol, 2017-12-17:
OK, I'll post something on the Ubuntu Budgie site in case they can think of something.

ChrisOfBristol, 2017-12-17:
UB replies:
_So it sounds like two issues. Something with your account. Second is something with the specific application version. Possibly you are missing a package dependency? Since you mentions space and characters possibly a font is missing? The application authors via their issue tracker is the best bet._
How can I check that all the necessary fonts are present?

in3otd, 2017-12-21:
The font stuff is managed by the Qt library we are using and also the check for spaces in the components Name field is done using a Qt function, so it could be a bug there.
Anyway, Qt will automatically substitute an installed font if the one the application asks for is not present; by default Qucs asks for an "Helvetica" font, which is very likely not present (it's not a free font) - on my system a "Nimbus Sans L" font is used instead.
You might try to explicitly choose another font (File->Application Settings->Font , exit and re-start Qucs after changing font) to see if it makes a difference.

If you can you recompile Qucs passing `--enable-debug` to `configure` it will show on the console quite a number of debug stuff, *maybe* there could be a hint there about what's wrong/missing.

ChrisOfBristol, 2017-12-21:
I selected Deja Vu Sans as it's commonly used in Ubuntu. It's now doing what happens on the guest account - see post three up (the posts don't appear to have numbers!)

Compiling - haha! I tried that a few times when I first started using Linux years ago i I never got it to work.

in3otd, 2017-12-27:
in case you have some spare disk space and time, here are some commands that should work there, as they come from what I did to test Qucs on Ubuntu Budgie 17.10 in the Virtualbox.
The steps below will install Qucs in `/usr/local`; I think you currently have Qucs in `/usr` so this will not overwrite it but you may want to double check.

- first install some needed stuff
```
sudo apt-get install git
sudo apt-get install build-essential
sudo apt-get install libqt4-dev libqt4-qt3support
sudo apt-get install automake libtool gperf flex bison
sudo apt-get install libcanberra-gtk3-module
```
- then install `adms` from the sources; you may try skipping this as you may already have `admsXml` installed from the previous Qucs install you did. Just type in a shell `admsXml -v` to see if it's already there.
```
wget http://sourceforge.net/projects/mot-adms/files/adms-source/2.3/adms-2.3.6.tar.gz
tar zxvf adms-2.3.6.tar.gz 
cd adms-2.3.6/
./configure
make
sudo make install
sudo ldconfig
cd ..
```
- finally you can get the Qucs sources and compile that
```
git clone https://github.com/Qucs/qucs.git
cd qucs/
./bootstrap 
```
- now if you already had `admsXml` installed type
`./configure --disable-doc --enable-debug`
if you just installed `admsXml` instead, type
  `./configure --disable-doc --enable-debug --with-mkadms=/usr/local/bin/admsXml`
- then
```
make
sudo make install
sudo ldconfig
cd ..
```

now typing `qucs` should start this local Qucs install

To remove this Qucs:
```
cd qucs
make uninstall
```

ChrisOfBristol, 2017-12-27:
chris@Dell-DXP051:~$ admsXml -v
[usage..] <release name="admsXml" version="2.3.6" date="Dec 27 2017" time="19:08:46"/>
chris@Dell-DXP051:~$ ^C
Does that mean that admsxml is installed or not?

ChrisOfBristol, 2017-12-27:
I got a few error messages when running Qucs, but it runs and won't allow me to enter spaces into a component name, or to move the cursor of the end of the name.

```
chris@Dell-DXP051:~$ qucs
Gtk-Message: Failed to load module "canberra-gtk-module"
Debug: trying to read "/usr/local/share/qucs/library/Bridges.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Diodes.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Ideal.lib" 
Debug: trying to read "/usr/local/share/qucs/library/JFETs.lib" 
Debug: trying to read "/usr/local/share/qucs/library/LEDs.lib" 
Debug: trying to read "/usr/local/share/qucs/library/MOSFETs.lib" 
Debug: trying to read "/usr/local/share/qucs/library/NMOSFETs.lib" 
Debug: trying to read "/usr/local/share/qucs/library/OpAmps.lib" 
Debug: trying to read "/usr/local/share/qucs/library/PMOSFETs.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Regulators.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Substrates.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Transistors.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Varistors.lib" 
Debug: trying to read "/usr/local/share/qucs/library/Z-Diodes.lib" 
```

in3otd, 2017-12-31:
so the Qucs you compiled from the sources works as expected... (BTW, not sure I understood what you meant by "[won't allow] to move the cursor of the end of the name").
I got rid of a similar Gtk warning after installing the `libcanberra-gtk3-module`, maybe for some reason you need to install `libcanberra-gtk-module` instead (just guessing). The Debug messages are ok.

I can try to install Qucs the same way you did initially to at least check if I can reproduce the issue here using that. Could you provide the command you used to install Qucs?

ChrisOfBristol, 2017-12-31:
> (BTW, not sure I understood what you meant by "[won't allow] to move the cursor of the end of the name"). Qucs...won't allow me to...move the cursor of[f] the end of the name
Is there a way I can capture all the commands I have typed since I installed the OS, so I can check?

in3otd, 2018-01-01:
> Qucs...won't allow me to...move the cursor of[f] the end of the name

you mean that when you try to edit a component Name the cursor is always at the end of the line, i.e. cannot be moved using Home or the arrow keys? Because the usual behavior here is that the cursor can be moved around while editing a Name.

Besides `history`, I do not know other commands for getting a list of what has been typed in a terminal - but the list may be incomplete and retain only a limited number of commands.

ChrisOfBristol, 2018-01-01:
It allows me to move the cursor anywhere along the name, as it should do. It won't let it off the end to the right where there is no text, which is as it should be. I think it was allowing it off the end of the text before.

I think this is about the point after the last installation?
` 1871  sudo apt install vlc
 1872  budgie-desktop --version
 1873  sudo apt update && sudo apt dist-upgrade
 1874  exit
 1875  sudo apt purge mutter
 1876  uname -a
 1877  sudo add-apt-repository ppa:ubuntubudgie-dev/mutter-crash
 1878  sudo apt update && sudo apt upgrade
 1879  sudo ppa-purge ppa:ubuntubudgie-dev/mutter-crash
 1880  sudo apt update
 1881  exit
 1882  gufw
 1883  sudo apt install gufw
 1884  gufw
 1885  sudo apt install thunderbird
 1886  sudo apt install firefox
 1887  sudo apt remove chromium
 1888  sudo apt install chromium
 1889  sudo apt install audacity
 1890  sudo apt install zim
 1891  sudo apt install boinc_client
 1892  sudo apt install boinc_manager
 1893  easytag
 1894  sudo apt install easytag
 1895  easytag
 1896  sudo apt install brasero
 1897  brasero
 1898  rsync -tzhhP <rsync://cdimage.ubuntu.com/ubuntu-budgie/daily-live/20171016/artful-desktop-amd64.iso>
 1899  rsync -tzhhP <rsync://cdimage.ubuntu.com/ubuntu-budgie/daily-live/21071016/artful-dektop-amd64.so>
 1900  rsync -tzhhP <rsync://cdimage.ubuntu.com/ubuntu-budgie/daily-live/21071016/artful-dektop-amd64.so
 1901  rsync -tzhhP <rsync://cdimage.ubuntu.com/ubuntu-budgie/daily-live/20171016/artful-desktop-amd64.iso
 1902  sudo apt dist-upgrade
 1903  sudo apt autoremove
 1904  sudo apt update
 1905  exits
 1906  exit
 1907  gnome-desktop-item-edit ~/Desktop/ --create-new
 1908  sudo apt dist-upgrade
 1909  sudo apt update
 1910  torrent
 1911  bittorrent
 1912  qbittorrent
 1913  ktorrent
 1914  ctorrent
 1915  transmission
 1916  sudo apt dist-upgrade
 1917  sudo apt update
 1918  lscpu
 1919  sudo gedit /usr/share/themes/elementary/gtk-3.0/scrollbar.css
 1920  gsettings set com.canonical.desktop.interface scrollbar-mode normal
 1921  sudo gedit /usr/share/themes/Ambiance/gtk-3.0/gtk-widgets.css
 1922  sudo gedit /etc/environment
 1923  sudo cp /etc/environment /etc/environment.ORIGINAL
 1924  sudo gedit /etc/environment
 1925  chromium
 1926  sudo gedit /etc/environment
 1927  sudo apt-get install flashplugin-installer
 1928  synapse
 1929  dir ~/.config/gtk-3.0/gtk.css
 1930  dir /root/.config/gtk-3.0/gtk.css
 1931  sudo dir /root/.config/gtk-3.0/gtk.css
 1932  dconf-editor
 1933  sudo apt install dconf-editor
 1934  dconf-editor
 1935  sudo gedit ~/.config/gtk-3.0/gtk.css
 1936  sudo gedit ~/.config/gtk-3.0/gtk.css~/.config/gtk-3.0/gtk.css
 1937  sudo rm ~/.config/gtk-3.0/gtk.css
 1938  exit
 1939  ls /usr/share/themes/elementary/gtk-3.0/ 
 1940  /usr/share/themes
 1941  dir /usr/share/themes
 1942  sudo apt install clementine
 1943  gsettings set org.gnome.nautilus.preferences sort-directories-first true
 1944  gsettings set org.gnome.nautilus.preferences default-sort-order 'name'
 1945  sudo add-apt-repository ppa:vlijm/budgie-extras && sudo apt install budgie-hotcorners-applet
 1946  sudo apt update
 1947  sudo add-apt-repository ppa:vlijm/budgie-extras && sudo apt install budgie-hotcorners-applet
 1948  sudo apt install clementine
 1949  evince
 1950  sudo add-apt-repository ppa:inkscape.dev/stable
 1951  sudo apt update
 1952  inkscape
 1953  sudo apt install inkscape
 1954  inkscape
 1955  clementine
 1956  sudo apt install vlc
 1957  vlc
 1958  sudo gedit /etc/environment
 1959  sudo /etc/environment
 1960  sudo gedit /etc/environment
 1961  clementine
 1962  di Spare
 1963  dir Spare
 1964  dir /Spare
 1965  dir /dev/Spare
 1966  dir /mnt/Spare
 1967  dir /mnt
 1968  dir /mnt/Spare/*
 1969  dir /mnt/Spare/*.*
 1970  ./backup.sh
 1971  gnome-desktop-item-edit ~/Desktop/ --create-new
 1972  dconf reset /org/gnome/desktop/interface/gtk-theme
 1973  sudo gedit /etc/environment
 1974  sudo gedit ~/.config/gtk-3.0/gtk.css
 1975  sudo shutdown -r now
 1976  wget -q -O- http://archive.getdeb.net/getdeb-archive.key | sudo apt-key add -
 1977  sudo apt update
 1978  sudo apt remove electric
 1979  sudo apt install ngspice
 1980  locale
 1981  sudo apt install libdvdcss libdvdread4 libdvdnav4
 1982  sudo apt install libdvdcss2
 1983  sudo apt-get install libdvdread4 libdvdnav4
 1984  sudo apt-get install libdvd-pkg
 1985  sudo apt update
 1986  sudo apt install libdvd-pkg
 1987  sudo dpkg-reconfigure libdvd-pkg
 1988  ./qucs.sh
 1989  admsXml -v
 1990  ./configure --disable-doc --enable-debug
 1991  cd qucs/
 1992  ./bootstrap
 1993  ./configure --disable-doc --enable-debug
 1994  make
 1995  sudo make install
 1996  sudo ldconfig
 1997  cd ..
 1998  qucs
 1999  history
 2000  history > history.txt`

ChrisOfBristol, 2020-02-20:
I hadn't used this software for quite a while but needed a circuit drawing so came back to it. Did a fair bit of work then found my file wouldn't load - and eventually remembered this bug. I've managed to bodge the file to make it loadable, so I can at least see the drawing. 

This bug has been around for 4 1/2 years and the last activity on this thread was 2 years ago. Either this bug has been abandoned or the software itself has been abandoned - either makes it unusable for me.

guitorri, 2020-02-20:
So far you seem to be the only user affected. Sorry, your bug report sunk to the bottom of the pile.
Which version are you using now? 
Is the attach above still giving the same error? 
Perhaps we can try to fix the port handling to accept any string... it would be nice thou if we could reproduce the error first.

ChrisOfBristol, 2020-02-20:
@guitorri Thanks for your helpful response to my somewhat blunt post.....I was rather cross about the loss of my work when I posted it.
"Version 0.0.19 (unknown) " on Ubuntu Budgie 19.10.
I'm not sure what you mean by "_Is the attach above still giving the same error?_"

in3otd, 2020-02-20:
uh, indeed quite some time has passed :grin:

the main problem for fixing this bug is that we cannot reproduce it.
I've tried again to have two components with spaces in their reference (e.g. `R 1` and `R 2`) but I cannot insert any space (and this is the expected behaviour).
I've then manually edited a schematic file on disk to have components reference with spaces and, while the file still loads, those components are not really loaded correctly, as the text position is off and one has the "short-circuited" symbol over it (the green cross you mentioned [above](https://github.com/Qucs/qucs/issues/557#issuecomment-353375887)). But this is expected, as space in component references are not allowed.

Can you check if the spaces you are able to insert in the components references are really spaces? Just copy/paste the text [here](https://www.soscisurvey.de/tools/view-chars.php) or [here](http://www.asciivalue.com/index.php) and see if it shows any strange character.

ChrisOfBristol, 2020-02-20:
The text below all appeared on one line it your first link:
<60
0x3cR82
0x52·32
0x20T84
0x54h104
0x68r114
0x72e101
0x65e101
0x65·32
0x20r114
0x72e101
0x65s115
0x73i105
0x69s115
0x73t116
0x74o111
0x6fr114
0x72s115
0x73·32
0x20149
0x31·32
0x20553
0x35654
0x36048
0x30·32
0x20250
0x32553
0x35048
0x30·32
0x20-45
0x2d149
0x31149
0x31149
0x31·32
0x20-45
0x2d250
0x32654
0x36·32
0x20048
0x30·32
0x20351
0x33·32
0x20"34
0x22553
0x35048
0x30·32
0x20O79
0x4fh104
0x68m109
0x6d"34
0x22·32
0x20149
0x31·32
0x20"34
0x22250
0x32654
0x36.46
0x2e856
0x38553
0x35"34
0x22·32
0x20048
0x30·32
0x20"34
0x22048
0x30.46
0x2e048
0x30"34
0x22·32
0x20048
0x30·32
0x20"34
0x22048
0x30.46
0x2e048
0x30"34
0x22·32
0x20048
0x30·32
0x20"34
0x22250
0x32654
0x36.46
0x2e856
0x38553
0x35"34
0x22·32
0x20048
0x30·32
0x20"34
0x22e101
0x65u117
0x75r114
0x72o111
0x6fp112
0x70e101
0x65a97
0x61n110
0x6e"34
0x22·32
0x20048
0x30>
This was from the second one, "32" looks like a space to me.

in3otd, 2020-02-20:
yes, those look like spaces...
Just another experiment: what happens if you try to paste a component reference with spaces? For example, copy this text, `R 1` and try to paste it in a component reference in Qucs.
Here nothing happens, the text is not pasted. When trying with a string without spaces it works as expected (the text is pasted).

Looking at the code, the allowed text for component references is defined [here](https://github.com/Qucs/qucs/blob/develop/qucs/qucs/qucs_actions.cpp#L1176), which from the Qt docs means *Matches a word character (QChar::isLetterOrNumber(), QChar::isMark(), or '_').*. The only suspect thing is [`QChar::isMark()`](https://doc.qt.io/archives/qt-4.8/qchar.html#isMark) that could potentially match a lot of strange characters (see [here](https://doc.qt.io/archives/qt-4.8/qchar.html#Category-enum)). I've tried to input some strange unicode characters but was not able to, maybe it depends on the terminal/system settings.

ChrisOfBristol, 2020-02-21:
@in3otd Interesting result! I opened a working file with no spaces in component names. I then copied your example above into a component name, saved the file, closed and re-opened QUCS and opened the file. It loaded, but the component was missing.

Copying the example above and pasting it into this message looks like this
R 1
is that relevant?

in3otd, 2020-02-21:
ok, still is strange that you can paste a string containing spaces in a component reference. Here I've tried in several ways and never been able to do that, which also agrees with the string checks in the code.
The fact that the component was missing after you gave it a name with spaces is somewhat expected, since normally spaces are used as fields separators in the schematic file and that spurious space likely prevented reading back the component properties correctly.

ChrisOfBristol, 2020-02-21:
Do you think you will be able to fix it by trying a different way of checking the input characters?

in3otd, 2020-02-21:
Not sure if that will be useful: when you compiled Qucs from the sources a couple of years ago you [reported](https://github.com/Qucs/qucs/issues/557#issuecomment-354171484) that it was behaving as expected and, checking the source file history, the last time the input characters check was changed was 14 years ago (!).
Do you think you could try again to recompile Qucs and check how it behaves?

ChrisOfBristol, 2020-02-21:
I've recompiled it as above and it no longer permits the entry of anything other than alphanumeric characters. That's a bit of a pain, but a lot better than losing the file.

I note that this is Version 20. What is the version history, when was V20 released?

in3otd, 2020-02-21:
good that at least that version behaves as expected. By the way, where did you get the version which shows the issue?

The latest release is 0.0.19; the one you compiled from the sources of the develop branch should, hopefully, become 0.0.20 shortly.

ChrisOfBristol, 2020-02-22:
Post 7 on 14th December 2017 stated that "I'm on Ubuntu Budgie 17.10"  and from the QUCS website "22 | January | 2017 | Released Qucs 0.0.19", so first time round I was on V19. This time round I'm on Ubuntu Budgie 19.10 and still on QUCS V19 - from the Ubuntu Budgie Software Installer. I think the version I compiled in 2017 must have moved on from 19 and had some mods intended for V20.

ChrisOfBristol, 2020-04-15:
A **warning** about compiling and using V0.0.20 as suggested above. It won't work on Ubuntu 20.04 which is due this month (or any derived distributions), I think it's because Qt4 is no longer available. That means you would have to use the Qucs version which comes with it, which is a Snap of V0.0.19. Any schematics you have created/edited with V0.0.20 won't load. 

I used a text editor to bodge the first line of the schematics to read "<Qucs Schematic 0.0.19>" and the files loaded. I don't know whether that would create complications for some files. If anyone still wants to try V0.0.20 I can upload a bash script (taken from in3otd's post above) to install it.

felix-salfelder, 2020-09-26:
please open and/or reopen if the issue persists in the current version.

NB: there are ways to run recent snapshots on systems without qt<=4. watch out for "appimage" or so.

closed 2020-09-26 by felix-salfelder