Qucs/qucs #560: Qt5 Support Expected...

jiapei100, 2016-09-10:
Hi:

It looks like qucs is heavily based on Qt4, instead of Qt5...
Just wonder if there is a plan to shift from Qt4 to Qt5?

Cheers
Pei

in3otd, 2016-09-10:
yep, as shown in the [Qucs roadmap] (https://github.com/Qucs/qucs/wiki/Roadmap) it is "planned" to move to Qt5, but the problem is that the porting from Qt3 is not really completed still as we still use quite some `Qt3Support` classes which are no longer available in Qt5 and nobody is currently actively working on fixing that first.
There was a branch dedicated to removing the `Qt3PtrList` stuff but unfortunately the work there was never completed.

andresmmera, 2016-09-10:
For completeness, Q3ScrollView should be removed too... I think it could be replaced by QScrollArea, but after having done some experiments... I'm not completely sure

guitorri, 2016-09-14:
The problem really is the graphics. We need to port **all** components to QGraphicsItem and re-implement **all** the mouse interactions. Q3ScrollView is probably taken care by the QGraphicsView. The Q3PtrlList are  the "database" representing the schematic components. It would be better/sane if we could refactor what we have into a model/view. Hence having a clear separation between the database and the graphics. This opens up the possibility of having the database as a library/module, which could be loaded/exchanged with other tools. 

I have a branch I started with the port of the components. That sort of worked but does not scale. It makes more sense to have a generic/primitive graphic component that can load the symbol and properties of all other components from a library (be it plain text of binary).

I did a bit of the mouse interactions, but I feel I am doing it wrong. I  will push the code later this week  so you guys can have a look.

felix-salfelder, 2016-09-14:
> port all components to QGraphicsItem 

there seems to be an overlap with the SYMBOL class that i recenly drafted (in order to fix/extend qucslib).

it has a drawing routine already, and it is working (roughly) as intended. the use case in qucslib is not fully complete (sorry about that). that needs to be pulled over to qucs (the gui) some day... we probably can fix 2 things in one go.

felix-salfelder, 2016-09-14:
> generic/primitive graphic component that can load the symbol and properties of all 

yes, there should be some object that you can attach a SYMBOL to. a symbol then can be anything, such as a reference to an XML file/item or an svg file or a geda symbol or just anything.

dont mix the graphic primitive with the XML parser that's a different layer, and must be pluggable.

nvdl, 2016-11-10:
I think, we had discussion around it a while ago. The more I dig into the GUI code, the more I feel I am wasting time. That part needs a real overhaul.

For Qt5 port, I have done some basic UI design based on PyQt. My thinking is to make that part isolated from other tools like simulator, plotter etc. The GUI application calls those tools and development of each tool goes independently to avoid source conflicts and many people modifying the same code region.

Is it OK to move to PyQt (or PySide) for GUI part and possibly Netlisters as well?

The simulators can be binaries and compiled separately.
Output of a simulator goes through different data storage classes for different formats.

Post data processors (e.g. plotters) will be either Python modules or standalone binaries called by GUI having access to the drawing part of GUI through a defined interface.

Thus, if some one wants to implement another way of plotting data, they can write their own code in any language as long as GUI can pass pointer to data (stored in a specific format) to it and it can call back drawing routines of GUI.

For example:

`GUI.addPlotter(type = "abc", locationAndSize = (x, y, z, w, h), ref = "plot1")`
`GUI.plot1.addPlot(myDataSet1)`

A component can be a Python class pointing to a model, GUI image and exported ports.
Even sub circuits can be designed that way with the related class handling all the internals; pointers to models, their ports and exported (to GUI) ports.

fer-rum, 2016-11-10:
Okay, since we opened "Pandoras refactoring box": Why stop there?
Separating the GUI is a good (and very necessary) step, but why refactor into the QtGui system which is supposed to deprecate at some point.
So why not switch to QML since this is where the current Qt-development is headed and catch up with the state of the art?

Also: instead of composing the application of dozens of binaries communication via IPC-hell or some other quirky mechanism (and who have to be maintained separately), how about defining a proper module interface, put all the parts into modules (Since there already is a mechanism for handling them in Qt) and build a proper, extensible and customizable application?

Qucs is quite big and thus there is no way it can get away with deeds half-done in the long run.

nvdl, 2016-11-10:
No experience with QML. Any advantage or just a new shiny tool?
Can you explain more on the modules and interfacing? Any example or reference?

> Qucs is quite big and thus there is no way it can get away with deeds half-done in the long run.

This is what is happening now. People are hacking on their own and the PRs get incompatible in a few months as someone else already changed some stuff in that region.
It is important to stabilize the core and keep it as it is without changing it just for the addition of new features. New features are bringing in new issues and the project is swinging up and down.

If I am convinced that QML has advantages, we can start with that. I have already created main UI but that is very basic. QML ignorant here.

The main focus should be multi-platform compatibility and script based extensibility.
It is OK to have Qucs scripting engine but I don't think that is a necessity.

fer-rum, 2016-11-15:
In short QML is a description language for your UI layout and connections to the UI backend, tweaked to work in combination with Qt. 
So you can e.g. take advantage of the signal/slot event handling right from the beginning. It further allows to integrate JavaScript code (But don't overdo it for the sake of good design :smile: ). Also it is really nice for UI design and debugging, since any change you make in the deployed QML files is instantly reflected in the running application. And the [documentation](http://doc.qt.io/qt-5/qtqml-index.html) is quite extensive.

It is a fixed part of Qt since version 4.8 (iirc) and there are [many examples](http://doc.qt.io/qt-5/qtquick-codesamples.html) out there (They even come as example projects with your QtCreator by default.)

We used QML+Qt5 in an industry project of comparable size and it worked out quite well so far.

> This is what is happening now. People are hacking on their own and the PRs get incompatible in a few months as someone else already changed some stuff in that region.

I am not that familiar with the Qucs community structure. Is there any project lead/ teams or some kind of "management"?

> Can you explain more on the modules and interfacing? Any example or reference?
I was talking about [these ones](http://doc.qt.io/qt-5/plugins-howto.html#the-low-level-api-extending-qt-applications) as an alternative to having dozens of separate applications. These all tend have a different styles of communication, so instead with modules one could come up with a well-defined interface every module has to adhere to. However, the idea still is more of an concept at this stage.

OursDesCavernes, 2018-04-12:
Hello,
Do you have updates on Qt4 to Qt5 migration ?
Gentoo is removing Qucs because Qt4 is being deprecated.
Thx

guitorri, 2019-01-10:
IMHO the next steps on the UI front are to refactor the mostly broken mouse actions with proper Command and Undo functionalities. See examples [1] [2].

I am a little lost on the branches. Where should such code land? 
In the `qt5-*` or `develop+qt5*`series?

[1] <https://doc.qt.io/qt-5/qtwidgets-tools-undoframework-example.html>
[2] <https://doc.qt.io/qt-5/qtwidgets-tools-undo-example.html>

felix-salfelder, 2019-01-10:
On Thu, Jan 10, 2019 at 12:21:54PM -0800, Guilherme Brondani Torri wrote:
> IMHO the next steps on the UI front are to refactor the mostly broken
> mouse actions with proper Command and Undo functionalities. See
> examples [1] [2].

one of the cruxes seems MouseActions::selectElement. selection is much
simpler and more efficient with the Qt5 scene. I don't yet know how to
delegate actions to widgets in the scene (i.e. diagrams). But i am
getting closer to making the non-diagrams work.

[1] and [2] looks useful, i was indeed wondering how to do the undo
stack. The implicit undo stack breaks down with removing the element
erase/create cycles buried in the mouse actions.

> I am a little lost on the branches. Where should such code land?
> In the `qt5-*` or `develop+qt5*`series?

[3] will become develop+qt5-11, it addresses deadlocks in -10... but
there are issues left.

thanks

[1] https://doc.qt.io/qt-5/qtwidgets-tools-undoframework-example.html
[2] https://doc.qt.io/qt-5/qtwidgets-tools-undo-example.html
[3] https://github.com/felix-salfelder/qucs/tree/refactor%2Bqt5

rubund, 2019-01-21:
>> I am a little lost on the branches. Where should such code land?
>> In the `qt5-*` or `develop+qt5*`series?
> will become develop+qt5-11, it addresses deadlocks in -10... but
> there are issues left.

Why is there not just ONE branch in github.com/Qucs/qucs called "qt5", where all qt5 specific things go in? One can regularly merge in from master/develop.

I think that would make it easier for other people to help out.

felix-salfelder, 2019-01-21:
On Mon, Jan 21, 2019 at 05:33:54AM -0800, rubund wrote:
> Why is there not just ONE branch in github.com/Qucs/qucs called "qt5",
> where all qt5 specific things go in? One can regularly merge in from
> master/develop.

in an ideal world, a sequence of merges into develop should be able to
do the trick. but we are currently lacking these merge requests as well
as the reviewers.

nonlinear commit history does not seem to work, as it breaks bisection,
rebase and review. the best approximation to the above is hence to use
revision numbers in the branch names.

note that we (usually) rebase before we merge into develop.

there are a couple of things to be fixed (call it "clean up" or
"refactoring") in develop before a switch to qt5 seems realistic. I
collect these changes in my "refactor" branch (a merge request, not a
branch in the main repo.) Most parts of it could go into develop,
without breaking anything (except CMake).

the refactor+qt branch are a couple of commits on top that will do the
actual switch. it will be squashed into one commit in the end.

> I think that would make it easier for other people to help out.

I will appreciate and consider commits against both, my refactor branch
and the refactor+qt5 branch. the latter will become develop+qt5-11 as
soon as all tests pass, likely after FOSDEM.

glixx, 2019-02-20:
> Why is there not just ONE branch in github.com/Qucs/qucs called "qt5", where all qt5 specific things go in? One can regularly merge in from master/develop.
>I think that would make it easier for other people to help out.

Yes

felix-salfelder, 2019-02-20:
On Tue, Feb 19, 2019 at 09:33:19PM -0800, glixx wrote:
> > Why is there not just ONE branch in github.com/Qucs/qucs called "qt5", where all qt5 specific things go in? One can regularly merge in from master/develop.
> >I think that would make it easier for other people to help out.
> 
> Yes

A quick update: we sort of decided to do the ground work in the develop
branch. first, do the changes that do not require qt5, and do not break
anything. those things we cannot get away without. examples, from easy
to hard.

* tests. there are some and some new ones in develop+qt5. it would be
  really helpful to have those in develop.
* import qt_compat.h. essentially, replace Qt3PtrList header use, but
  there is more.
* refactor mouse actions. mouse actions are entangled with component
  typology. It is pretty nonobvious how to fix it. look at the Diagrams.

the first ones will easily get you into business, the last one is the
main missing bit in my branch. there are more. i will rebase once
develop has moved.

thanks

rajasrijan, 2020-06-18:
Is there any update on qt5 support? Ubuntu 20.04 has dropped qt4 altogether.

felix-salfelder, 2020-06-18:
On Wed, Jun 17, 2020 at 08:31:31PM -0700, Srijan Kumar Sharma wrote:
> Is there any update on qt5 support?

Work in progress. Looking for help.

> Ubuntu 20.04 has dropped qt4 altogether.

You can still use statically linked binaries, containers or VMs. I think
this is what people do.

Rmano, 2020-06-18:
If you are interested, I managed to create an AppImage on my 18.04 in https://github.com/Qucs/qucs/issues/1006#issuecomment-643792968 --- it is working on a 20.04 virtual machine, at least after a half-an-hour test.

It would be very nice to add the AppImage generation to the release CI, but I am a bit at a loss here, so help is welcome.

rajasrijan, 2020-06-18:
@felix-salfelder would love to help. I just don't know where to start.

felix-salfelder, 2020-06-19:
On Thu, Jun 18, 2020 at 11:41:06AM -0700, Srijan Kumar Sharma wrote:
> I just don't know where to start.

please read the mailing list or the code. in fact there is no proper
todo list, and could need help with that, too. e.g. find what is hard to
do, what is easy, and what is there...

thanks

rajasrijan, 2020-06-22:
looking at the code it seems like a lot of stuff just needs to be copied. I'm try to get wire placing working.

felix-salfelder, 2020-07-28:
The OP question has been answered.

closed 2020-07-28 by felix-salfelder

cederom, 2021-02-08:
Hello from year 2021 :-) `qucs` port was deleted from FreeBSD in March 2019 because it is not Qt5 compatible and Qt4 was removed from the OS. This utility looks amazing, would be really nice to have it working with Qt5 or Qt6 :-)
https://www.freshports.org/cad/qucs/

thomaslepoix, 2021-02-08:
Hey! Many distros are currently in this case, the port to Qt5 is in progress. For now the best alternative is the AppImage from Rmano, but I don't know if AppImages run on FeeeBSD.

cederom, 2021-02-08:
Yes I saw that, thanks @thomaslepoix :-) macOS is based on FreeBSD but these are different OS and does not run their binaries.

QT5 is also step back as QT6 release was recently introduced to the public :-)

Maybe switching to Python QT would be a reasonable choice between portability and maintenance? Then QUCS cuold be installed on any platform as easy as `pip install qucs` :-)