Qucs/qucs #562: HiDPI could be improved

guiweber, 2016-09-16:
On a 15" 4k screen, only the first few letters of the components are displayed. The font size is fine, but the component images are very small. Some menu icons are so small that it's even hard to click on them.

This was tested under Fedora Linux and Windows 10 with the default DPI scaling.

[image]

guiweber, 2016-09-23:
The property list in the Edit Component window is also too small which makes editing the values tedious. The window can be resized but this has to be done every time you edit a component since the size of the window is not saved.

[image]

Infi9292, 2017-09-25:
If you are using Windows 10, you can update properties of the qucs.exe program file located in Program Files (x86)/qucs/bin. Right click on the qucs.exe file, this will open the property window. Go to Compatibility tab and check box 'Override high DPI scaling beavior, Scaling performed by: '. Pick 'System (Enhanced)' and save.

The next time you will call qucs.exe, you will get a nice scaled window. You may have to enter the same setup for other program files in the qucs/bin directory if you need to run them

[image]

[image]

thomasska, 2019-10-31:
On Windows 10 (Version 1903 Build 18362.416) this no longer works :(
On a 4K 15" laptop, the micro-icons and large-text occur with all combinations of high-DPI settings.
It also affects QUCS even when shown on a full-HD display, if there is a 4K monitor active on the system.