Qucs/qucs #563: Discussion: port graphics to Qt Graphics View Framework

guitorri, 2016-09-21:
To be able to get rid of Qt3Support, the current graphics need to be ported/reimplemented. The proposal is to use the [Graphics View Framework](http://doc.qt.io/qt-4.8/graphicsview.html). Some old bits of discussion in our [Wiki](https://github.com/Qucs/qucs/wiki/QEP%3A-QGraphicsView).

Some time ago I did a bit of exploration (very rough and unstable!), in the following branch:
  - https://github.com/guitorri/qucs/tree/QGraphicView-exploration

It is based in a old version. To build you can:
```./autogen.sh && ./configure && cd qucs/qucs && make```
or
```cd qucs/qucs && mkdir build && cd build && cmake .. && make```

I did a bit of brute force to get the following working:
- load/render most of the existing schematics 
- Pick and place (insert) components from dock
- Move and edit (some?) component properties
- Mouse hover changes for visual feedback and tooltip (eye candy and debug)
- Partial fix for zoom and show all
- Partial fix for scrolling (bars and key arrows)
- partial fix for schematic area and grid
- (other things you can see in the commit history)

For all the other features it can easily crash...

The following important things are not yet done:
- mouse interactions for drawing/connecting wires
- the contents/curves inside of _diagrams_
- majority of mouse interactions (rotate, select, ...) 
- netlist/schematic to file
- other things I don't see now...

So, it is certainly doable. But should we continue down this path? Do you see another route? Shall we do graphics first and refactor the model/view later?

Please take a careful look, let us discuss and try to move forward with this important topic.

felix-salfelder, 2016-09-21:
two questions

> "The proposal is to make all components subclasses of QGraphicsItem"

what does "components" refer to in this sentence? imo only graphical items should inherit from QGraphicsItem.

> So, it is certainly doable. 

yes, that makes sense. do you see any chance of splitting this down, such as "implement graphical items properly" (one at a time) then "switch schematic view to QGraphicsView".

guitorri, 2016-09-22:
Yes, components should be replaced by graphical item. Any item that is added to the scene and rendered in the view should inherit from QGraphicsItem. 

What do you mean by "implement properly one at a time"? Switching the view to QGraphicsView was the first thing I did in order to see the implications. We can certainly go thru the global changes and merge one by one as we see more clearly the path forward. 

We need at least one scene and one view.  If I recall correctly after loading objects from the schematic file, on instantiation they are added to the scene and references are kept in lists of pointers, the scene is rendered and modified through the view. So, my question is whether we keep lists of pointers or replace it by another data-structure which we can more easily traverse. Ideally, I guess, the schematic data-structures should not depend on Qt.

felix-salfelder, 2016-09-23:
> What do you mean by "implement properly one at a time"?

haven't looked at the code much, but the todo list is very long... looks like the first step breaks everything, and then requires lots of refactoring... just wondering if it makes sense to refactor first and do the switch afterwards. (i will have a closer look!)

> another data-structure which we can more easily traverse.

i guess, a double linked list is optimal for the purpose. of course every "thing" must hold a reference to the list token. i don't know how this is currently (not) the case.

> Ideally, I guess, the schematic data-structures should not depend on Qt.

yes, definitely.

ldpgh, 2016-10-06:
In case of big refactoring ... could you take in to your considerations something which may allow on the long term to wrap all elements for Python  (e.g. SIP for pyQt, ?? for pySide).

felix-salfelder, 2016-10-06:
ldpgh. sounds interesting. can you expand on it? e.g. are you referring to "circuit elements" or to "graphical elements"?

(there is modularisation in progress, would you consider to join/contribute?)

ldpgh, 2016-10-06:
Do you know Cadence Virtuoso & Skill   .... respective ... FreeCAD & Python. This is what I have in my mind.

If "circuit elements" are refering to elements itself usually residing in a kind of library ... NO, interpreter/binary are just using the elements.
Use the lib elements, place it and/or do some automatic fancy things (e.g. mixed-signal netlist) ... YES

I was looking already for emails/discussions regarding modularization but could not really find something.

Join/Contribute ... yes, no, maybe ... I'm not sure. At least I'm always curious.

felix-salfelder, 2016-10-06:
> modularization but could not really find something

i think it makes sense to add a "modularity" label to attach to issues/PRs. can't find the button, maybe i lack the privileges...

fer-rum, 2016-10-06:
On the danger of digressing from the discussion: Have you considered/ evaluated making the full leap forward and using QML?

ldpgh, 2016-10-06:
**If** (QML is meant for GUI-design at the development stage)
  - I don't care if at the end the Python-wrapper are available - it is the SW-experts area :-)

**Else**
 - Just for my curiosity ... how would you add a netlist-formatter using QML at CAD-use stage?

fer-rum, 2016-10-06:
QML is a complete alternative approach to GUI design compared to the traditional "programmed" QGraphics…-way used in older Qt-applications. We used QML at a work project for a comparable sized engineering application and it handles itself quite well, especially with respect to extensibility.
I have no experience yet mixing the two approaches.

### TLDR 
__QML from top to bottom__ = probably more initial work (more learning required to do every little bit you want from your UI) but then you have one nice modern approach to rule them all
__QML for user CAD only__ = Nicer for a smoother development transition, but may contain unforeseen pitfalls.

Maybe reading a few pieces from the [qmlbook](http://qmlbook.github.io/) (chapter 4 onward, if you want to skip the generic Qt-stuff) can help you to form a mental image how it might be.

ldpgh, 2016-10-06:
"user CAD" != "CAD-user" ... or do you mean the same?

fer-rum, 2016-10-10:
I am not sure, if I mean what you mean :confused: I talk about the UI area employed by the user to draw circuits etc. 
Sorry, I am not very familiar with the qucs-terminology (yet).

PS: Is this still considered on-topic or should we move this discussion elsewhere?

guitorri, 2016-10-19:
@ldpgh indeed an EDA environment with scripting capability is very powerful. Another nice example relevant here is KLayout with Ruby an Python interfaces. However the author seem to have rolled out a custom wrapper generator, no SWIG, no SIP, no PySide. IMHO, to get to that level, it would make sense to converge to a form of database, split things into namespaces and set up an API to enable any sort of interface. This is definitely in the wish list, but as you said it would be part of a big refactor. 

@fer-rum I am not so sure about QML. Frankly I never tried it. I did read some [comments] (http://lists.qt-project.org/pipermail/interest/2015-June/017412.html) and watched the related [video](https://www.youtube.com/watch?v=6vhIq3Nxmz0). The above developer does paint a very pretty picture about QML in general, specially for desktop applications. Again QML would be require a big refactor.

What I had in mind here is a first mid-sized refactor, just enough to get rid of Qt3Support and continue from there.

ldpgh, 2016-10-19:
Just FYI ... to be deleted next week
... fer-rum & ldpgh had an open discussion last week at http://b-liebig.de
... topics were QML, script interface, structural topics, what next, question by Felix
Due to the nature of the location, there are no meeting notes :-o

Btw. .... have send an email to the Klayout contact adress.

alysogorov, 2016-11-06:
Imo, Qt Graphics View Framework looks as good choice for schematics editor. it has everything, people need, to develop such tools. Else you'll have to develop something similar, to implement this editor.

nvdl, 2016-11-06:
Any thought about PySide (PyQt) as user interface of Qucs?
The modules that have to do the main job (processing, simulation) act as plugins.

alysogorov, 2016-11-06:
what is plugin? python script, or just a dynamic library, written for example in C++?

felix-salfelder, 2016-11-06:
> If someone wants to add a new type of plot, they write a new script (or class).

that sounds good. if we manage to turn the current plot "items" into plugins, we will have figured out the interface for that. i have started this, in the hope for a 3dplot replacement... right now, components take prececence, as they are required to resolve the non-free debacle. (they are missing in your list @ndvl), 

> python script, or [..] a dynamic library

in python, these are interchangeable. theres some overhead of course, but you will have to figure that out only once.

guitorri, 2016-11-17:
I was curious to see what would it take to render the `Graph`s used in the `Diagram`s. 
I took a hammer and these are the results so far:
 - qucs-0.0.19-rc2
[image]
 - https://github.com/guitorri/qucs/tree/QGraphicView-exploration
[image]

nvdl, 2016-11-18:
This looks nice. I can help with mouse interactions and other stuff.
Before we proceed, it is good to layout the classes that interact and their links.

Importantly, we have to start from scratch and can include code from current base if needed.
But, cleanly formatted code to start with; no more organic growth.
Doxygen comments for each function etc.

These are the classes we need as I see it (add more as needed):

Painter class (with Qt front end only now)
Editor class (change names, labels, positions etc. add/remove components)
User input class (track user input; mouse and keyboard)
Shortcuts class (connect to "User input class") and other classes to invoke actions
Selector class (use data from "User input class") to select components
Netlister class
Librarian class (good sharing option; download model directly from URL; security?)

And most importantly the Component class.

Each component is represented by a SVG file with a Python file.
The Python file contains the locations and names of the ports in the graphic image along with the netlist and other specifics for different simulators. It also points to the model(s) files if that is applicable.
The scripted nature of this file means that it can be modified for future changes to adopt to new base classes if needed. No more compilation just because a new component arrives.

A subcircuit is also a component with internal netlist and exported ports.

Netlister class queries this script to create the netlist.
Painter class queries this script to draw the symbol.

Addition of new component means addition of the aforementioned files to the "components" folder and refreshing the "Librarian class".

Others can chime in so that we do it correctly this time.

felix-salfelder, 2016-11-18:
> And most importantly the Component class.

yes. please help with the refactoring and review. i would really like to finish #589 this year.

> Each component is represented by a SVG file with a Python file.

this is completely arbitrary and should be done in optional pluggable code. go ahead. implement that. we will soon be able to load it.

> do it correctly this time.

yes. arbitrary choices must go to plugins, to they can be easily replaced. e.g. what if in 10 years somebody wants to use julia instead of python? that *must be made possible*. do we agree on this?

nvdl, 2016-11-18:
Any starter on how you have done it so far?
What is already done and what needs to be done?
Classes and their interfaces?

I can get started with the "Component" part.
With "Editor" class we can move them around in Qt5.
GUI stuff in the beginning only; graphics and ports and how mouse hover detects those.
R and C symbols are already done!!!

We should keep the GUI drawing stuff and mouse/keyboard interaction stuff as much localized as possible for future plug-ability of any other GUI renderer and input interface layer.

To switch from Python to "whatever", we can write a translator later.
That should not be an issue. First goal is to encompass all the functions within a Python class that are now entangled in the main source tree.

felix-salfelder, 2016-11-18:
> What is already done and what needs to be done?

loadable modules and the dispatcher are in place. i am currently untangling the symbol list on the left side of the screen, it must be possible for a component to register to that list. currently the list is compiled in.

the roadmap would be to start with a few components, handle them properly, make everything else work. and then convert all the others.

be aware that the components are massively entangled with "commands" and "diagrams", maybe more. i would like to touch them as little as possible... (fixing these could be scheduled for 0.0.21, maybe).

guitorri, 2016-11-18:
You're losing me guys. Are we ok with Qt Graphics View or not?

@nvdl, I might be wrong but I don't think starting from scratch is a good idea. In order to keep things testable and compatible with existing QUCS projects I would rather do multiple refactoring passes instead of starting afresh. I am not a software developer by training, but I read in most places that refactoring is usually preferred instead of a big rewrite.

Frankly, switching to Python/PySide/PyQt (which I know well and like) or QML (which I don't know and dislike) at this stage is out of question, at least for me. I don't have the bandwidth to help with that.

The following is what I mean by refactoring passes, very specific and more or less in this order:
- remove Qt3 compatibility, **with as little code changes as possible**.  Mainly related to the exploration linked to this issue and a few remaining Q3PtrList. Document methods as we go. 
- separate things into namespaces. The new loader, dispatcher, netlisters, plugins, multi-simulator,..., should easily fit in somewhere.
- introduce unit tests where possible
- move/rename code around as necessary, push/pull methods across classes as needed, create/rename classes as reasonable
- decouple the schematic storage/database from graphics
- refactor database to handle hierarchy and libraries
- turn database code into shared library with an API (expose to wrapper SWIG or the like)
- turn graphics code into shared library with an API (expose to PySide or PyQt)
- deprecate Qucs proprietary file formats in favor of something more expressive and standard. Provide ways to port existing projects forward.
- I have a longer list somewhere... I should put it into the wiki for future reference.

Let's say it takes 3 volunteers 1 months per pass... who is in for this?

Alternatively we could look into [Caneda](https://github.com/Caneda/Caneda) which forked from QUCS. Maybe force into it compatibility for existing QUCS projects and call it a day.

nvdl, 2016-11-21:
We go with Qt5 graphics but there should only be one class that handles this. All drawing requests go through it. For later updates/changes in GUI front-end, only this class needs to be modified/rewritten.

If starting from scratch is not a good idea, code formatting should be allowed. If I reformat a piece of code to make sense of it, it creates a review issue. That forces me to format the code (untangle the spaghetti), find the issue, modify the lines and tangle the code again before creating a PR.

Just give "mouseactions.cpp" a look, it is like chasing function pointers all the time with hacks when class calls do not work (though they were designed to be generic).
When the class calls do not work, there are more hacks on top of that plus the legacy code that does not do anything now or make drawing worse.

This is why wire label is handled in a different way because its own implementation is broken.
Moreover, we are accepting PRs on top of that code base before a reformat/rewrite.
I gave the latest PR a look and the person commented out legacy code still unsure if it is to be used or not.

As an example:
https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/mouseactions.cpp#L818
I am not a formatting Nazi here but it makes difficult to follow code flow and find an issue.

I am in for the graphics and can take some stuff from one year old PR plus the current code base provided that we agree on code formatting and cleanup (where needed). This is going to be a major rewrite especially in some parts.

> I have a longer list somewhere... I should put it into the wiki for future reference.

Yes a Wiki for this.

> Alternatively we could look into Caneda which forked from QUCS.

It looks better. At least they have code documentation and are using XML!
Any historical reason for the fork?

> and call it a day.

I don't think we are getting short of time or there is a deadline to meet. This is a part of Qucs that needs an ongoing involvement to make it work firstly and add more functions to it secondly. We have many good simulators in the arena, what we need is a good schematic capture and plotter software with flexible components extensibility.

nvdl, 2016-11-21:
Moreover, to give life to the scripting side, we need another branch; something like `devel-scripting`.

https://github.com/ClemensFMN/QucsPython
https://github.com/ClemensFMN/QucsPython

From discussion: https://github.com/Qucs/qucs/issues/263

ldpgh, 2016-11-21:
@guitorri ... regarding QGraphicsView ... please check the drawing performance by increasing pts=50 to pts=50000 (50 thousand).

Though it's more a TRAN-simulation issue ... using complex numbers would be an interesting add-on stress.

If I remember right ... my experience (but can't find my testcase, @ 2010, performance dropped significantly) was a slow default screen raster algorithm in Qt. The default setting has drawn each point of the data vector while using drawPolyline regardless the pixel was already set by the previous line segment in the data vector or not. 

With about 50000 points in the data vector and a screen of 2000x2000pixel just 2000 line segments (4%) has to be drawn. 
I chopped the data vector into single Line segments. Line segments are only drawn if the distance to the previous point is 1 pixel (regardless x or y direction. It has improved the speed by 8x.

Btw. It's so

guitorri, 2016-11-21:
> We go with Qt5 graphics but there should only be one class that handles
> this. All drawing requests go through it. For later updates/changes in GUI
> front-end, only this class needs to be modified/rewritten.
I am not sure I follow. Each of the QGraphicsItems must draw themselves.
That is what make it fast.
The children of `Element` must all handle its drawing an interactions.
See 7f5a3324edbc8
for reference. Later I roughly implement the paint and bounding methods.

> If starting from scratch is not a good idea, code formatting should be
> allowed. If I reformat a piece of code to make sense of it, it creates a
> review issue. That forces me to format the code (untangle the spaghetti),
> find the issue, modify the lines and tangle the code again before creating
> a PR.
Which format? I don't want start discussing that. Can't we just get the job
done and later take care of the code format? Clearly, chunks that are moved
around and to new files you can format it the way you like, at the risk of
being re-formated again once we decide about a coding standard.

> Just give "mouseactions.cpp" a look, it is like chasing function pointers
> all the time with hacks when class calls do not work (though they were
> designed to be generic).
> When the class calls do not [...]
Yes, `mouseactions.cpp` must be redone, eventually even removed. See one
pass in 5bbb7ccd4f.

> This is why wire label is handled in a different way because its own
> implementation is broken.
> Moreover, we are accepting PRs on top of that code base before a
> reformat/rewrite.
> I gave the latest PR a look and the person commented out legacy code still
> unsure if it is to be used or not.

We cannot stop the world and fix it all. Yes, might be a waste to accept
PRs that are going to be deprecated. Please don't be confused with code
format and rewrite, these are different things. The legacy code is
sometimes difficult to figure out, the developers are no longer around to
give us a hint of their intentions... But if someone indadvertedly put time
to fix the issue on the legacy code, what do we do? Reject because we are
planing to do a rewrite?

> As an example:
> https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/m
> ouseactions.cpp#L818
> I am not a formatting Nazi here but it makes difficult to follow code flow
> and find an issue.
Yes, see 5bbb7ccd4f as a first step. Later commits might have moved more
code around.

Note my branch was really an exploration, it is very dirty, and it is on
purpose. I wanted to cover more ground instead of stopping at each tiny
detail.

> I am in for the graphics and can take some stuff from one year old PR plus
> the current code base provided that we agree on code formatting and cleanup
> (where needed). This is going to be a major rewrite especially in some
> parts.
I don't follow what do you mean by old PR. Yes some parts will need a
rewrite. But be mindfull of what the QGraphics framework already offer so
we don't reinvent the wheel.

[snip]

> Alternatively we could look into Caneda which forked from QUCS.
>
> It looks better. At least they have code documentation and are using XML!
> Any historical reason for the fork?
They started by trying to fix the graphics while porting to Qt4. Then
overhauled everything, including the file format. They lost compatibility
with legacy Qucs projects and that is what I know about it.
Some comments by Gopala Krishna, one person initially involved in the fork,
can be found here:
https://sourceforge.net/p/qucs/mailman/message/33224830/

We cannot afford to break users projects.

> and call it a day.
>
> I don't think we are getting short of time or there is a deadline to meet.
> This is a part of Qucs that needs an ongoing involvement to make it work
> firstly and add more functions [...]

Right, take your time. Read my commits for some overview and let us
continue as we can. When I have some time I will squash related commits and
cherry-pick base changes into develop (like 9e07d9ffb), so we don't diverge
too much from mainline.

guitorri, 2016-11-21:
Off-topic. Open another issue and define what do you mean by scripting.
I guess you want more than just modifying schematic/netlist files and rerun
the tools.
To properly embed scripting capability into the inner-workings of
Qucs/Qucsator will take awhile.

guitorri, 2016-11-21:
@ldpgh yes I can try to check how the graphs are being drawn. Thank you for the hint about drawing points, polylines and lines. We certainly need to investigate the performance hit.

The slowdown might be related with the still ongoing port. At some stage fransschreuder had to serialize the drawings to make the Qt3Support work at all. 

Talking about graphics and plots. We could also think about using Qwt. However (last time I checked) Qwt uses QWidget as canvas and I really don't know how easy would it be to replicate the diagrams the way we do, in the schematic area, with it. Adding a QWidged to a QGraphicsView/Scene is certainly possible. Thou the examples I saw didn't look very good, but I didn't look hard enough.

felix-salfelder, 2020-09-26:
This discussion touches most aspects of QGraphicsView and refactoring. Mainly there does not seem to be an alternative to QGraphicsView, as long as Qt is involved. (also, I have tried, and it does the job pretty nicely).

Some problems are still open, but please open a new issue as needed.

closed 2020-09-26 by felix-salfelder