Qucs/qucs #564: spice4qucs: DC analysis not found in netlist

ldpgh, 2016-10-05:
Hello, it seems the DC analysis is not exported to Ngspice (no .DC in the netlist seen)

If using Qucsator it's working as expected and saveOp=yes provides all required information

Win7Pro : spice4qucs 0.0.19

PS. What's the recommended bug-tracker ... sourceforge seems to have low run rate

ldpgh, 2016-10-05:
AC analysis works as expected

ra3xdh, 2016-10-06:
`DC analysis` in Qucs is approximate equivalent of SPICE `.OP` analysis. OP is performed automatically when TRAN or AC is executed and .DC don't require to be explicitly added into the SPICE netlist. Also you can use "Show bias" (press F8 key)  to see operation point. SPICE `.DC` is emulated with Qucs DC+ParSweep. I haven't previously used `saveOp` option to look at DC bias. I will consider later how to add putting OP into dataset feature. 

You may use the following two places for bugtracker for Qucs-S:
1. The following spice4qucs discussion thread #77 
2. My home repository https://github.com/ra3xdh/qucs You can open issues related to SPICE supports here. Don't use it to report bugs of mainline Qucs!

I think main Qucs repository should not be used for posting issues on `spice4qucs` (except #77 and #181 ), because it can lead to mixing up of mainline Qucs and Qucs-S bugs.

ldpgh, 2016-10-06:
Regarding 1.
- With all those very intersting informations (w.g Xyce) in [#77](https://github.com/Qucs/qucs/issues/77) I did not mixed it with my small issues.
- What about adding a Wiki page for those nice informations.

Regarding 2.
- There is no "issue" Tab on this page ... that's the main bottleneck for me.
- 1st tried http://qucs.sourceforge.net bug-tracker, but got the impression all folks are running GitHub.

Agree with you keeping spice4qucs separated from main-line.

ra3xdh, 2016-10-06:
I have enabled issues at my home repository. See https://github.com/ra3xdh/qucs/issues

ra3xdh, 2016-10-30:
Transferred to my home repository as related to `spice4qucs` subsystem only. See https://github.com/ra3xdh/qucs/issues/10 to continue discussion.

closed 2016-10-30 by ra3xdh