Qucs/qucs #570: Qucs Build Fails creating modelist.lo

AndyKirby, 2016-10-19:
Guys

The build is currently failing to compile on Debian unstable:-

4.7.0-1-amd64 #1 SMP Debian 4.7.6-1 (2016-10-07) x86_64 GNU/Linux

ADMS built and installed OK, but I had to run ldconfig manualy as root before qucs configure would complete sucessfuly.

Thereafter the qucs -core build using make fails on nodelist.cpp

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/components -I../src/interface -I.   -O2 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT nodelist.lo -MD -MP -MF .deps/nodelist.Tpo -c -o nodelist.lo nodelist.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/components -I../src/interface -I. -O2 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT nodelist.lo -MD -MP -MF .deps/nodelist.Tpo -c nodelist.cpp  -fPIC -DPIC -o .libs/nodelist.o
In file included from nodelist.cpp:32:0:
logging.h:37:8: error: ‘FILE’ does not name a type
 extern FILE * file_status;
        ^~~~
logging.h:38:8: error: ‘FILE’ does not name a type
 extern FILE * file_error;
        ^~~~
Makefile:860: recipe for target 'nodelist.lo' failed
make[5]: *** [nodelist.lo] Error 1
make[5]: Leaving directory '/home/andy/Apps/qucs/qucs/qucs-core/src'
Makefile:906: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1

in3otd, 2016-10-22:
this sounds like the issue fixed in #536 ; did you pull the `release-0.0.19` branch? The `master` branch is not up to date at present, it will contain the release 0.0.19 code once this is released

AndyKirby, 2016-10-25:
Thanks in3otd.

That worked fine.I can confirm that the qucs 0.0.19 branch compiled, 
installed and looks to be working fine on Debian unstable.

I did have a good google around and a skim through the issues before 
pestering you guys, but obviously missed that one.

Might be worth putting a note to this effect on the readme for qucs in 
github at least in the interim. Just to save folk like me that arrive to 
install qucs and don't have the context.

Are there any plans to update the versions etc in the debian repo's ?? 
These do feed into a lot of other distributions, so are likely to be 
worth the effort.

Cheers

Kirbs

ra3xdh, 2016-10-30:
@AndyKirby , You can read about plans for Qucs inclusion in Debian repos here https://github.com/Qucs/qucs/pull/454

closed 2016-10-30 by ra3xdh

AndyKirby, 2016-10-30:
Thanks Vadim