Qucs/qucs #571: vcresistor.analogfunction.cpp Error -1073741515 ... qtcreator312, qt486, cmake & cmake362

ldpgh, 2016-10-27:
File "qucs-core/src/components/verilog/vcresistor.analogfunction.cpp" does not exist.
error code looks strange to me.
Source has been cloned from Github 1hour ago into fresh directory

Building Qucs using Cmake within QtCreator creates an error at 61% progress.
It's not clear to me whether it's something in my setup (most likely) or a real issue?

I tried to to bypass the Adms topic, but was not successfull. To get the Adm-executable built I was using my cygwin setup (because it compiled it already in the past) and gave the path to cmake.

OS: Windows7
build_environment: qtcreator312, qt486, cmake362
amds: admsXml build with cygwin (!)

Tail of the log-message including the error message
[ 61%] Built target gperfappgen
Scanning dependencies of target equation
[ 61%] Generating gperfapphash.gph
[ 61%] Generating gperfapphash.cpp
[ 61%] Built target equation
[ 61%] Generating vcresistor.analogfunction.cpp, vcresistor.analogfunction.h, vcresistor.cpp, vcresistor.core.cpp, vcresistor.core.h, vcresistor.defs.h
qucs-core\src\components\verilog\CMakeFiles\coreVerilog.dir\build.make:915: recipe for target 'qucs-core/src/components/verilog/vcresistor.analogfunction.cpp' failed
CMakeFiles\Makefile2:1798: recipe for target 'qucs-core/src/components/verilog/CMakeFiles/coreVerilog.dir/all' failed
mingw32-make[2]: *** [qucs-core/src/components/verilog/vcresistor.analogfunction.cpp] Error -1073741515
Makefile:148: recipe for target 'all' failed
mingw32-make[1]: *** [qucs-core/src/components/verilog/CMakeFiles/coreVerilog.dir/all] Error 2
mingw32-make: *** [all] Error 2
15:54:10: Der Prozess "C:\ProgrammeLutz\Qt\Qt4.8.6\Tools\mingw482\bin\mingw32-make.exe" wurde mit dem Rückgabewert 2 beendet.
Fehler beim Erstellen/Deployment des Projekts qucs-project (Kit: Qt 4.8.6)
Bei der Ausführung von Schritt 'Make'
15:54:10: Verstrichene Zeit: 06:28.

ra3xdh, 2016-10-27:
`qucs-core` building with CMake seems to be broken on all platforms. But `qucs-gui` could be successfully built with CMake. The simplest way for Windows is usage of MSYS2+autotools. It seems there is no exact build instruction. Look into https://github.com/Qucs/qucs/blob/master/.appveyor.yml and you can extract build commands form here. 

If you are considering taking part in Qucs development, consider migration of development subsystem to Linux platform, because GNU tools for Windows may be buggy and poor documented.

in3otd, 2016-10-27:
The build instructions for Windows with MSYS2+autotools on the [wiki](https://github.com/Qucs/qucs/wiki/Build-Windows) should be up to date... if these need to be adjusted, please help updating them :grin:

ldpgh, 2016-10-27:
Getting Qucs-Gui built is my goal at the moment.
QtCreator with graphical GDB-wrapper as cross-platform tool is what I like to have on both platforms.

Any hint how to jump over the hurdle to keep the focus on Qucs-Gui exclusively?
Where the missing file comes from? Could my AdmsXml.exe@cygwin be the issue?
What's the recommended Linux-Distro in case of Qucs?

@in3otd 
I was able to build Qucs (-g) using MSYS2+autotools according to the wiki. The gdb-commandline use was just to debug a crash.
Btw. it's more an use-case issue. After `make install` Qucs need to be started from the `mingw64.exe` window. Looks like a DLL topic ... an appropriate PATH setting may fix it. But it was just a test compile.

ra3xdh, 2016-10-27:
You don't need ADMS to build Qucs-GUI only. You can even open
qucs/qucs/CMakeLists.txt as a project in QtCreator and it will build Qucs
without qucsator. It works for Linux and should work for Windows, if tools
are configured properly. You can use any Linux distribution with Qt4 and
C++ development kit. For example, Debian will be good choice. Other
distributions also have no problems.

in3otd, 2016-10-27:
@ldpgh FYI, we do copy some DLLs to build binary user packages (see the AppVeyor script [here](https://github.com/Qucs/qucs/blob/master/.appveyor.yml#L57-L70)) and these may also be needed to start Qucs outside of the shell, if you do not have them in the path.

ldpgh, 2016-10-27:
Have Qucs-core disabled (changed to comment) ... rebuild cmake and compile ... and gdb stops at the desired location :-)

ra3xdh: "qucs-core building with CMake seems to be broken on all platforms." ... this means to me qucs-core is treated in a different issue.

Close this issue with my focus just on Qucs-gui.

closed 2016-10-27 by ldpgh

guitorri, 2016-10-28:
For the record, CMake for qucs-core is working just fine for OSX 10.9.5 (tried `release-0.0.19` branch just now).

@ldpgh The error seems to be related to admsXml. C++ files are not being generated for the Verilog-A modules. Indeed please open another issue if want to further build the core.

@ra3xdh please open an issue if you find CMake to be broken in another platforms.

ra3xdh, 2016-10-29:
@guitorri , I have tested CMake just now. Qucsator was successfully compiled. It seems CMake is operational. Previously I have tried to build Qucsator with CMake last year and there were problems.

guitorri, 2016-10-29:
Ok. Thanks for the update.