Qucs/qucs #572: Rectangle Voltage source not handling transient step size correctly

AdvancedRFLLC, 2016-10-28:
In a transient simulation, the rectangle voltage does not appear to properly handle step sizes.  In particular, with InitialStep and MaxStep equal, the timing is highly dependent upon the MinStep setting, and does not correlate with the times set in the source.  In fact, the timing is also dependent upon the number of points and the stop time.  

With smaller MinStep values (i.e. InitialStep=MaxStep=1e-6, MinStep=1e-7), the source output appears unstable, oscillating between the output states.  

Setting InitialStep=MaxStep=MinStep is potentially useful for simulating discrete-time systems, and it would be very helpful if it worked correctly.  The unpredictability of the Rectangle Voltage source is a bit frustrating.   

(As a side note, it looks like the ideal Delay element has a few quirks as well when MinStep is set to 1e-7.)

[Pulse_Test.zip](https://github.com/Qucs/qucs/files/558581/Pulse_Test.zip)

AdvancedRFLLC, 2016-10-28:
Issue applies to Qucs 0.0.19, operating on Windows 7 Enterprise

andresmmera, 2016-10-28:
Hello,

I'm not the best one to help with transient simulation issues, but, as a
rule of thumb, I usually set MinStep < 1/(1000*f) where 'f' is the maximum
frequency I'm interested on. Concerning your case, I have set MinStep =
1e-9 and it works as expected on my side.

https://dl.dropboxusercontent.com/u/21294824/Selection_011.png

Please notice that the transient simulator is not as good as it should be.
In case you feel disappointed with it, I suggest using spice4qucs with
Ngspice

https://github.com/ra3xdh/qucs/tree/spice4qucs
https://qucs-help.readthedocs.io/en/spice4qucs/Intro.html

Regards,

Andrés

in3otd, 2016-10-28:
currently the transient simulator is not very good at handling (sources with) fast transients but his may not be the main problem here.

The parameters `InitialStep`, `MaxStep` and `MinStep` may not be what you expect them to be. These parameters control the transient simulator internal timesteps, not the print time step, which is set by the transient simulation `Step` (unless you set `RelaxTSR` to `yes`, but that's another story).
If you would like to have an output point every microsecond, set the transient simulation `Step` accordingly and let the simulator choose its internal timestep (use the default values)

Anyway there is surely a bug, since the simulator does not handle well a case like this, where it is asked to print results more often than they are computed:  here the print step is set to a value (150 ns) less than the simulation timestep imposed by `MaxStep` and `MinStep` (1 us).
The problem you see is not related to the rectangle voltage source but to the way the print and simulation timestep interact.

AdvancedRFLLC, 2016-10-29:
Apparently, the main bug was on me.  Looks like I messed up when setting the number of points, which of course sets the (print) step size - that was intended to be 151 points from 0 to 150 usec,  leading to a step size of 1 usec., not 150 nsec.  (It must have been Friday...)  With that value, the generator seems to work pretty much as expected.   I'll check it out thoroughly next week in the complete circuit when I get back into the office.

A copy of the corrected file is attached, with the step size parameters correctly calculated in the equation block.  

Fyi, I'm setting all of the step size parameters equal in an attempt to ensure calculation only once per time step, enabling simulation of discrete-time systems.

Thanks for the feedback!

[Pulse_Test.sch.zip](https://github.com/Qucs/qucs/files/559949/Pulse_Test.sch.zip)

closed 2017-10-10 by guitorri