Qucs/qucs #578: Export as image, timing diagram text shifted up.

guitorri, 2016-11-02:
Spotted this while testing #523.

See exported image below. Note the text shifted up in the timing diagram :
[image]

The project is here: [05-verilog_counter_prj.zip](https://github.com/Qucs/qucs/files/566768/05-verilog_counter_prj.zip)

in3otd, 2016-11-03:
uhm, this seems to have appeared with c9fcce60b1ddcefcc425a6c909ef6914ccb02a98 but I don't understand why...

Another small printing issue is that when printing from the command line the text painting (in blue above) is cut (but not when printing from the GUI)

ra3xdh, 2016-11-04:
There may exist not obvious connection between Marker and Timing diagrams. But I don't understand how they may be connected too. I will try to dig into legacy code in the next few days.

guitorri, 2016-11-04:
in3otd have you checked before and after c9fcce6. I was pretty sure I did not include it when I reported. I will check again.

in3otd, 2016-11-04:
yes, I've checked before and after, did `make distclean` in between but I cannot exclude a mistake :grin:

in3otd, 2016-11-06:
ok, it's clearer now: we are [now](https://github.com/Qucs/qucs/commit/c9fcce60b1ddcefcc425a6c909ef6914ccb02a98#diff-12763b2bbd1d04ec7aa8054d6f637f86R710)  calling `paintDiagram()` to paint every diagram while we should actually call `paint()`, as done before.
`paintDiagram()` is the method of the `Diagram()` base class, while the timing diagram is a `TabDiagram()` (a child of `Diagram()`) and we need to use its own `paint()` method (which also prints the Tab Diagram scrollbar, which is now missing) to have it printed correctly.
Something needs to be modified to have the marker print stuff working as expected now...

closed 2016-11-11 by guitorri