Qucs/qucs #582: Compilation Error (master)

nvdl, 2016-11-04:
https://github.com/Qucs/qucs/blob/master/qucs-core/src/converter/check_spice.cpp#L682

Need to change to:
for (entry = spice_definition_available; entry->type != NULL; entry++)

guitorri, 2016-11-04:
Can you try `release-0.0.19` branch? it is going to become `master` and `develop` in a couple of days. I am wrapping up the release.

guitorri, 2016-11-04:
I should have said, going to be merged....

nvdl, 2016-11-04:
Same issue with release-0.0.19 branch:

Debian clang version 3.5.0-10 (tags/RELEASE_350/final) (based on LLVM 3.5.0)
Target: x86_64-pc-linux-gnu
Thread model: posix

clang++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/components -I../src/interface -I.   -O2 -std=gnu++11 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__ -MT ucs.o -MD -MP -MF .deps/ucs.Tpo -c -o ucs.o ucs.cpp
clang: warning: -ldl: 'linker' input unused
mv -f .deps/ucs.Tpo .deps/ucs.Po
/bin/bash ../libtool  --tag=CXX   --mode=link clang++  -O2 -std=gnu++11 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__ -Wl,-rpath,/usr/local/lib  -o qucsator ucs.o libqucs.la -lm 
libtool: link: clang++ -O2 -std=gnu++11 -pipe -fno-exceptions -D__STRICT_ANSI__ -Wl,-rpath -Wl,/usr/local/lib -o .libs/qucsator ucs.o  -ldl ./.libs/libqucs.so -lm
./qucsator -l > qucsdefs.h
/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src/.libs/lt-qucsator: symbol lookup error: /home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src/.libs/lt-qucsator: undefined symbol: _ZN4qucs6module5printEv
Makefile:1216: recipe for target 'qucsdefs.h' failed
make[5]: *** [qucsdefs.h] Error 127
make[5]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src'
Makefile:896: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src'
Makefile:587: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src'
Makefile:762: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core'
Makefile:648: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core'
Makefile:370: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

clang: warning: -ldl: 'linker' input unused
check_spice.cpp:688:16: error: use of undeclared identifier 'qucs_definition_available'; did you mean 'spice_definition_available'?
  for (entry = qucs_definition_available; entry->type != NULL; entry++)
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               spice_definition_available
check_spice.cpp:96:17: note: 'spice_definition_available' declared here
struct define_t spice_definition_available[] =
                ^
1 error generated.
Makefile:524: recipe for target 'check_spice.o' failed
make[5]: *** [check_spice.o] Error 1
make[5]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src/converter'
Makefile:896: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src'
Makefile:587: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core/src'
Makefile:762: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core'
Makefile:648: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-core'
Makefile:370: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

nvdl, 2016-11-04:
When Octave is no installed, "no --path" is evaluated.

Making all in qucs-doc
make[1]: Entering directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-doc'
Making all in technical
make[2]: Entering directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-doc/technical'
no --path . mszl.m
/bin/bash: no: command not found
Makefile:606: recipe for target 'mszl.eps' failed
make[2]: *** [mszl.eps] Error 127
make[2]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-doc/technical'
Makefile:373: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-doc'
Makefile:370: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

guitorri, 2016-11-04:
I saw this octave error yesterday. Should turn it into an error. Related to #160.

felix-salfelder, 2016-11-04:
> When Octave is no installed, "no --path" is evaluated.

configure.ac is incomplete. use `--disable-doc` (at top level) for now.

nvdl, 2016-11-04:
OK for docs.

for (entry = spice_definition_available; entry->type != NULL; entry++)

is still needed. Cannot find "qucs_definition_available" in the source.

guitorri, 2016-11-04:
It seems like an error, if so I don't know why it was not caught before...

The `qucs_definition_available` is made available by `qucsator` to bootstrap `qucsconv`.
https://github.com/Qucs/qucs/blob/release-0.0.19/qucs-core/src/module.cpp#L344
https://github.com/Qucs/qucs/blob/release-0.0.19/qucs-core/src/Makefile.am#L124
The `qucsdefs.h` should contain the  `qucs_definition_available`

Either you build is not finding  `qucsdefs.h` of there is something really wrong going on and it should be `spice_definition_available`.

I am away of my development box. Cannot check it now.

nvdl, 2016-11-04:
If I start with core:

`cd qucs-core`
`./bootstrap`
`./configure`
`make`

check_spice.cpp:688:16: error: use of undeclared identifier 'qucs_definition_available'; did you mean 'spice_definition_available'?
  for (entry = qucs_definition_available; entry->type != NULL; entry++)
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               spice_definition_available
check_spice.cpp:96:17: note: 'spice_definition_available' declared here
struct define_t spice_definition_available[] =
                ^
1 error generated.
Makefile:524: recipe for target 'check_spice.o' failed

Is this data type generated by "bootstrap" or "configure" or is it somewhere in the source?

guitorri, 2016-11-04:
Sorry for the confusion, it is not related to the `bootstrap` script. 

It strings in `module.cpp` are written to a file during `make`. First `qucsator` gets build then it is run to get its definitions to a file:
`./qucsator -l > qucsdef.h`  (see [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs-core/src/Makefile.am#L124))

Than the `qucsdefs.h` file is used by `qucsconv`.

Have you tried cleaning up your working directory start over? 
Something like `git clean -xdf` (caution, will remove any non tracked files).

nvdl, 2016-11-04:
The creation of the header via `qucsator` is failing.

./qucsator -l > qucsdefs.h
/home/nvd/software/simulators/qucs/qucs-nvdl/qucs-core/src/.libs/lt-qucsator: symbol lookup error: /home/nvd/software/simulators/qucs/qucs-nvdl/qucs-core/src/.libs/lt-qucsator: undefined symbol: _ZN4qucs6module5printEv
Makefile:1216: recipe for target 'qucsdefs.h' failed
make[5]: *** [qucsdefs.h] Error 127

Am digging more.

in3otd, 2016-11-04:
did you already install qucs before in the same target directory? I've seen similar strange issues which were resolved basically by uninstalling qucs or removing just the installed qucs libraries. I had the impression that sometimes libtool gets confused for some reason and tries to use old installed libraries with the newly compiled code.

felix-salfelder, 2016-11-04:
a common reason for that is the use of `-l<libname>` during the build? (it should be LDADD instead, and point directly to the .la file...)
at least when using `.._LDADD=some/library.la` it's no longer possible to accidentally link to system libraries.

nvdl, 2016-11-04:
git log

Take the latest commit.

git reset --hard 2c94ca0a67dd7c90ed3c37f08fc6b25e4e5148cf
HEAD is now at 2c94ca0 Merge pull request #579 from guitorri/update-news

git clean -xdf

./bootstrap 
./configure --disable-doc

make

Same error. Is it caused by disabling the document generation?

guitorri, 2016-11-04:
`qucsator` cannot find find the correct symbols on the shared library.

Do you have any `libqucs` in `/usr/local/lib`?

It might be trying to use an older version of the library.

nvdl, 2016-11-04:
That was it. Old libraries. I moved them away from `/usr/local/lib`
Can the order of library linking be prioritized?

What if someone is trying to upgrade and follow these steps?
Can `qucsator` be invoked through Makefile to use to newly created library?

in3otd, 2016-11-04:
I've seen this issue sometimes, it comes and goes randomly. I'll say it happens less than 1 % of the times here, most often after checking out and trying several different commits.

Note in your command line above you have

`libtool: link: clang++ -O2 -std=gnu++11 -pipe -fno-exceptions -D__STRICT_ANSI__ -Wl,`**-rpath** `-Wl,`**/usr/local/lib**` -o .libs/qucsator ucs.o -ldl ./.libs/libqucs.so -lm`

Why there is `-rpath /usr/local/lib` ? It should point to the local library and I think it usually does, when things go as expected.

nvdl, 2016-11-04:
No idea about `-rpath` and why it is set that way.

After `sudo make install` from the last time, the build went well this time.
Looks like it comes and goes based the difference in the installed library vs newly created one.
If the installed one suffices, things go unnoticed.

I do not see any reference to `qucs-core/src/.libs` in the `configure` scripts.

qucs-core/configure:10381:  sys_lib_search_path_spec="/lib /usr/lib /usr/local/lib"
qucs-core/configure:10485:  sys_lib_search_path_spec="/shlib /usr/lib /usr/X11/lib /usr/contrib/lib /lib /usr/local/lib"
qucs-core/configure:10486:  sys_lib_dlsearch_path_spec="/shlib /usr/lib /usr/local/lib"
qucs-core/configure:10613:  sys_lib_search_path_spec="$sys_lib_search_path_spec /usr/local/lib"
qucs-core/configure:10614:  sys_lib_dlsearch_path_spec='/usr/local/lib /lib /usr/lib'
qucs-core/configure:10699:      sys_lib_search_path_spec="/usr/lib/hpux32 /usr/local/lib/hpux32 /usr/local/lib"
qucs-core/configure:10701:      sys_lib_search_path_spec="/usr/lib/hpux64 /usr/local/lib/hpux64"
qucs-core/configure:10775:  sys_lib_search_path_spec="/usr/lib${libsuff} /lib${libsuff} /usr/local/lib${libsuff}"
qucs-core/configure:10941:  sys_lib_search_path_spec="/usr/shlib /usr/ccs/lib /usr/lib/cmplrs/cc /usr/lib /usr/local/lib /var/shlib"
qucs-core/configure:11016:    sys_lib_search_path_spec='/usr/local/lib /usr/gnu/lib /usr/ccs/lib /usr/lib /lib'
qucs-core/configure:15374:  sys_lib_search_path_spec="/shlib /usr/lib /usr/X11/lib /usr/contrib/lib /lib /usr/local/lib"
qucs-core/configure:15375:  sys_lib_dlsearch_path_spec="/shlib /usr/lib /usr/local/lib"
qucs-core/configure:15501:  sys_lib_dlsearch_path_spec='/usr/local/lib /lib /usr/lib'
qucs-core/configure:15586:      sys_lib_search_path_spec="/usr/lib/hpux32 /usr/local/lib/hpux32 /usr/local/lib"
qucs-core/configure:15588:      sys_lib_search_path_spec="/usr/lib/hpux64 /usr/local/lib/hpux64"
qucs-core/configure:15662:  sys_lib_search_path_spec="/usr/lib${libsuff} /lib${libsuff} /usr/local/lib${libsuff}"
qucs-core/configure:15828:  sys_lib_search_path_spec="/usr/shlib /usr/ccs/lib /usr/lib/cmplrs/cc /usr/lib /usr/local/lib /var/shlib"
qucs-core/configure:15903:    sys_lib_search_path_spec='/usr/local/lib /usr/gnu/lib /usr/ccs/lib /usr/lib /lib'
qucs-core/libtool:281:sys_lib_dlsearch_path_spec="/lib /usr/lib /lib/arm-linux-gnueabihf /usr/lib/arm-linux-gnueabihf /usr/lib/x86_64-linux-gnu/libfakeroot /lib/i386-linux-gnu /usr/lib/i386-linux-gnu /lib/i586-linux-gnu /usr/lib/i586-linux-gnu /usr/local/lib /usr/lib/x86_64-linux-gnu/panda3d /lib/x86_64-linux-gnu /usr/lib/x86_64-linux-gnu /lib32 /usr/lib32 "

felix-salfelder, 2016-11-04:
> No idea about -rpath and why it is set that way.

cargo cult, maybe?

>  do not see any reference to qucs-core/src/.libs

`.libs` is an implementation detail of libtool. automake might know about it, but only in the clean targets.

nvdl, 2016-11-04:
> cargo cult, maybe?

Freedom of though for everyone including humans.

felix-salfelder, 2016-11-04:
> Freedom of thought for everyone including humans.

no idea what that is supposed to mean.

any objection to remove the rpath flag, no matter where it came from, maybe?

nvdl, 2016-11-04:
> any objection to remove the rpath flag, no matter where it came from, maybe?

I have no issue removing it if I know where to look for it. Have not worked with "autoconf", "configure" tools configuration before.

felix-salfelder, 2016-11-04:
its in qucs-core/src/Makefile.am line 113

in3otd, 2016-11-04:
`git blame` says that the `rpath` stuff was added in 92501dfb4a4b116def7eb4c6e789435029059584 ; @guitorri  can you remember why that was needed?

felix-salfelder, 2016-11-04:
let me guess: if `$prefix/lib` is in /etc/ld.so.conf, then libtool does not set an rpath. usually that happens for prefix==/usr/local or prefix==/usr, depending on what you set in /etc/ld.so.conf.

but: ldopen uses ld.cache. and if ld.cache is out-of-date, dynamical loading will not work.

that's why libtool runs `ldconfig` (with superuser privileges) after `make install`. however, if you install with a DESTDIR, and pack a tarball, libtool will kindly ask you to run ldconfig after unpacking. you have to run libtool manually.. and if you forget to do that, the broken rpath can kick in and make it seem to work anyway...

likewise (second scenario), if you build a tarball on a machine with $prefix/lib listed in /etc/ld.so.conf, and then install/unpack it on a machine without that, -rpath seems to work around that.

guitorri, 2016-11-04:
I think Felix guess in on the right direction. The commit just after `git blame` explains a bit the change cb7e2ba. 
People were being bitten by not running `sudo ldconfig`.
I remember I read a bit around and `rpath` could be a solution.  I think there was also something on OSX to make the libraries relocatable by using `rpath`.  I am sorry I don't remember all the details.

felix-salfelder, 2016-11-04:
i claimed that `make install` triggers ldconfig. this is not entirely true and cannot work in some cases
- on my end, `make install` as root, runs `ldconfig -n` my manual says, `-n` implies `-N`, "do not rebuild cache". (?!)
- a (staff) user with write permissions in /usr/local might not have permissions to write /etc/ld.so.cache, so ldconfig wont help.

i think we should remove the -rpath's from the makefiles, it's not the right solution, can we agree on that?
and some relief: lets have distros package/distribute QUCS, so nothing of the above will hit the users anyway.

but: we should maybe revert cb7e2ba or add a pointer to ldconfig in INSTALL or in README

nvdl, 2016-11-07:
Or at least update the compilation Wiki that such things can happen and the work arounds.

guitorri, 2016-11-07:
Feel free to revert and document. I will let you know if something breaks on OSX. I have to check, but I think the rpath will be necessary to make it an OSX bundle relocatable. But this is a packaging issue for the future...

felix-salfelder, 2016-11-07:
guilherme: if the goal is a relocatable package, then consider to set a prefix p different to one without p/lib in /etc/ld.so.conf. libtool will set the rpath to p/lib for you. well *should*. cross fingers.

guitorri, 2017-10-10:
I don't know if this was fixed or not. If not reopen.

closed 2017-10-10 by guitorri

kenogo, 2017-10-25:
It's not fixed. I just ran across this same issue.

felix-salfelder, 2017-10-25:
@kenogo, can you check if this is related to an already installed libqucs? if so, could you remove the rpath (ab)use discussed above and retry?

thanks.

kenogo, 2017-10-25:
@felix-salfelder This is my first install of qucs on this machine, so that can't be the issue I think.

kenogo, 2017-10-25:
> When Octave is no installed, "no --path" is evaluated.
>
> Making all in qucs-doc
> make[1]: Entering directory '/home/nvd/software/simulators/qucs/qucs-0.0.19/qucs-doc'
> Making all in technical
> make[2]: [...]

This is the issue I get, by the way.

felix-salfelder, 2017-10-25:
ok, it seems diffent issues are mixing up here

your bug is that qucs-doc should complain about missing octave. and tell users about `--disable-doc` maybe.
please fix that, if you feel adventurous `:)`

kenogo, 2017-10-25:
@felix-salfelder Ahh thanks! Running
```
./bootstrap
./configure --disable-doc
```
fixed the issue. I'll try working on a PR to make qucs-doc check for octave this weekend.