Qucs/qucs #587: qDebug()

nvdl, 2016-11-05:
Not getting any qDebug() messages on the console.

`./configure --disable-doc --enable-maintainer-mode`

Don't know if `--enable-maintainer-mode` matters or not here; gave it a shot.

Compilation guidelines do not say anything about enabling/disabling the debug mode.

felix-salfelder, 2016-11-05:
this is a Qt feature... compile define QT_DEBUG, e.g.

    $ make CPPFLAGS=-DQT_DEBUG

we should have an --enable-debug flag that exchanges -DNDEBUG for -DQT_DEBUG. please implement it, if you like.

nvdl, 2016-11-05:
I will tend to keep `--enable-debug` and add `--enable-qt-debug` if allowable.
Any pointer which place to start with? Have not dug into build scripts before.

felix-salfelder, 2016-11-05:
an extra qt-debug cannot hurt. go for it!

it's in qucs/configure.ac

in3otd, 2016-11-05:
? `--enable-debug` currently adds `-DQT_DEBUG` and works here.

`--enable-maintainer-mode` may have enabled also debug in the past (don't remember) but this has changed. (and `--enable-maintainer-mode` is now deprecated, IIUC)

nvdl, 2016-11-05:
Configure Information:
  C++ Compiler    : clang++
    DEFS          :   -DHAVE_CONFIG_H
    CPPFLAGS      :   
    CXXFLAGS      :   -g -O2 -pipe -fno-exceptions -Wno-deprecated-register -W -Wall -std=c++0x -O0

  Linker          : 
    LDFLAGS       :   
    LIBS          :   

  Prefix          : /usr/local
  QTDIR           : /lib
  QT_LIBS         : -lQtCore -lQtGui -lQtXml -lQt3Support -lQtSvg -lQtScript
  QT_DEF          : -DQT_DEBUG
  QT_LDFLAGS      : -L/lib 
  QT_INC          : -DQT_DEBUG -DQT3_SUPPORT -DQT3_SUPPORT_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT

configure: Appending gcc optimisation flag -O0 due to --enable-debug setting.

Looks like `-DQT_DEBUG` is not making its way up to CPP flags.

`$ make CPPFLAGS=-DQT_DEBUG` is giving the debug messages.

> (and --enable-maintainer-mode is now deprecated, IIUC)

Have to update the Wiki after we (I) understand how it works now.

nvdl, 2016-11-05:
libtool: compile:  clang++ -DHAVE_CONFIG_H -I. -I../../.. -I../../../src -I../../../src/math -I../../../src/components -I../../../src/components/devices -g -O2 -std=gnu++11 -O0 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__ -MT msmbend.lo -MD -MP -MF .deps/msmbend.Tpo -c msmbend.cpp  -fPIC -DPIC -o .libs/msmbend.o
clang: warning: -ldl: 'linker' input unused

in3otd, 2016-11-05:
Here I have

```
Configure Information:
  C++ Compiler    : clang++
    DEFS          :   -DHAVE_CONFIG_H
    CPPFLAGS      :   
    CXXFLAGS      :   -g -O2 -pipe -fno-exceptions -Wno-deprecated-register -W -Wall -std=c++0x -O0

  Linker          : 
    LDFLAGS       :   
    LIBS          :   

  Prefix          : /usr/local
  QTDIR           : /usr/lib64/qt/lib
  QT_LIBS         : -lQtCore -lQtGui -lQtXml -lQt3Support -lQtSvg -lQtScript
  QT_DEF          : -DQT_DEBUG
  QT_LDFLAGS      : -L/usr/lib64/qt/lib 
  QT_INC          : -DQT_DEBUG -DQT3_SUPPORT -DQT3_SUPPORT_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT
```

> Looks like -DQT_DEBUG is not making its way up to CPP flags.

apparently the `QT_*` flags are added in the sub-`Makefiles`, in some way.

Interestingly, my compile command looks a bit different from yours:
`clang++ -DHAVE_CONFIG_H -I. -I../..   -I/usr/lib64/qt/include -I/usr/lib64/qt/include/Qt -I/usr/lib64/qt/include/QtGui -I/usr/lib64/qt/include/QtCore -I/usr/lib64/qt/include/QtSvg -I/usr/lib64/qt/include/QtXml -I/usr/lib64/qt/include/QtScript -I/usr/lib64/qt/include/Qt3Support -DQT_DEBUG -DQT3_SUPPORT -DQT3_SUPPORT_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT -I../../qucs   -g -O2 -pipe -fno-exceptions -Wno-deprecated-register -W -Wall -std=c++0x -O0 -MT msmbend.o -MD -MP -MF .deps/msmbend.Tpo -c -o msmbend.o msmbend.cpp`

nvdl, 2016-11-05:
Is that variation acceptable if it is due to version difference in tools?
I don't think so.

Is it `libtool` causing issue on my side?
`libtool: compile: clang++`

I am doing:
`./bootstrap`
`./configure --disable-doc --enable-maintainer-mode`
`make`

in3otd, 2016-11-05:
ah, got it, your compile command was for `msmbend.cpp` in `qucs-core` not for the GUI... this is why you do not have the Qt flags there :grin:
There is a file with the same name in `qucs`, my command line is for compiling this latter, not the former.

(FYI, here is the compile command for `msmbend.cpp` in `qucs-core`:
`libtool: compile:  clang++ -DHAVE_CONFIG_H -I. -I../../.. -I../../../src -I../../../src/math -I../../../src/components -I../../../src/components/devices -g -O2 -std=gnu++11 -O0 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__ -MT msmbend.lo -MD -MP -MF .deps/msmbend.Tpo -c msmbend.cpp  -fPIC -DPIC -o .libs/msmbend.o`
`clang-3.8: warning: -ldl: 'linker' input unused`)

nvdl, 2016-11-06:
IIRC, GUI had the same issue and I ended up passing the explicit define `-DQT_DEBUG` to `make`. This behavior is different than previously observed (roughly 1 year ago) where debug was there possibly taken care by `--enable-maintainer-mode`.

Will come back to it later.

in3otd, 2016-11-23:
any news on this issue?
My understanding is that you configured with `--enable-maintainer-mode` expecting to have the debug messages enabled but they actually weren't. 
This worked in the past but in 6c35c46d42d0d444988ffffe18f7bfeafe74067b the "maintainer mode" was removed and now you must use `--enable-debug` to get the debug messages.

felix-salfelder, 2016-11-23:
> configured with `--enable-maintainer-mode`...

should produce a warning. however, `AC_CONFIG_SUBDIRS` disables that, so options can be passed to subpackages more easily.

BUT: qucs-core/configure (does not have submodules) prints a warning... perhaps it should error out instead?

(have not tried `--enable-debug` in a while)

in3otd, 2016-11-23:
ah, I was wondering why passing an unsupported option gave no messages...
When playing with `git bisect` I pass both `--enable-maintainer-mode` and `--enable-debug`, just to be sure - so I don't need to remember when the behavior was changed :grin:

(here I always use `--enable-debug`)

nvdl, 2016-11-23:
"qDebug()" works with `--enable-debug`. Don't know what I missed to begin with.

Separating `--enable-qt-debug` from `--enable-debug` will suffice where only messages are needed.
Now, if I understood it correctly, `--enabled-debug` set -O0 which is not needed for Qt messages.

Compiler debug should be separated from functional (qDebug() as one function here; broadly, :wink:) debug.
https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L735

Gave `configure.ac` a look but did not edit the file.
Don't want to mess it without understanding the switches involved.
https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L28

`./configure` should give out an error and stop if an unknown switch is passed.

felix-salfelder, 2016-11-23:
> `./configure` should give out an error and stop if an unknown switch is passed.

yes. looks much like the problem described/fixed here http://www.midnight-commander.org/changeset/d362fc964ed7c2ffea185ce0138ba3fe141129e9.

is it that simple? please try to turn that into a patch for qucs ...

nvdl, 2016-11-23:
Looking into that.

`CXXFLAGS      :   -g -O2 -pipe -fno-exceptions -Wno-deprecated-register -W -Wall -std=c++0x -O0`

`-O` is appended twice when debug is enabled in a hope that the last one will override the previous one.

`./configure --enable-option-checking=fatal --enable-debug --enable-maintainer-mode` works to detect the unused mode but it is not recommended as the other scripts may be using it.

nvdl, 2016-11-23:
https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L418

Debug should having nothing to do with compilation warnings or am I missing the :thought_balloon:?

nvdl, 2016-11-23:
Another issue?

`./configure: line 7371: QUCS_CHECK_FUNC_QTABWIDGET_SETMOVABLE: command not found`

guitorri, 2016-11-23:
Building out-of-three I observed an issue with `QUCS_CHECK_FUNC_QTABWIDGET_SETMOVABLE` see #576. But after cleaning the local copy (`git clean -xdf.`) and rerunning `./bootstrap` the issue went away... 😕

felix-salfelder, 2016-11-23:
> Debug should have nothing to do with compilation warnings or am I missing the :thought_balloon:?

yes and no. it's definitely possible to over-engineer the configure script. imo it's perfect to only have two modes, debug and production...

NB: there's *always* more than only the options you hardcode into configure. try passing custom flags into the build...
e.g. `make clean all CXXFLAGS=-O0\ -g` should do something useful (no matter how configured).
and `touch some/file.cpp; make CPPFLAGS=-UNDEBUG\ -DQT_DEBUG` should similarly just plain work.

i'm not saying it does not work right now, but i definitely do not want to have this in configure.ac. and i am not sure if -Wall should only go into the debug mode YMMV -- it just does not matter that much if overrides work.

nvdl, 2016-11-23:
Replace `AC_HELP_STRING` with `AS_HELP_STRING`.

https://www.redhat.com/archives/libvir-list/2013-September/msg00324.html
``

nvdl, 2016-11-23:
>  it just does not matter that much if overrides work.

Works for you. Should we make it more friendly for the newbies who read the wiki and start compiling?
I think compilation warnings have nothing to do with debugging.

`Hey Joe, What are you upto? Eh, I am debugging the compilation.`

nvdl, 2016-11-23:
`QUCS_CHECK_FUNC_QTABWIDGET_SETMOVABLE`

https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L729

Is it a script or autoconf declaration?
It shows up "as it is" in the `configure` script.

guitorri, 2016-11-24:
It is a m4 macro: 8932f71fb59

in3otd, 2016-11-24:
> -O is appended twice when debug is enabled in a hope that the last one will override the previous one.

at least for `gcc` it's not a hope but a certainty, see the [`gcc` docs](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#Optimize-Options): *If you use multiple -O options, with or without level numbers, the last such option is the one that is effective.*

> Is it a script or autoconf declaration?

it's a `m4` macro, see https://github.com/Qucs/qucs/blob/master/qucs/m4/qt_features.m4#L18

> I think compilation warnings have nothing to do with debugging.

I beg to differ: debug starts by looking at the compile warning messages (this is why, ideally, we should polish the code so that there are no compile warnings). Then one looks at the `qDebug()` messages printed at runtime. Then one runs the app under `gdb` and looks at the debug messages to understand where to break the execution, inspect the call stack and add breakpoints.

nvdl, 2016-11-24:
What if I want to see compile warnings only and do not want to use -O0 (no optimization) to fix warnings?

The way I see it:
`--enable-compiler-warnings`
`--enable-debug`
`--enable-qt-debug`

And regarding `gdb`, do we need `-g3` or is `-g` enough?

nvdl, 2016-11-24:
Is `--enable-doc` a better choice (disabled by default)?
Will it break something in the current build system?

in3otd, 2016-11-24:
> What if I want to see compile warnings only and do not want to use -O0 (no optimization) to fix warnings?

I do not see why you would *need* to do that but, as I think felix was suggesting, every flags combination is (should be) possible by passing the appropriate custom flags on the command line. No need to think of every possible combination of compile-time flags and make them available as `configure` options. Just `--enable-debug` for enabling "all" debug stuff is enough, custom cases will be handled by every user/developer to their own wishes with appropriate command lines they will figure out :grin: .

nvdl, 2016-11-24:
`CFLAGS : -g -O2 -pipe` with `--enable-debug`.
On purpose? Any C code in this project?

nvdl, 2016-11-24:
https://github.com/Qucs/qucs/pull/614