Qucs/qucs #591: Error messages in Qucslib

ra3xdh, 2016-11-06:
Every user library with `Default symbol` defined gives an error message when trying to open in library manager. Steps to reproduce:

* Move `NMOSFETs.lib` library into `$HOME/.qucs/user_lib`
* Select this library at ComboBox. Use user version of library.
* The following error message will be shown:
[image]

I think, it appears after #556

felix-salfelder, 2016-11-06:
FTR: i have rebased the non-hotfix approach to #556. see #592. (help definitely appreciated).

ra3xdh, 2016-11-06:
#592 is not a good solution to fix this particular issue, because it is a deep refactoring proposal. I don't recommend start refactoring right at release time. Let's discuss your refactoring concepts after 0.0.19 will be released.

felix-salfelder, 2016-11-06:
yes, the non-hotfix approach is currently targeting `devel`, only with some help from others that might change.

and no, there's no "deep refactoring" involved just for this, i intentionally postponed that. read the comments/commits and disussion.

ra3xdh, 2016-11-11:
There exists another issue in Library Manager. It always show an an empty symbol for devices (not subcircuits) libraries. Try to select `Transistor` or `Diode` library and you will see the following. Component symbol will be empty.
[image]

guitorri, 2016-11-19:
Seems fixed with #594

closed 2016-11-19 by guitorri