Qucs/qucs #593: Cannot paste schematic from previous version(s)

in3otd, 2016-11-06:
In the GUI we can open schematics from previous versions but when trying to paste a schematic from a previous version a "Wrong document version" error pops up.

This comes from [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/schematic_file.cpp#L131) where the pasted schematic version needs to be the same as the current Qucs version.

To easily reproduce:
- open a schematic
- select all, copy and paste in a text document
- change the version in the first line to `0.0.18`
- select all, copy and try to paste in a schematic

(could maybe reuse the first [commit](https://github.com/Qucs/qucs/pull/340/commits/85bb09531dfe8ca0237ab17233feaccf144c12a1) of #340 to ease checking)

ra3xdh, 2016-11-06:
Yes, I know about this problem. I always use clipboard editing to allow copy-pasting between versions. This issue should not concern simple users, because they usually uses only single Qucs version. But it makes development and debugging a bit more difficult.

felix-salfelder, 2016-11-06:
perhaps, as long as loading old schematics is not fully tested, we should warn "loading old schematic, please be careful", instead? `return false` seems to be entirely pointless to me.

in3otd, 2016-11-06:
yep, I was sure this was already known, but wanted to have a remainder to fix it :grin:

Currently we load old schematics without any user warning, I think we should have the same behavior (emit warning or not) when pasting. When `0.0.19` will be out, all existing users' (vs. developers) schematics will become old.

guitorri, 2016-11-06:
There also something about loading future versions:
https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/dialogs/qucssettingsdialog.cpp#L134

in3otd, 2016-11-19:
I was trying to understand the logic of the current code about loading from future versions: at present having it checked or unchecked makes practically no difference...

- when trying to load a file from a future version, the user is *always* asked if ey really wants to open the file (see the code [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/schematic_file.cpp#L868-L885)), regardless of the related setting
- also, currently we show all the `.sch` files in the tree view, also from future versions, regardless of the related setting status (see the code [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/projectView.cpp#L130-L138); `n` is negative for future versions but this case is not actually treated differently)

So, what I propose is to modify the current behavior as:
- if *load from future versions* is enabled:
 - show future-version schematic in the tree view
 - open them without any warning message
 - paste from future versions without any warning message
- if *load from future versions* is **not** enabled:
 - do **not** show future-version schematic in the tree view
 - abort opening them (if forced thru the *Open* dialog) with an error message
 - abort pasting from future versions with an error message
This is different from the behaviour of previous versions: in `0.0.17` the files from future versions were not shown (see the code [here](https://github.com/Qucs/qucs/blob/qucs-0.0.17/qucs/qucs/qucs.cpp#L860-L872))

seems reasonable ?

in3otd, 2016-11-23:
Fixed in #607

closed 2016-11-23 by in3otd