Qucs/qucs #601: Component Renaming

nvdl, 2016-11-14:
Add a component c1.
Add another component c2.
Rename c2 to same name as c1.

Nothing happens.

It should follow the same routine used for copy->paste i.e. rename with a number in the end.
Or
Warn the user and ask for new name again.

in3otd, 2016-11-14:
interesting; here the behavior is different

nvdl, 2016-11-14:
If "Lg" is renamed to L2, there is no warning. It stays "Lg".
This is a simple example.
If there are many components and one wants to give one the next available "free" number, that does not work. If there is R1 and R100. Renaming R100 should warn and propose R2.

[image]

in3otd, 2016-11-14:
ah, it was really not clear what the issue was. That behavior is as expected.
Note that the reference designators are really arbitrary: you can name your inductor as `C1` or as`MyNiceResistor`, etc.
What *might* be useful is a mass-renaming, to reassign all the references, maybe. I think Kicad has that.

nvdl, 2016-11-14:
Yes, something like that.
Copy paste has some "intelligence" but auto renaming when the designator number is too high and one wants to guess wildly is silently dropped.
I was thinking to fetch code from there.

[image]

Copying C1 and pasting creates C201. Good but it should be C2. At least for numbered components.

nvdl, 2016-11-14:
A designator that does not end with a number is appended with one which is good and then it keeps on incrementing.

[image]

in3otd, 2016-11-14:
> Copying C1 and pasting creates C201. Good but it should be C2.

Not always. "Holes" in numbering are sometimes used to separate different functional parts of a circuit.

nvdl, 2016-11-14:
Then mass renaming seems plausible or selecting components via selection box and auto-renaming only those with a starting sequence number without conflicts.

ra3xdh, 2016-11-16:
@in3otd , Yes every PCB design CAD (PCAD, Eagle) supports mass renaming. But simulators (MicroCAP) often have no such function. PCB CADs often are used for end product documentations set preparation. And many national standards forbid usage of combined letter-number part names (such as `Lg`) and part number gaps. But circuit simulator should not be used for this purpose. And automatic component renaming has no sense for it. 

@nvdl  Combined letter-number part names (such as `Rd` and `Cs1`) are often used in parametric circuits. And such names should not be forbidden. The same for component numbering gaps. Suggested copy-paste behavior (C201->C202) also covers only specific case. User can copy C201 and paste into another circuit part or even in another schematic fragment where components are numbered as C2, etc.

nvdl, 2016-11-16:
I agree, it is not so critical here. I was looking into the generated netlist for ngspice and wanted to have continuous or somewhat controlled "auto naming" to follow the components.

Can you point me to the source that is doing qucs schematic to ngspice netlist conversion?
Since, I did not add parameter to the switch, the netlist was not correct. Shouldn't it have default parameters to begin with?

I saw some other issues as well possibly related to the switch with no definitions.