Qucs/qucs #609: Drag and drop schematic file behavior

guitorri, 2016-11-23:
Example:
- open Qucs
- drag into existing schematic into it
- run the simulation

It will load the schematic and even simulate it, but the containing project is not open, so it does not list the other contents associated with the schematic.

Should it open the containing project first? Should schematics only work if they are inside a `_prj` directory?

in3otd, 2016-11-23:
IMHO...

> Should it open the containing project first?

not unconditionally. I often have a (new) project open and need to open single schematics from other projects to take a look at some previous design implementations and/or copy some parts.
It may be useful to warn the user that the schematic is part of a project and ask if ey want to open the containing project also.

>  Should schematics only work if they are inside a _prj directory?

I think this has been briefly discussed in the past also; I'm against because sometimes I just need a quick simulation of a single schematic and not really to have a nice, separate, project. But I have to admit I have a `tmp_prj` where I move all the test/temporary/unfinished single schematics when there are too many of them in the Qucs projects directory...
For beginners the concept of a "project" is sometimes not immediately obvious, they just want to simulate a schematic. Sure, the concept is readily grasped after creating too many schematics :grin:
I recognized these above may be weak reasons to allow schematics outside projects, but I wonder what we will gain by forcing to use projects always.

ra3xdh, 2016-11-23:
>  Should schematics only work if they are inside a _prj

Definitely no. I often use standalone schematics for design purposes. And I often send schematic detached from the project via e-mail. I have report form other users that they also use standalone schematics.

guitorri, 2016-11-23:
Fine. What about the project contents dock? Should the GUI it try to populate with files found in the same location as the schematic file?
 
Standalone schematics will work as long as it contains only Qucs primitives or system libs. Right?

in3otd, 2016-11-23:
> Should the GUI it try to populate with files found in the same location as the schematic file?

? you mean when already in a project and opening a "foreign" schematic? I'll say no, it's the *project content* dock, so only stuff belonging to the project should go there

IMHO, standalone schematics/data displays/etc. could be made visible in the `Projects` dock (to be better renamed...) and opened by double clinking them - if they are in the Qucs home dir. We now have icons there, ideally project dirs would have different icons than real dirs and one would be able to navigate the filesystem there. But I digress, all this is post `0.0.19`, maybe.

>Standalone schematics will work as long as it contains only Qucs primitives or system libs. Right?

not sure, but I think they should "always" work (assuming they are opened outside of a project)

closed 2016-11-24 by guitorri