Qucs/qucs #625: Retrieve German translations

guitorri, 2016-11-30:
A number of German translations were lost before 0.0.17, in this commit 6663635

See comments in the end of #353 for context.

aimylios, 2017-10-14:
I took care of this and retrieved all German translations which were still present [six years ago](https://github.com/Qucs/qucs/blob/8a89b51a92e6d296fb6b034130a3c7f74fa20ca8/qucs/qucs/qucs_de.ts) but missing from the current translation on Transifex. I tried to upload the new qucs_de.ts as described [here](https://docs.transifex.com/translation/importing-translations), but that had no effect, it still shows 50,52% translated (should be much more now). Please let me know if you have any idea how to do this properly.

aimylios, 2017-10-14:
OK, I just tried it once again, and it worked. We're now at 87,64% for the German translation. This issue can be closed.

Maybe I'll take some time another day to review the existing strings and add what is missing. But I'd already be happy if the new old translation would find its way into the next release.

guitorri, 2017-10-14:
Awesome. Thank you @aimylios.
I will make sure to update the translations for the next release.

closed 2017-10-14 by guitorri