Qucs/qucs #626: Replace translated strings on comparisons.

guitorri, 2016-11-30:
After #353 comparisons involving translatable strings seem a little fragile.

Is there a way to avoid them?

For example for things like this:
```
$ grep -r 'tr(' . | grep 'if *(.*=' --color
./components/componentdialog.cpp:  if(show == tr("yes"))
./components/optimizedialog.cpp:    if (typeStr == tr("linear double")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("logarithmic double")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("linear integer")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E3 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E6 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E12 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E24 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E48 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E96 series")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("E192 series")) {
./components/optimizedialog.cpp:    if (typeStr == tr("minimize")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("maximize")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("less")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("greater")) {
./components/optimizedialog.cpp:    } else if (typeStr == tr("equal")) {
./diagrams/diagramdialog.cpp:    if(Diag->Name == "SP") VBox11Layout->addWidget(new QLabel(tr("number")));
./qucs.cpp:  if (CompChoose->itemText(0) == tr("Search results")) {
./qucs.cpp:  if (item == QObject::tr("verilog-a user devices")) {
./qucs.cpp:  if (CompChoose->itemText(0) != tr("Search results")) {
./qucs.cpp:  if (CompChoose->itemText(0) == tr("Search results")) {
./qucs.cpp:  if (CompChoose->itemText(0) == tr("Search results")) {
./qucs.cpp:  if(category == tr("Schematics")) {
./qucs.cpp:  else if(category == tr("VHDL"))
./qucs.cpp:  else if(category == tr("Verilog"))
./qucs_actions.cpp:        if (it.value()->category == QObject::tr("verilog-a user devices")) {

```

in3otd, 2016-11-30:
those are for sure fragile: a translator may set two different translations for the same string. OTOH, I think that Qt by default uses the same translation for the same string, unless the translation file has different translations, but I need to check if this is really true.

I think most of those comparisons check the content of a `QComboBox` or similar so in theory we should be able to just compare index value instead of the actual string.

felix-salfelder, 2016-11-30:
> Is there a way to avoid them?

indeed. that's what virtual functions are for.

in C++ you would call a virtual function e.g. `something->do_stuff()`. i.e. `function` is virtual in the common baseclass of the objects involved. it can be specialized for the type of `something`, so it does special stuff, instead of (what we have)

    if(something->somestring=="somethingspecial"){
        do_special_stuff_on(something);
    }

virtual function overrides also force the code into the right place (to the object, not to the caller), so more and other objects can be added later on.

> I think most of those comparisons check the content of a QComboBox or similar so in theory we should be able to just compare index value instead of the actual string.

that makes sense (but only where applicable...)