Qucs/qucs #632: Error messages in "graph.cpp" slow down debugging

ldpgh, 2016-12-17:
While doing debug using QtCreator and latest commit ....
* My Cartesian diagram is showing just a part of the simulated data range (5n->6n)
* In this context the console-output is flooded with error message. The messages are associated to negative values given/returned by the listed functions in `graph.cpp`
   * `void Graph::ScrPt::setScrX(float x)`
   * `void Graph::ScrPt::setScrY(float x)`
   * `float Graph::ScrPt::getScrX() const`
* Those function are called very very often if the diagram is redrawn (e.g. clipping algorithm).

It's a heavy performance drop in case of debugging.
The impact on normal use case has not been checked yet.

Test schematic: [test_negativescreencoordinates.sch](https://gist.github.com/ldpgh/15297b1de09c3d7699a4cbb2d32a42d1#file-test_negativescreencoordinates-sch).

in3otd, 2016-12-17:
I think there is already a PR including a fix for this, see https://github.com/Qucs/qucs/pull/314#issuecomment-141641565 and #370 , please check if that works ok .
(yes, sometimes all those debug messages slow down the graphs redraw quite a lot)

ldpgh, 2016-12-17:
Now I'm definitely nervous ... maybe just confused ...
#314 is closed, but I do not see any of the changes made in 2015 while cloning the repo 2016/Nov (same for #370)
#370 may not easily merge with #631 ... (changes in `rect3ddiagram.cpp`)

Btw. ... have seen already graph-segments drawn outside of the regular 2D Cartesian diagram region.

in3otd, 2016-12-17:
#314 was closed "by mistake" and #370 is its continuation, so the code there was not yet merged anywhere. My comment above was just to say that the debug messages should be reduced once that PR is merged.

I don't remember seeing segments out of place in 2D graphs, please open an issue if you happen to see that again.

closed 2018-07-19 by guitorri