Qucs/qucs #633: Unexpected Duplicate Schematic Behaviour

nouwaarom, 2016-12-17:
Steps to reproduce:
* create a new project
* create a  schematic
* make a simple circuit
* simulate the circuit
* add some diagrams in the data display
* duplicate the schematic file (using right mouse button -> duplicate in the content menu)

Expected:
the data display file and dataset corresponding to the original schematic to be duplicated too

Actual:
both the original and the duplicated schematic use the original data display file and dataset

in3otd, 2016-12-17:
personally, I'm used to that behavior but I see it can be confusing at first, also because it's not consistent with what "Save as" does.
It may make sense to do what "Save as" does also when duplicating the schematic, i.e. using the new (base)name also for the default Dataset and Data Display of the duplicated schematic.

I'm not in favor of having "Duplicate" create other files also because it should apply only on the selected file, IMHO; also the Data Displays and Datasets can be duplicated manually if one wants/needs to.

cvgaviao, 2017-02-22:
It would at least to change the names of files in "document settings" to the new name that was choose.

in3otd, 2017-02-22:
I thought (a little) about that and I think the current solution is the best compromise: using "Save As..." changes the names of files in "Document Settings", while the operations available in the drop-down menu from the mouse right button in the project content view operate at the file-level only, so without changing anything else. We should of course document the behavior...

ra3xdh, 2017-06-28:
See also https://github.com/ra3xdh/qucs/issues/26 for the same issue report. Maybe it's better to emulate a `SaveAs` operation in `QucsApp::slotCMenuCopy()` instead of simple copying of file? Current behavior when file is just copied  may mislead user.

in3otd, 2017-06-29:
to expand a little bit on my previous comment, we currently have two functionalities
- the *Duplicate* in the *Content* tab context menu:
  this just copies the schematic file, it does not modify the copy in any way; the file content is the same as the original file, only the name is different. The schematic still points to the original Dataset and Data Display files
- the "Save as..." menu item in *File*:
  this can be used to save an existing schematic with a different name **and** it will change the schematic Dataset and Data Display names to use the same basename as the new schematic file

I'm reluctant to change any of these because I actually use both: I may have several different version/refinements of a schematic, which I create incrementally using the *Duplicate* menu, so I still have one Data Display will all the graphs and markers/etc. I need and I use *Save as...* to create a separate schematic from some existing one I have around.

I will tend to leave things as they are, since I see the *Content* tab context menu as operating on single files only - we have *Duplicate*, *Rename* and *Delete*. To make things less confusing for new users we may change *Duplicate* to *Copy file* or put a header *File operations* to the context menu (if possible). And, of course, we could actually have some documentation for this :grin:

ra3xdh, 2017-07-10:
> we may change Duplicate to Copy file 

Yes, it will be the best solution

> And, of course, we could actually have some documentation for this 

We can add a note on "Duplication" here: https://qucs-help.readthedocs.io/en/latest/short.html#mouse-in-content-tab

I can prepare a fix for this in this week.

in3otd, 2017-07-15:
fixed in #695 and  Qucs/qucs-help#15

closed 2017-07-15 by in3otd