Qucs/qucs #636: Question on qucs-filter development

andresmmera, 2016-12-30:
Hello all, 

I'm working on qucs-filter in my spare time and I wanted to ask you a few things before going further:

1) So far, the schematics are hardcoded on the filter design functions (for example, [see](https://github.com/Qucs/qucs/blob/master/qucs/qucs-filter/line_filter.cpp#L116)). I reckon that this approach is poorly scalable and leads to write much more lines than actually needed. To my mind, filter design and schematic generation should be treated separatedly. I mean, it would be 
more convenient to have an (unique) class (let's say 'SchematicGenerator' class) to convert 
the filter values and the ladder structure into a Qucs schematic. That way, every filter design function 
would instantiate an object of 'SchematicGenerator' and forget about schematic generation (component location, wires, simulation blocks, etc...).

'SchematicGenerator' should receive (at least) two parameters:

- vector< int > Structure: Contains the network structure according to a look-up-table

Code | Description
------------ | -------------
0 | Series cap
1 | Shunt cap
2 | Series inductance
3 | Shunt inductance
4 | Mid-series resonator
5 | Shunt resonator
... | shunt/open stub, etc.

- vector< double > Values: Contains the values of the network components.

Example:

Let's say we want to design a fifth order lowpass Chebyshev filter. Then, the function which designs
Chebyshev I filters instantiates a 'SchematicGenerator' object with the following parameters:

Structure={1,2,1,2,1} //Shunt cap, Series ind, Shunt cap, Series ind, Shunt cap
Values={7.025e-12, 8.976e-9, 9.875e-12, 8.976e-9, 7.025e-12}

After that, the SchematicGenerator object, through a public method (say QString SchematicGenerator::getQucsSchematic() ) returns the Qucs schematic to the 'lc_filter.cpp' and this is copied into clipboard.

I did something similar [here](https://github.com/andresmmera/qucs/blob/MatchingTool/qucs/qucs/dialogs/matchdialog.cpp#L1553) for ladder structures and it saved quite a number of lines. I think that this code could be extensible to qucs-filter by adding series/shunt resonators (for Cauer and Chebyshev II filters)

2) Concerning the synthesis of LC filters, two classes are involved:

[lc_filter](https://github.com/Qucs/qucs/blob/master/qucs/qucs-filter/lc_filter.cpp): It calculates Butterworth, Chebyshev I and Bessel filters from the lowpass prototype and generates the schematic in a pretty entangled way. I honestly think that 1) would help a lot here...

[qf_cauer'](https://github.com/Qucs/qucs/blob/master/qucs/qucs-filter/qf_cauer.cpp): It designs Cauer filters. It seems that someday it belonged to qucs-filter-v2 and, at some point, it was ported to qucs-filter. The documentation is not as detailed as I'd like and it works for 3rd order filters. I understand more or less what it does, but I cannot identify where the problem is. Some time ago, I was in the need of having a Cauer filter design tool, so I wrote a new class for designing symmetric Cauer filters from scratch (zero-pole calculation, input impedance calculation and ladder filter synthesis, refs. [2] and [3]). It works for even order filters (conventional odd order Cauer filters are antisymmetrical, Rsource != Rload) and I think it brings more light than 'qf_cauer'. Should I continue this way or, maybe should we keep 'qf_cauer' class?

The current qucs-filter LC filter design workflow is as follows: In https://github.com/Qucs/qucs/blob/master/qucs/qucs-filter/qucsfilter.cpp#L398, qucs-filter checks if the response prototype is Cauer or not. If the response is Cauer, 'qf_cauer' will design the filter, otherwise, 'lc_filter' is used.

To my mind, 'lc_filter' should provide Cauer filters too. IMHO, the way to achieve it is to split Butterworth, Chebyshev I, Bessel and Cauer into separate classes and instantiate them in 'lc_filter' when needed. I hope not being pedantic, but I'm thinking about adding a few more response prototypes (Legendre, Linear phase with equirriple error and maximally delay flat [1]), but implementing them on the current code state will lead to a mesh.

The thing is, do you like the ideas 1) and/or 2)?

Regards,

Andrés

[1] 'Handbook of filter synthesis'. Anatol I. Zverev. John Wiley and Sons. 1967
[2]  Cauer Filters: A research monograph. Klaus Huber. 2011
[3]  Analog Filter Design using MATLAB. Lars Wanhammar. Springer. 2009

in3otd, 2016-12-30:
1.

> filter design and schematic generation should be treated separatedly

yes, this will also help in having a consistent schematic style for the different filter types...
I think that in the long term `qucs-filter` should handle also designs based on coupling matrix synthesis, so in general with cross-couplings; the resulting topology is not a simple ladder so it can not be handled by a circuit description like you suggested. Does it make sense to try to generalize that to include cross couplings? Maybe it's too early and trying to handle this will generated unneeded/used complexity.

2.

It would be nice if `qucs-filter` could be extended to "arbitrary" filter responses and the corresponding (ladder) network obtained by, e.g., Brune synthesis. This should be also somewhat similar to a broadband matching network design so maybe some code/structures could be common.
So if having separate classes for the different filter design types will still allow in the future to have an "arbitrary" filter class, I do not see a problem with that :grin:

andresmmera, 2016-12-31:
I'm not familiar with the CMS method, so I've done a quick search on the internet... I've found [this article](11_Vishal_Mishra_778_Research_Article_EEC_July_2012.pdf) which shows the generalized equivalent circuit of a multicoupled filter (Fig.1, page 2). At the first glance I guess that the circuit description above could be extensible to multicoupled inductor.

guitorri, 2017-02-18:
Related to #161.

closed 2017-11-19 by andresmmera