Qucs/qucs #637: Path issue building Verilog-A module in Windows

in3otd, 2017-01-12:
On my Windows system, trying to build a Verilog-A module using the latest Qucs portable fails:
```
mingw32-make.exe -f C:\qucs-0.0.19-rc3-win32-mingw482-asco-freehdl-adms\include\qucs-core\va2cpp.makefile ADMSXML=C:\qucs-0.0.19-rc3-win32-mingw482-asco-freehdl-adms\bin\admsXml.exe PREFIX=C:\qucs-0.0.19-rc3-win32-mingw482-asco-freehdl-adms MODEL=TestRC

sh: C:\Program: No such file or directory
C:\qucs-0.0.19-rc3-win32-mingw482-asco-freehdl-adms\include\qucs-core\va2cpp.makefile:60: recipe for target 'TestRC.cpp' failed
mingw32-make.exe: *** [TestRC.cpp] Error 127
```
(and the `admsXml` tab still show a green checkbox, not sure if that should reflect the status)

Google tells me that this is due to the fact that I have also Git Bash installed, see [here](http://www.openstm32.org/forumthread769) and [here](https://github.com/STAT545-UBC/Discussion/issues/54) .

Inserting a `SHELL=cmd.exe` [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs-core/src/components/verilog/va2cpp.makefile#L27) in `va2cpp.makefile` fixed that. But it should probably handled a bit differently, looking at the check for `MINGW32` just after. 

(BTW, I used [RC_prj.zip](https://github.com/Qucs/qucs/files/133353/RC_prj.zip) as a test project for this)

felix-salfelder, 2017-01-15:
> But it should probably handled a bit differently

yes. i think it's important to have exactly one place for those platform specific choices. there are many ways, e.g. the environment.

just set a shell in the environment and use that (if set). otherwise let `make` choose. should be very similar for `RM` and perhaps more. how about `QUCS_SHELL` and `QUCS_RM` for now?

and... how does qucsator know that it needs to call `mingw32-make.exe`? is there a `QUCS_MAKE` variable already? in particular the makefile is not the right place for platform configuration.

guitorri, 2017-01-18:
@in3otd 

> (and the admsXml tab still show a green checkbox, not sure if that should reflect the status)

The admsXml tab status looks for a few error strings and `Error 127` is not one of them, see [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/messagedock.cpp#L105-L132). Some stderr catching mechanism would be better. Sorry for the fancy half done feature. I never had time to go back to it...

>Google tells me that this is due to the fact that I have also Git Bash installed

I also have Git Bash installed, for some reason `sh` is not a problem here. I recall some issues with using MSYS with Git Bash installed, maybe I took Git out of the PATH or changed the PATH order.

I think that setting the default `SHELL=cmd.exe` on Windows will do for now.  I will test also here and send a PR.

> But it should probably handled a bit differently, looking at the check for MINGW32 just after.

The MINGW32 check is to detect if the makefile is running under MSYS, which is not the case for the 'official' portable Windows package.

@felix-salfelder 
yes, we can move to use variables and do platform settings elsewhere. After the release maybe?
The `mingw32-make.exe` is hardcoded [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/qucs_actions.cpp#L1456-L1460). Yes, we should pull the hardcoded names out into env. vars set in a configuration file somewhere.

`quscator` should, but it does not call `admsXml` or `make`. At the moment it is all done interactively in the GUI. `qucsator` just loads the dynamic libraries when told to do so. 
I want to push these steps down to `qucsator`. Something like a `.hdl myModel.va` netlist command needs to be implemented.

guitorri, 2017-01-20:
Added the SHELL default on Windows.

closed 2017-01-20 by guitorri