Qucs/qucs #644: Travis CI, random issues running qucs-test.

guitorri, 2017-01-21:
After the OSX image was downgraded to xcode6.4 random issues started to show up when running qucs-test.

Examples of errors below:
```

Project :  /Users/travis/build/Qucs/qucs/qucs-test/testsuite/DC_AC_bbv_prj
Running :  /Users/travis/build/Qucs/qucs/qucs-core/src/qucsator -i /Users/travis/build/Qucs/qucs/qucs-test/testsuite/DC_AC_bbv_prj/netlist.txt -o /Users/travis/build/Qucs/qucs/qucs-test/testsuite/DC_AC_bbv_prj/test_bbv.dat
WARNING:Process return code: -11
Failed with error code, saving: 
   /Users/travis/build/Qucs/qucs/qucs-test/testsuite/DC_AC_bbv_prj/error_code.txt
```

```
Terminating process, timed out 1 s
 +++ Qucsator ERROR +++ 
 +++  no result +++ 
[  >FAIL ]  TEST 069  [ arccos ] ==> [TAG_DOUBLE] = ['TAG_DOUBLE']
---
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV"saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
Eqn:Eqn1 check="arccos(-1.0)" Export="yes"
Expected:
arccos(-1.0)
    3.14159265359
Got:
Traceback (most recent call last):
  File "run_equations.py", line 667, in <module>
    with open(outDat, 'r') as myDat:
IOError: [Errno 2] No such file or directory: '/tmp/qucstest/test068.dat'
make: *** [eqncheck] Error 1
The command "if [ "$OSX" = true ]; then make; make qucscheck; make eqncheck; else make distcheck DISTCHECK_CONFIGURE_FLAGS="${DISTCHECK_CONFIGURE_FLAGS}"; fi" exited with 2.
```

I tried disabling qucsatorcheck, but now the issues moved up to eqncheck.
It might be related to the multiprocessing or numpy (used to calculate the differences).

guitorri, 2017-02-13:
It is an issue with qucs-test.
I managed to reproduce. Seems related to the multiprocessing module.

in3otd, 2017-02-13:
...maybe we could avoid using `multiprocessing` when only one process is actually requested (the default), or disable it when not on Linux, similarly to what is already done for Windows (see also Qucs/qucs-test#18).

guitorri, 2017-02-13:
Rigth, disabling `mutliprocessing` seems to do the trick `--qucsator`. 
Now `run_equation.py` needs some attention.

closed 2020-02-15 by guitorri