Qucs/qucs #646: Releasing binaries to GitHub

VINTproYKT, 2017-01-22:
I'm experiencing problems with downloading Windows binaries from SourceForge.

It is possible to upload binaries also on GitHub, please do it if you're not tempted with it.

I really want to give a try for Qucs.

in3otd, 2017-01-26:
release 0.0.19 has been just closed and the GitHub release contains only the source code, as usual.

Could you please try again to download from Sourceforge - latest version is at https://sourceforge.net/projects/qucs/files/qucs-binary/0.0.19/

VINTproYKT, 2017-02-14:
SourceForge makes a lot of sense to show advertisements and not download. I have even tryed "direct link" but this couldn't work.

There is way to upload binaries under release's version tag to GitHub:
https://help.github.com/articles/creating-releases/
Look at step 7.

guitorri, 2017-10-09:
We haven't discussed so far about moving binaries to Github.
For the moment they will stay in SourceForce.

closed 2017-10-09 by guitorri