Qucs/qucs #651: Merge new RF devices and tools

ra3xdh, 2017-01-29:
There exists a lot of excellent new RF devices and tools (mainly by @andresmmera ) that are waiting to be merged. Now it's time to merge it in the main branch. Here will be the countdown of these features:

- [ ] Rebase and merge circular waveguide #356 
- [ ] Test again, rebase and merge the matching tool #339 
- [ ] Rebase and merge tapered line coponent #380 
- [ ] Rebase and merge lumped RF devices #416 
- [ ] Adjust and merge an enhanced markers #407 
- [ ] Rebase and merge transcalc stripline #510 
- [ ] Adjust and merge power combining tool #509 

@andresmmera , Are there other related stuff? Is this list complete? If you are time-constrained, I can rebase you PRs. 

@guitorri , @in3otd  should we target these PRs right to the `develop` branch, or use an intermediate branch, for example `RF_devices`. And then merge this intermediate branch to the develop?

guitorri, 2017-01-29:
Let's try to follow the branching model more closely. 
Ideally we should merge individual feature branches into develop and later create the release branch. Release branches should be only about bug-fixes, no new features (the opposite of what we did for so far).
Intermediate feature branch should be fine, but it delays the merge of individual features into develop.
If it helps collaboration, move the branches to the main repo, and delete after merge.

andresmmera, 2017-01-29:
> @andresmmera , Are there other related stuff? Is this list complete? 

Thanks Vadim for your interest on these PRs. From my point of view, #356, #339, #380, #510 and #509 are quite mature and work as expected. However, I'd like to take a close look to #416 and #407 again since I cannot remember their state right now. 

On the other hand, you may also add #555 to the list (as long as you and @in3otd , etc. agree, of course )
Just for the record, I've closed #415 since I've been working on that lately...

Despite not being related to RF/MW design, I think that the shortcut editor from @yodalee could be merged in 0.0.20 (see #558)

> If you are time-constrained, I can rebase you PRs.

Please, feel free to do that. You can count on me, but I'm really short of time these days...

ra3xdh, 2017-01-29:
@guitorri , Summarizing all above, I will merge these RF-related PRs directly into the develop.

ra3xdh, 2017-01-29:
@andresmmera , I will start, from #356 . I will prepare a rebased version for you tomorrow.

tipofthesowrd, 2017-10-09:
Maybe other components which could be added to the RF devices are related to multiple resonator structures. Currently Qucs has only one resonator models.
Especially paging @ra3xdh and @andresmmera :smiling_imp: 
I once tried adding a component like this to Qucsator but quickly got lost.

i.e.

Multiple Rectangular Coupled Coaxial Lines
https://awrcorp.com/download/faq/english/docs/Elements/RCCOAX.htm

And

Multiple Edge Coupled Microstrip Lines
https://awrcorp.com/download/faq/english/docs/Elements/MXCLIN.htm

andresmmera, 2017-10-09:
@tipofthesowrd  I'm not familiar with Galerkin's method, but I completely agree with you. According to what @guitorri said [here](https://github.com/Qucs/qucs/pull/416): 

> We should not add more stuff to the core without checking for coverage of the new code.

we'd better wait until the next release to tackle this. On the other hand, you/we/me/whoever can start scratching that in octave (or any other environment). Surely that will speed up things in the future.

tipofthesowrd, 2017-10-09:
I will check if I still have some papers on this. I started once on this for microstrip lines and have to look up my sources.

guitorri, 2017-10-09:
@tipofthesowrd  please open a separate issue with your feature request. 

I am closing this one. The checklist is almost complete, the review of the missing items is assigned to me.

closed 2017-10-09 by guitorri