Qucs/qucs #653: error compiling qucs 0.0.19 (master)

zerberros, 2017-01-29:
O.S.:              Arch linux
Architecture:  x86-64
error (last lines):

`make[3]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-core/tests'
make[3]: Entering directory '/home/zerberros/source/qucs/qucs/qucs-core'
make[3]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-core'
make[2]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-core'
make[1]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-core'
Making all in qucs-doc
make[1]: Entering directory '/home/zerberros/source/qucs/qucs/qucs-doc'
Making all in technical
make[2]: Entering directory '/home/zerberros/source/qucs/qucs/qucs-doc/technical'
no --path . mszl.m
make[2]: no: Command not found
make[2]: *** [Makefile:618: mszl.eps] Error 127
make[2]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-doc/technical'
make[1]: *** [Makefile:383: all-recursive] Error 1
make[1]: Leaving directory '/home/zerberros/source/qucs/qucs/qucs-doc'
make: *** [Makefile:380: all-recursive] Error 1
`

zerberros, 2017-01-30:
O.k. I do not have octave-epstk installed.
there is not in AUR.

in3otd, 2017-01-30:
the docs have a lot of dependencies, unless you plan to modify them it's better to grab the precompiled ones on Sourceforge and pass `--disable-doc` to `configure`. We should maybe have `--disable-doc` by default there.
Which icons are missing in the GUI? Both the components icon in the Components tab on the left and the application icons in the toolbar?

zerberros, 2017-01-30:
> Which icons are missing in the GUI? Both the components icon in the Components tab on the left and the application icons in the toolbar?

Yes, all those.

in3otd, 2017-01-30:
uhm, could you try modifying [this line](https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L710) in `qucs/configure.ac` to read
```
AC_PATH_PROGS(RCC, rcc-qt4 rcc, :, [$QT_DIR/bin:$PATH])
```
and re-build/install again from the beginning, starting from the `./bootstrap` step.

If still no icons are shown, please post the output of the `./configure` step.

zerberros, 2017-01-30:
this does not work
here the ./configure --disable-doc output:
`
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for style of include used by make... GNU
checking dependency style of g++... none
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... none
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: executing depfiles commands
=== configuring in qucs (/home/zerberros/source/qucs/qucs/qucs)
configure: running /bin/sh ./configure --disable-option-checking '--prefix=/usr/local'  '--disable-doc' --cache-file=/dev/null --srcdir=.
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking for clang++... clang++
checking whether we are using the GNU C++ compiler... yes
checking whether clang++ accepts -g... yes
checking dependency style of clang++... gcc3
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for ranlib... ranlib
checking for ar... (cached) ar
checking how to run the C preprocessor... gcc -E
checking for X... libraries , headers 
[... 425 lines cut ...]
config.status: creating tutorial/opamp/Makefile
config.status: creating tutorial/opamp/pics/Makefile
config.status: creating tutorial/timer555/Makefile
config.status: creating tutorial/timer555/pics/Makefile
config.status: creating tutorial/spicetoqucs/Makefile
config.status: creating tutorial/spicetoqucs/pics/Makefile
config.status: creating tutorial/equations/Makefile
config.status: creating tutorial/equations/pics/Makefile
config.status: creating tutorial/bjtbias/Makefile
config.status: creating tutorial/bjtbias/pics/Makefile
config.status: creating tutorial/textmode/Makefile
config.status: creating tutorial/textmode/pics/Makefile
config.status: creating tutorial/RFResModels/Makefile
config.status: creating tutorial/RFResModels/pics/Makefile
config.status: creating tutorial/activefilter/Makefile
config.status: creating tutorial/activefilter/pics/Makefile

 qucs-doc version 0.0.19 configured successfully.

Configure Information:
        OCTAVE    : /usr/bin/octave
    OCTAVE_EPSTK  : yes
	 

`

guitorri, 2017-01-30:
There must be an issue running `rcc` (resource compiler). It bundles the resources (bitmaps) into the executable.

Try to build only the Qucs-GUI, something like
```
./bootstrap
./configure --disable-doc
cd qucs/qucs/qucs/
make clean # if you ran make before
make
./qucs  # run the executable
```
Watch for something like the following in the log:
```
...
/usr/bin/rcc-qt4 -o qrc_qucs.cpp qucs.qrc
...
```
In other works the `qrc_qucs.cpp` should be generated.

zerberros, 2017-01-30:
The icons are still unseen. The only line that appears in terminal after execute ./qucs is:
Warning: QucsConv not found:  "/home/zerberros/source/qucs/qucs/qucs/bin/qucsconv"

guitorri, 2017-01-30:
Can you post the output of `make`?

zerberros, 2017-01-30:
This is the errors output. If you want to see the standar output, tell me:

sp_sim.cpp:27:13: warning: 'findRev' is deprecated
      [-Wdeprecated-declarations]
  int b = s.findRev(" ");
            ^
/usr/include/qt4/QtCore/qstring.h:535:28: note: 'findRev' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT int findRev(const QString &s, int i = -1, b...
                           ^
1 warning generated.
param_sweep.cpp:26:13: warning: 'findRev' is deprecated
      [-Wdeprecated-declarations]
  int a = s.findRev(" ");
            ^
/usr/include/qt4/QtCore/qstring.h:535:28: note: 'findRev' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT int findRev(const QString &s, int i = -1, b...
                           ^
1 warning generated.
hb_sim.cpp:26:13: warning: 'findRev' is deprecated
      [-Wdeprecated-declarations]
  int a = s.findRev(" ");
            ^
/usr/include/qt4/QtCore/qstring.h:535:28: note: 'findRev' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT int findRev(const QString &s, int i = -1, b...
                           ^
1 warning generated.
dc_sim.cpp:27:13: warning: 'findRev' is deprecated
      [-Wdeprecated-declarations]
  int b = s.findRev(" ");
            ^
/usr/include/qt4/QtCore/qstring.h:535:28: note: 'findRev' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT int findRev(const QString &s, int i = -1, b...
                           ^
1 warning generated.
ac_sim.cpp:27:13: warning: 'findRev' is deprecated
      [-Wdeprecated-declarations]
  int b = s.findRev(" ");
            ^
/usr/include/qt4/QtCore/qstring.h:535:28: note: 'findRev' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT int findRev(const QString &s, int i = -1, b...
                           ^
1 warning generated.
subcircuit.cpp:149:34: warning: 'read' is deprecated
      [-Wdeprecated-declarations]
  QString FileString = ReadWhole.read();
                                 ^
/usr/include/qt4/QtCore/qtextstream.h:249:32: note: 'read' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT QString read() { return readAll(); }
                               ^
1 warning generated.
spicefile.cpp:219:77: warning: 'absFilePath' is deprecated
      [-Wdeprecated-declarations]
  ...qCritical() << "Spice file not found:" << localFileInfo.absFilePat...
                                                             ^
/usr/include/qt4/QtCore/qfileinfo.h:161:32: note: 'absFilePath' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT QString absFilePath() const { return absolu...
                               ^
1 warning generated.
spicedialog.cpp:42:7: warning: 'QDialog' is deprecated
      [-Wdeprecated-declarations]
    : QDialog(d, 0, TRUE, Qt::WDestructiveClose)
      ^
/usr/include/qt4/QtGui/qdialog.h:67:29: note: 'QDialog' has been
[... 1276 lines cut ...]
                              ^
arrowdialog.cpp:115:30: warning: 'setPaletteBackgroundColor' is deprecated
      [-Wdeprecated-declarations]
  if(c.isValid()) ColorButt->setPaletteBackgroundColor(c);
                             ^
/usr/include/qt4/QtGui/qwidget.h:1077:22: note: 
      'setPaletteBackgroundColor' has been explicitly marked deprecated
      here
inline void QWidget::setPaletteBackgroundColor(const QColor &c)
                     ^
arrowdialog.cpp:141:41: warning: 'setCurrentItem' is deprecated
      [-Wdeprecated-declarations]
    case Qt::SolidLine      : StyleBox->setCurrentItem(0);
                                        ^
/usr/include/qt4/QtGui/qcombobox.h:261:29: note: 'setCurrentItem' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT void setCurrentItem(int index) { setCurrent...
                            ^
arrowdialog.cpp:143:41: warning: 'setCurrentItem' is deprecated
      [-Wdeprecated-declarations]
    case Qt::DashLine       : StyleBox->setCurrentItem(1);
                                        ^
/usr/include/qt4/QtGui/qcombobox.h:261:29: note: 'setCurrentItem' has been
      explicitly marked deprecated here
    inline QT3_SUPPORT void setCurrentItem(int index) { setCurrent...

in3otd, 2017-01-31:
> Warning: QucsConv not found: "/home/zerberros/source/qucs/qucs/qucs/bin/qucsconv"

are you launching `qucs` without installing it? At present this is not well supported, you should install it first (even to a local directory by specifying `--prefix` during the `./configure` phase).

Still, the icons should be there... Later I'll try to install Arch in a VM here to see what happens.
I suspect you may have a stray, zero-length `qrc_qucs.cpp` from your first try, where `rcc-qt4` was likely not found but no error thrown. This file may not be recreated when you re-compiled, unless you made a `make clean` or `make distclean` before and maybe even if you did.
Please check if you have a zero-length `qucs/qucs/qrc_qucs.cpp`; in case, delete it and recompile.

The tons of warnings above are "normal", they are mostly there to remind us we need to move away from the old Qt3 compatibility code (it's a long story :grin:)

zerberros, 2017-01-31:
Well, the last time, you don't tell me to install. Maybe I have to learn more about compiling with make. I uninstalled qucs for every attempt. I don't know if this is correct.

the file qucs/qrc_qucs.cpp have 10278 lines.

I think It's a good idea remove all Qt3 labels ;)

By the way, I actually have antergos, no arch. I mean do not spend 3 or 4 hours installing arch, install a faster arch derivative, Hehe.

in3otd, 2017-01-31:
> By the way, I actually have antergos, no arch. I mean do not spend 3 or 4 hours installing arch, install a faster arch derivative, Hehe.

At the beginning you said you were using Arch, so I installed that to try to replicate your issue. Why didn't you simply mention Antergos in the first place? Never mind.
Anyway, in Arch compilation fails since `rcc-qt4` is not detected and I get an error message
```
g++: error: qrc_qucs.cpp: No such file or directory
g++: fatal error: no input files
```
after modifying `configure.ac` as described above, compilation works fine - my `qrc-qucs.cpp` is 10027 lines long. Even running locally, without installing, I get the icons and all the other stuff.

zerberros, 2017-02-01:
OK. I downloaded the tarball from sourceForge, and it works perfectly, without any errors, and without modifying the configure.ac file.

But the version that I tried to compile it was the master branch from github. I don't know why this branch don't show the icons. I have clone again from github, and there is the problem. 

Thanks for your time, I think this issue can be closed

in3otd, 2017-02-01:
the tarball from sourceForge is a "distribution package" and already contains a `qrc_qucs.cpp` while a clone from GitHub or a release tarball from there is a "source package" and does not include a `qrc_qucs.cpp` (and you need to run `./bootstrap` there to create the `configure` script).

At least on Arch the Qt tools have `-qt4` appended, unless you run `qtchooser` before, so `rcc` is not found and this should be fixed.

>  I have clone again from github, and there is the problem.

I used the release tarball from GitHub for my tests in Arch.
Please list the exact steps you did so I can try to replicate the issue here, after installing Antergos...

zerberros, 2017-02-01:
> I used the release tarball from GitHub for my tests in Arch.
Please list the exact steps you did so I can try to replicate the issue here, after installing Antergos...

git clone https://github.com/Qucs/qucs.git
cd qucs
./bootstrap
./configure
make
sudo make install

joe-skb7, 2017-02-06:
The same issue on Debian testing ("stretch"). Yes, I don't have `octave-epstk` package installed. But the ugly thing is, there is no such package in Debian anymore. So I had to install it manually, taking it from "jessie" repository. So it would be really nice, if we can build qucs-docs cleanly with no `octave-epstk` package installed.

closed 2017-02-09 by zerberros

keesj, 2018-04-18:
I had similar issues on ubuntu installing from the sourceforge sources worked for me

BOSSY1388, 2022-04-26:
THATS BECAUSE THA
T IS ALL MY DATA WHOM I ENTRUSTED WITH MY SIBLING WHOM THEN ALTERED MY DETAILS AND PICTURES IN ORDER TO ACCESS MY TEAMZ MONEY ACCOUNTS H.B.I.C .....
[image]