Qucs/qucs #656: Fix deprecation warnings (Qt3Support)

guitorri, 2017-01-31:
There is a truckload of deprecation warnings when `-DQT3_SUPPORT_WARNINGS` is enabled.

See log in #653 as an example.

guitorri, 2017-01-31:
I started working in qucs/qucs/qucs.

in3otd, 2017-01-31:
IIRC 70% of the warnings can be fixed by simply updating a function name or similar; I had somewhere a script, which I think came originally from the KDE repository, which handled this automatically but I cannot find it anymore at present. I never used it actually, but it could be useful.

guitorri, 2017-01-31:
Yes, most are simple. If you find the script let me know.
For now I am doing it in QtCreator that helps jump around the issues, then I grep and sed for bulk changes. 
I wan't to clean up a bit the stage before start cherry-picking form the QGraphics work I did so far.

in3otd, 2017-01-31:
...they were from the KDE kdesk app, cannot find the original repo but a copy seems to be at https://github.com/wyuka/kdesdk/tree/master/scripts/qt4 . There is a `remove-qt3-support.pl` plus other stuff.

guitorri, 2017-02-01:
With #655 and #662 there are no more warnings.
The relevant flags used where: `DQT3_SUPPORT_WARNINGS -Wall  -std=c++0x  -Wno-deprecated-register`

I need help to check the buttons with background color. I cannot see colors on the Mac, see #74.

I have yet to test:
- import/export of QUCS packages
- the octave dock.

in3otd, 2017-02-02:
most of the buttons which should have non-default background colors do not have the color actually  changed also on my Linux box. It one of those things that depends on the window manager and its settings.
I'm just used to that (...) and see the right color background only in Windows.

guitorri, 2017-02-02:
The create/extract packages is mangling directory names on extract. I have seen this before when taking some packages from Mike. I don't think the issue was introduced now, but please check. See below. The files are there, just with very weird names.
[image]

There are a few patches about the coloured buttons #74 and related. I will try to get back to them.

Still haven't tested the Octave dock.

in3otd, 2017-02-03:
> The create/extract packages is mangling directory names on extract.

I don't think I've seen this before... please attach a package that shows this issue on your side so I can try with the same. Or does it happen with all packages??

guitorri, 2017-02-06:
Funny thing. I just created 6 packages on 0.0.19, did a rebuild and `make install` the PR#662 and and tried to extracted the packages. One package got mangled as the figure above. I tried again a few minutes later and the package extracted just ok.
I did also the same thing on reverse. Created packages based on PR#662 and extracted on 0.0.19. One package failed and a few minutes later it extracted just fine. 

I don't know where to go from here. It seems there is a random mangling of directory names on extract. Maybe just on OSX? 

Please test on your side. It is not a big issue as the data is not lost, just renaming the directory fix the issue.

The buttons will be addressed in another PR.

I will test the octave dock now.

in3otd, 2017-02-06:
I have just 3 packages here; I can create some more for testing but maybe could be useful if you attach one  you created there.
At present, 2 packages work fine, one gives an error on extraction, but could have been damaged somewhere (don't know where it comes from)

guitorri, 2017-02-06:
Here they are. 
[qucs-pgk-tests.zip](https://github.com/Qucs/qucs/files/755857/qucs-pgk-tests.zip)
The ones prefixed by `d` where created with the #662 patch. The other ones created with 0.0.19.

guitorri, 2017-02-06:
I just did a`diff 01.qucs d01.qucs` on all 6 file pairs. They are the same binary.

in3otd, 2017-02-06:
BTW, compiling here I get
```
$ make 2>&1 >/dev/null | grep warning
spicedialog.cpp:42:7: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
spicedialog.cpp:374:29: warning: 'QMessageBox' is deprecated [-Wdeprecated-declarations]
spicedialog.cpp:441:27: warning: 'QMessageBox' is deprecated [-Wdeprecated-declarations]
3 warnings generated.
diagramdialog.cpp:109:23: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
diagramdialog.cpp:331:27: warning: 'QGridLayout' is deprecated [-Wdeprecated-declarations]
diagramdialog.cpp:426:26: warning: 'QSlider' is deprecated [-Wdeprecated-declarations]
diagramdialog.cpp:442:26: warning: 'QSlider' is deprecated [-Wdeprecated-declarations]
diagramdialog.cpp:458:26: warning: 'QSlider' is deprecated [-Wdeprecated-declarations]
5 warnings generated.
markerdialog.cpp:30:24: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
1 warning generated.
graphictextdialog.cpp:35:37: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
1 warning generated.
arrowdialog.cpp:32:37: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
arrowdialog.cpp:37:13: warning: 'QGridLayout' is deprecated [-Wdeprecated-declarations]
2 warnings generated.
filldialog.cpp:45:26: warning: 'QGridLayout' is deprecated [-Wdeprecated-declarations]
filldialog.cpp:76:26: warning: 'QGridLayout' is deprecated [-Wdeprecated-declarations]
2 warnings generated.
simmessage.cpp:353:27: warning: 'convertSeparators' is deprecated [-Wdeprecated-declarations]
1 warning generated.
savedialog.cpp:35:6: warning: 'QDialog' is deprecated [-Wdeprecated-declarations]
savedialog.cpp:55:27: warning: 'QVBoxLayout' is deprecated [-Wdeprecated-declarations]
savedialog.cpp:68:24: warning: 'QHBoxLayout' is deprecated [-Wdeprecated-declarations]
3 warnings generated.
mouseactions.cpp:74:23: warning: 'QMouseEvent' is deprecated [-Wdeprecated-declarations]
1 warning generated.
qucs.cpp:180:24: warning: 'argc' is deprecated [-Wdeprecated-declarations]
qucs.cpp:181:25: warning: 'argv' is deprecated [-Wdeprecated-declarations]
2 warnings generated.
```
(plus other known `fabs()` warnings)

guitorri, 2017-02-06:
right... these things I cannot see on OSX 😕 
I tried to enable -DQT_DEPRECATED_WARNINGS but no no avail. I will dig in to the qgloba.h to see how to trigger -Wdeprecated-declarations

guitorri, 2017-10-09:
Flags are enabled. Closing for now.

PR #662 (2017-02-19): Fix warnings qt3support
Closes #656.
closed 2017-10-09 by guitorri