Qucs/qucs #663: Check if is needed to set PATH for processes

guitorri, 2017-02-01:
The following snipped is used in 6 places:
```
QProcessEnvironment env = QProcessEnvironment::systemEnvironment();
env.insert("PATH", env.value("PATH") );
Program->setProcessEnvironment(env);
```
It tries to set the environment of the calling process for the process to be executed.

Most (?) processes are started with an absolute path to the executables, this should not be needed.
Needs to be checked.

in3otd, 2017-02-02:
maybe not really related but still around the path settings stuff, there is an issue reported [here](https://sourceforge.net/p/qucs/discussion/311050/thread/cedc297e); I guess there we have the opposite problem, that  `PATH` is not set so [`QProcess::execute()`](https://github.com/Qucs/qucs/blob/373d7373a6f79e6ff6bbf76a814fedbde684b518/qucs/qucs/imagewriter.cpp#L77-L85) fails. And, IIRC, it does not work only in Windows, in Linux it uses the default path (?)

felix-salfelder, 2017-10-12:
there seems to be a bug in the way an asco process is set up and/or how asco is invoked and/or asco input is generated.

felix-salfelder, 2017-10-13:
yes. the way asco is invoked now, it does not respect the `QUCSATOR` envoronment variable.

steps to reproduce: 0) make sure there is no qucsator in `$PATH`, or sticking around in your builddir 1) run `QUCSATOR=/path/to/qucsator qucs`. 2) somehow invoke asco.

in3otd, 2017-10-13:
good point, instead of `QucsSettings.BinDir` we should have used `canonicalPath()` of`QucsSettings.BinDir` or something like that.

in3otd, 2018-02-15:
adding the 0.0.20 milestone as a reminder to fix the ASCO invocation (at least)

guitorri, 2018-07-25:
Well, current state is that only `octave_window` does the 
```
env.insert("PATH", env.value("PATH") );
```
I will check if can work without it.

PR #730 (2017-10-11): fix PATH use
PATH is en environment variable, lets not mess with it. fixes #663