Qucs/qucs #664: S-parameter simulation

z-Wind, 2017-02-02:
Hi

Window 7 x32
0.0.19

I want to follow the getstarted.pdf to make the same result between AC simulation and S-parameter simulation.
But, I find it's different, and I check the AC circuit parameter as shown in the picture. It's right.
So I think the s-parameter simulation maybe is wrong.
And even I change the impedance of the port1 and port2 to the same value like 500, the s-parameter are changed, but I think it should be the same.

Thanks a lot for providing so amazing tool.

[image]
[image]

in3otd, 2017-02-02:
interesting exercise; please check again the S-parameters definition (e.g. in http://www.ece.rutgers.edu/~orfanidi/ewa/ch14.pdf ), in particular
- the signs of the currents in the equations for <tt>a<sub>2</sub></tt> and <tt>b<sub>2</sub></tt>. You can use current probe components to avoid issues with the currents signs.
-  the actual definitions. e.g. <tt>S<sub>11</sub>=b<sub>1</sub>/a<sub>1</sub></tt> **with <tt>b<sub>2</sub>=0</tt>**. If you keep the second generator on <tt>b<sub>2</sub></tt> cannot be zero. This is why you need two simulations to compute the two-port S-parameters with a standard AC analysis in spice :grin:

z-Wind, 2017-02-10:
Thanks a lot. I would try to study the document.

closed 2017-02-10 by z-Wind