Qucs/qucs #668: dB function with the same value but different answer

z-Wind, 2017-02-08:
Hi

Window 7x32
Qucs 0.0.19

From "functions.pdf", the dB function is defined "y=dB(x) = 20 log |x|".

But I get a wrong answer, if I just do dB(10) as shown below.

It seems to be "y = 10 log |x|".

However, when I do it with the voltage, it seems OK.

But the two value is the same.

Do I make something wrong? Thanks

[image]

felix-salfelder, 2017-02-08:
(i am not sure if functions.pdf is sufficiently correct or precise.)

"dB" is used to express ratios. usually, power ratios. so, if x=p/q is the ratio between two powers, then that ratio can be put as log_10(x)B = 10*log_10(x)dB.

however, ratios are unitless, so the "dB function" cannot infer whether what you pass is a power ratio at all. now if you pass something with unit volts, that's not the correct unit for an amplitude ratio. but: still somehow QUCS guesses that you _intend_ to express an amplitude ratio in dB... so you get twice the result instead (which looks "correct"). the reason why its "twice" is that power scales quadratically with the amplitude, and log turns "^2" into "*2"...

tldr: "dB" alone is not well defined (and it's a unit, not a function), always expect random results. when in doubt, do the computation manually, using "log".

z-Wind, 2017-02-08:
Thank you very much

The example is "y=db(10) returns 20." in "function.pdf". so I misunderstand it.

in3otd, 2017-02-19:
reopening this since, as z-Wind said, the documentation is not in line with the current behavior so needs to be corrected.
Need to dig into the code to understand what it actually does.