Qucs/qucs #673: qucs Mac installation should not pollute /usr/local

d235j, 2017-02-12:
The installer for 0.0.19 for Mac currently pollutes /usr/local. Ideally the software would be packaged like KiCad, where a main app has the other apps, libraries, and binaries inside its bundle, and symlinks to the copies inside the bundle were provided.

However, /usr/local is commonly used for user-installed software, often with Homebrew, and polluting it causes problems. It would be much better (for now) to have a defined prefix such as /usr/local/Qucs or /opt/local/Qucs instead, as that would avoid such pollution.

guitorri, 2017-02-12:
Thank you for reporting.
I agree with you. There a bunch of issues with the Mac package, see #131. 
I would like to fix them but I have been busy with other things. 
If someone volunteers to help us out I will certainly review/test the changes.

d235j, 2017-02-12:
@guitorri I'm aware of most of these issues and the fact that fixing them will take some time. However, the issue of polluting /usr/local is a big problem for me (and probably others), so moving Qucs into its own private prefix for the time being seems like it would be fairly straightforward.

guitorri, 2017-02-12:
Fair enough. I will think about rebuilding 0.0.19 package with a private prefix. 
I can also add a bottle to the brew [qucs formula](https://github.com/guitorri/homebrew-tap/blob/master/qucs.rb).

guitorri, 2017-02-12:
Well, making a homebrew bottle for qucs does not seem like an ideal solution. Qt4 from cartr/qt4/qt does not include Qt3Support. So building Qt4 from source to have Qt3Support and then pour the qucs bottle is kind of a waste of time.

d235j, 2017-10-30:
@guitorri looks like at least at this point Qt4 from cartr/qt4/qt does include Qt3Support. It may have been added at some point.

guitorri, 2017-11-02:
@d235j, yes I asked Carter to add Qt3Support. I will give it a try to clean up things for the upcoming release.

felix-salfelder, 2025-03-02:
There is no more need for Qt3support. The rest of this issue seems stale/outdated. closing.

closed 2025-03-02 by felix-salfelder