Qucs/qucs #676: Crash due to do "pointer being freed was not allocated"

guitorri, 2017-02-17:
Here is the crash info:
```
qucsator(16385,0x7fffa694b3c0) malloc: *** error for object 0x7fb016036800: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug

ERROR: Simulator crashed!
Please report this error to qucs-bugs@lists.sourceforge.net
```

Here is a test example that works just fine with 0.0.18 and it is failing with develop.
[bug_malloc_prj.zip](https://github.com/Qucs/qucs/files/784468/bug_malloc_prj.zip)

I did not bisect yet, but it is probably related to either f9dbd6, or 654651ca50.

felix-salfelder, 2017-10-04:
>  either f9dbd6, or 654651c

f9dbd6 does not seem right. it should be `if(p) { free(p); p=NULL;]`.

the other one looks wrong, as it removes `release`, which is (was) right.

revert both?

guitorri, 2017-10-04:
So, the context for some changes come from here: https://github.com/Qucs/qucs/issues/265

The 654651c is no longer applicable, the code was refactored a few commits later.

For the rest, `free` and `delete` should not need check for `NULL` or post assignment to `NULL`. If so, there are other issues lurking around.

The traceback with current `develop`:

```
~/qucs-bugs/bug_malloc_prj $ lldb -- ~/local/qucs/devel/bin/qucsator -i netlist.txt -o data.dat
(lldb) target create "/Users/guitorri/local/qucs/devel/bin/qucsator"
Current executable set to '/Users/guitorri/local/qucs/devel/bin/qucsator' (x86_64).
(lldb) settings set -- target.run-args  "-i" "netlist.txt" "-o" "data.dat"
(lldb) r
Process 78839 launched: '/Users/guitorri/local/qucs/devel/bin/qucsator' (x86_64)
project location: 
modules to load: 0
factorycreate.size() is 0
factorycreate has registered:
parsing netlist...
checking netlist...
netlist content
      3 C instances
      2 VCVS instances
      1 Vfile instances
      1 DC instances
      3 R instances
      1 TR instances
creating netlist...
NOTIFY: TR1: average time-step 1.40994e-08, 4121 rejections
NOTIFY: TR1: average NR-iterations 2.5809, 1 non-convergences
qucsator(78839,0x7fffe99453c0) malloc: *** error for object 0x10107e400: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Process 78839 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fffe0af1d42 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fffe0af1d42 <+10>: jae    0x7fffe0af1d4c            ; <+20>
    0x7fffe0af1d44 <+12>: movq   %rax, %rdi
    0x7fffe0af1d47 <+15>: jmp    0x7fffe0aeacaf            ; cerror_nocancel
    0x7fffe0af1d4c <+20>: retq   
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fffe0af1d42 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fffe0bdf457 libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fffe0a57420 libsystem_c.dylib`abort + 129
    frame #3: 0x00007fffe0b46fe7 libsystem_malloc.dylib`free + 530
    frame #4: 0x0000000100126380 libqucs.0.dylib`qucs::spline::~spline() + 64
    frame #5: 0x000000010013308b libqucs.0.dylib`qucs::interpolator::~interpolator() + 27
    frame #6: 0x0000000100178074 libqucs.0.dylib`vfile::~vfile() + 68
    frame #7: 0x00000001000bb9fa libqucs.0.dylib`qucs::net::~net() + 42
    frame #8: 0x0000000100001dec qucsator`main + 4460
    frame #9: 0x00007fffe09c3235 libdyld.dylib`start + 1
(lldb) 
```

The issue in the destructor of `spline`, https://github.com/Qucs/qucs/blob/master/qucs-core/src/spline.cpp#L343-L348

If I comment out all the `delete`s in there, the crash goes away... looking into it.

guitorri, 2017-10-04:
`delete[] f2;` is causing the crash

guitorri, 2017-10-04:
This is the culprit:
https://github.com/Qucs/qucs/blob/master/qucs-core/src/spline.cpp#L276

commit d346d62e (2017-10-09): Fix #676 crash

The destructor is is trying to free a std::vector leading to a crash.
closed 2017-10-09 by guitorri