Qucs/qucs #677: how to upgrade or remove intalled from source?

cvgaviao, 2017-02-18:
I've installed Qucs from source qucs-0.0.19S-rc5 on ubuntu.

Now I would like to install the released 0.19 version. what should I do?

Also there isn't a "make uninstall. how can I uninstall it?

guitorri, 2017-02-18:
Note QUCS-S is currently considered a fork of QUCS. @ra3xdh might be able to help you with qucs-0.0.19S-rc5 uninstall.

I see you got stuck with ADMS, so I figure you are already on track to build QUCS.

ra3xdh, 2017-02-19:
I used an unmodified Auttools build system from the latest Qucs for my Qucs-S package before the RC7 release candidate.  Qucs-S allows the installation into the one root with mainline Qucs since RC7 package.

I have just tried to build the old `qucs-0.0.19S-rc.tar.gz` and `uninstall` target was successfully created fro both `qucs-core` and `qucs` using the autotools. `CMake` allows to uninstall package too, but it was not recommended to build the particular RC5. I swtched to `CMake` since RC7. What build system did you use? If it was something wrong with your build system, you can always remove all files by hands. Here is the full directory tree of the RC5 installation: https://gist.github.com/ra3xdh/ebf0360ef3119343baf3b5c7ec9aa6e0

Please use the Qucs-S bugtracker https://github.com/ra3xdh/qucs/issues to report issues related to Qucs-S packages.

cvgaviao, 2017-02-19:
Hi @ra3xdh, 
So, I have installed soooo many things (related to java, c and c++) in my computer in the last couple of years that I really didn't remember what method I have used to this package. but the fact is that I had a uninstall target on my installation. 
Then I end up removing everything by hand ;)
thanks for the time !

closed 2017-02-19 by cvgaviao