Qucs/qucs #678: Build issue related to octave

cvgaviao, 2017-02-19:
I'm trying to build from source using ubuntu xenial. The installed octave is default 4.0.0-3ubuntu9.1.

Then I'm having this error:
```
/usr/bin/octave --path . stabgear.m
Graphic reduced !
GEAR1 error constants: 1-step, order 1, error constant -0.5
   0.00000   0.00000  -0.50000
GEAR2 error constants: 2-step, order 2, error constant -0.222222
   5.5511e-17   2.2204e-16   1.1102e-16  -2.2222e-01
GEAR3 error constants: 3-step, order 3, error constant -0.136364
  -5.5511e-17  -1.1102e-16  -5.5511e-17   5.5511e-16  -1.3636e-01
GEAR4 error constants: 4-step, order 4, error constant -0.096
   0.000000   0.000000  -0.000000   0.000000   0.000000  -0.096000
GEAR5 error constants: 5-step, order 5, error constant -0.0729927
   0.000000   0.000000   0.000000  -0.000000   0.000000   0.000000  -0.072993
GEAR6 error constants: 6-step, order 6, error constant -0.058309
 Columns 1 through 6:

  -1.8041e-16  -1.6653e-15  -1.0547e-15  -7.8965e-15  -5.5268e-15  -1.2441e-14

 Columns 7 and 8:

  -4.6733e-15  -5.8309e-02
/usr/bin/octave --path . stabmoulton.m
Graphic reduced !
AM1 error constants: 1-step, order 1, error constant -0.5
   0.00000   0.00000  -0.50000
AM2 error constants: 1-step, order 2, error constant -0.0833333
   0.000000   0.000000   0.000000  -0.083333
AM3 error constants: 2-step, order 3, error constant -0.0416667
   0.000000   0.000000   0.000000  -0.000000  -0.041667
AM4 error constants: 3-step, order 4, error constant -0.0263889
   0.000000   0.000000   0.000000   0.000000   0.000000  -0.026389
AM5 error constants: 4-step, order 5, error constant -0.01875
   0.000000   0.000000   0.000000  -0.000000   0.000000   0.000000  -0.018750
AM6 error constants: 5-step, order 6, error constant -0.0142692
 Columns 1 through 7:

   0.000000   0.000000  -0.000000   0.000000   0.000000   0.000000   0.000000

 Column 8:

  -0.014269
/usr/bin/octave --path . stabbashford.m
Graphic reduced !
AB1 error constants: 1-step, order 1, error constant 0.5
   0.00000   0.00000   0.50000
AB2 error constants: 2-step, order 2, error constant 0.416667
   0.00000   0.00000   0.00000   0.41667
AB3 error constants: 3-step, order 3, error constant 0.375
   0.00000  -0.00000   0.00000  -0.00000   0.37500
AB4 error constants: 4-step, order 4, error constant 0.348611
   0.00000   0.00000   0.00000   0.00000  -0.00000   0.34861
AB5 error constants: 5-step, order 5, error constant 0.329861
   0.00000   0.00000   0.00000  -0.00000  -0.00000  -0.00000   0.32986
AB6 error constants: 6-step, order 6, error constant 0.315592
   0.00000   0.00000  -0.00000  -0.00000  -0.00000  -0.00000  -0.00000   0.31559
/usr/bin/octave --path . stabgearex.m
Graphic reduced !
GEAR1 error constants: 2-step, order 1, error constant 1
   0   0   1
GEAR2 error constants: 3-step, order 2, error constant 1
   0   0   0   1
GEAR3 error constants: 4-step, order 3, error constant 1
   0.00000   0.00000   0.00000  -0.00000   1.00000
GEAR4 error constants: 5-step, order 4, error constant 1
   0.00000   0.00000   0.00000  -0.00000   0.00000   1.00000
GEAR5 error constants: 6-step, order 5, error constant 1
   0.00000   0.00000   0.00000   0.00000   0.00000  -0.00000   1.00000
GEAR6 error constants: 7-step, order 6, error constant 1
   0.00000   0.00000   0.00000   0.00000   0.00000  -0.00000   0.00000   1.00000
/usr/bin/octave --path . coplanarzl.m
Graphic reduced !
export TEXINPUTS=.:./..:./..:; : -interaction=nonstopmode technical.tex
export BIBINPUTS=.:; /usr/bin/bibtex technical
I couldn't open file name `technical.aux'
Makefile:586: recipe for target 'technical.dvi' failed
make[2]: *** [technical.dvi] Error 1
make[2]: Leaving directory '/home/cvgaviao/Development/Development_Linux/qucs/qucs-doc/technical'
Makefile:384: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/cvgaviao/Development/Development_Linux/qucs/qucs-doc'
Makefile:381: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

```
I have tried to open some of those Qucs **.m** files using the octave desktop and it crashed.

leon-vv, 2017-03-03:
I had the exact same issue. Also, whenever I tried to run a simulation it would complain about not being able to find 'qucsator'. Not sure if that's related to this build error though. In the end I was able to fix it by making a hardlink from /qucs/bin to /qucs/qucs-core/qucsator (ln ./qucs-core/qucsator ./qucs/bin/qucsator).

in3otd, 2017-03-03:
in general, building the docs can be skipped by passing `--disable-doc` to `configure`. All the documentation is also available already in `pdf` format on [Sourceforge](https://sourceforge.net/projects/qucs/files/qucs-binary/0.0.19/).

@leon-vv did you `make install` also? Running Qucs from the build directory does not work.

leon-vv, 2017-03-05:
> did you make install also?

Yes I think I tried without success. I retried running the following commands:
`
./bootstrap
./configure --disable-doc
make
sudo make install
`
I still get the same error in the dialog titled 'Qucs Simulation Messages':
`ERROR: Cannot start /usr/local/bin/qucsator (No such file or directory)`

`Running Qucs from the build directory does not work.`

Well, doing the following is actually the only way I got the simulation to work:
- Put the location of admsXml into my PATH
- Make directory qucs/bin
- Add a link from qucs/bin/qucsator to qucs-core/src/qucsator
- Run the qucs/qucs/qucs binary

If qucs is not meant to be ran from the build directory it is helpful if you put this into the Readme. I (and probably other people) always run programs I compile myself from the build directories. This way I don't have to run code with root privileges and I don't pollute my file system.

felix-salfelder, 2017-03-05:
> If qucs is not meant to be ran from the build directory

that bug has been fixed 1.5 years ago. see #660. 

> have to run code with root privileges and I don't pollute my file system.

exactly. read the discussion...

leon-vv, 2017-03-05:
> that bug has been fixed 1.5 years ago. see #660.

Then please tell me what I've done wrong while following the build instructions in the Readme.

> exactly. read the discussion... e.g. #374 

Why do I have to read a long discussion about technical stuff I'm not familiar with? I'm simply contributing a bug report.

felix-salfelder, 2017-03-05:
> Then please tell me what I've done wrong 

you did nothing wrong (the fix has not been merged.)

> I'm simply contributing a bug report.

thanks for that, i totally see your point. however, if you want to improve qucs, then a bug report is not sufficient. unfortunately, you need to argue that what you found is really a bug. sometimes that takes a week. sometimes a year. sometimes more.

maybe you can test or upvote the fix i wrote?

leon-vv, 2017-03-05:
> you did nothing wrong 

Good to know. It's a pity though that the build instructions listed in the Readme don't work for me (even after running `sudo make install`). This way other people are probably unable to use this (amazing!) software. But of course this is not your fault.

> (the fix has not been merged.)

Sorry, failed to notice. I have upvoted the pull request. Hopefully it's merged soon.

guitorri, 2017-10-18:
Should be fixed with the merge of #660.

closed 2017-10-18 by guitorri