Qucs/qucs #681: Left mouse click not working any more, system wide, after placing part

aktentasche, 2017-03-23:
Hi,

this is a really weird problem, as soon as I place a part in the schematic my left mouse click does not work any more, systemwide, except for the top bar and the system search screen.

I think it might not be a problem with Qucs but also my system or a combination of both.

distribution: arch linux
window manager: Gnome 3.22.2
Qucs version: 0.0.19

I am new to using Qucs, after placing a part is there a window supposed to pop up that somehow "locks" my mouse?
Regards,
aktentasche

in3otd, 2017-03-23:
hello,
strange problem, never seen that before.
After placing a part no window will pop up and you should be able to exit the "part placement mode" just by pressing `Esc`.
You mean you can only place a single part? Usually left-clicking on a schematic window after selecting a part should places more instances of it.

aktentasche, 2017-03-24:
Yes, I can only place a single part. However, I can drag & drop another part from the box on the left (for example, one resistor then one capacitor). The buttons in the top menu (zoom, cut, etc..) are also still clickable.

When I change to another window, for example my Browser or a PDF reader, no mouse clicks are possible any more. But closing Qucs by the x on the top right works, however, when the window "save as" pops up, I cannot click again. What also still works, like mentioned in the first post, are the system menus. Volume, WiFi etc.

Interestingly, aborting the placement mode with escape does not make a difference.

aktentasche, 2017-03-24:
Console output:

qucs 
Gtk-Message: (for origin information, set GTK_DEBUG): failed to retrieve property gtk-primary-button-warps-slider of type gboolean from rc file value "((GString*) 0x2127740)" of type gboolean

Drag& drop a resistor:
`Warning: QVariant::save: unable to save type 261.`

Drag& drop another one:
`Warning: QVariant::save: unable to save type 261.`

Drag& drop a capacitor
`Warning: QVariant::save: unable to save type 261.`

Smells like Qt doesn't like gtk.

aktentasche, 2017-03-24:
I checked where QVariant is used but I am not really familiar with Qt and the architecture of qucs. Maybe it helps anyways:

[image]

guitorri, 2017-10-11:
Wild guess: it might be a cast issue. 

Maybe a `qRegisterMetaType<>()` is needed somewhere around that `QVariant`?

See: 
https://forum.qt.io/topic/1690/qvariant-save-load-unable-to-save-type-x
https://github.com/qbittorrent/qBittorrent/issues/3921

closed 2018-07-19 by guitorri