Qucs/qucs #686: newlines break schematic loader

felix-salfelder, 2017-05-16:
see #685

ra3xdh, 2017-05-16:
Yes, it is well known feature (or bug?) of Qucs XML schematic format. There present also some other featurebugs that I cannot remember now. I firstly met them in 2014, when I started the design of my active filter synthesizer. The reason is that Qucs XML is not true XML. It uses  a custom home-made XML parser. And this parser has some limitations.

felix-salfelder, 2017-05-16:
do we have a specification of that format?

to me, it seems the newline in #685 was intentional...

> I cannot remember now

nor can I. but these days, we have a bugtracker `;)`

chgans, 2017-05-16:
I never found one, To understand it, I reverse-engineer it from the source code, which is a real pain.

Take for example the <text> pseudo-xml element, the last pseudo-xml attribute is a quoted token containing the text. But the text itself can contains quotes, and these quotes are not escaped in anyway... So to extract the text content, one has to find the first quote and then blindly swallow everything up the the last-1 character...

In my opinion, the best way to move forward, is to define a real XML format, and convert all existing files to the new format. 
I have been pushing for that, but without much success so far.

chgans, 2017-05-16:
> Take for example the pseudo-xml element

Sorry, I meant: Take for example the pseudo-xml "Text" element

felix-salfelder, 2017-05-16:
> and convert all existing files to the new format.

do that with your files. we don't need more arbitrary ad-hoc solutions, but flexibility. that means, support for more/multiple/other and arbitrary file formats. c.f. https://xkcd.com/927/

we could easily go for a standard exchange format *and* for something new (XML or not), while keeping the legacy. the cleanup sufficient for that approach ("refactoring") has to be done anyway...

i do not think that the author(s) of the xml-like schematic parser and printer messed up their format intentionally. it's just... nobody else uses it.

chgans, 2017-05-18:
Don't get me wrong, I totally respect the people behind this xml-like syntax, i have no problem with that. What was done 10 years ago was fit for the purpose at that time.

Anyway. What about starting documenting the Qucs file format then? 
I have acquired some knowledge, and keep discovering more stuff as I'm working on my Qcs related projects.

What is the best place/format/way to start this documentation process? 
I don't fancy write a latex document for that, so ASCIIdoc/MarkDown/Plain text would be preferred.

Should that be added to the source git repo? Do you guys have a wiki somewhere, or anything else?

chgans, 2017-05-18:
Side note: I've recently discovered as well the "version" effect of these document. AFAIU, Qucs approach to forward-compatibility is "use default if not found". There are magic things like the first and last properties have special purpose/meanings, ... Still a dark area to me.

felix-salfelder, 2017-05-18:
> Anyway. What about starting documenting the Qucs file format then?

do, if you care enough.

what i have in mind boils down to
http://git.savannah.gnu.org/cgit/gnucap.git/tree/apps/lang_spice.cc#n602 and 602

it's the one sensible and proven approach i know of that implements

- keep legacy behaviour
- don't care what it is
- permit alternatives

at the same time. the implementation ("if thats an old device, then do it the old way") transports 1:1 to the qucs schematic parser that we all desperately want. no side effects guaranteed. note how the gnucap code sample lives in plugin space, next to the other more sensible language plugins.

essentially, we just need to finish the gnucap parser to qucs port, rather than inventing something new.

chgans, 2017-05-18:
What is the best place/format/way to start this documentation process?

guitorri, 2017-05-18:
I suggest we document somewhere here:
https://github.com/Qucs/qucs/wiki

or here:
https://github.com/Qucs/qucs-help/blob/develop/source/internal.rst
http://qucs-help.readthedocs.io/en/latest/internal.html

chgans, 2017-05-18:
Thank, I think i'll go for the rst in qucs-help for technical details.

guitorri, 2017-10-09:
Is there something we can do right now? 
Such that the parser can ignore `\n`  till if finds the next `>`?

chgans, 2017-10-10:
I don't know if it's easy or not to fix the parser, I haven't touch Qucs for a while.

BTW, I just realised that i had some WIP regarding the documentation (https://github.com/Qucs/qucs-help/pull/13/commits/6ec07b9aa5b9827a4620976e408bbc89bac3c744)

I can't remember why i closed the PR, anyway i have just pushed an updated version here:
https://github.com/chgans/qucs-help/commit/f839758142dbb7dfc52e6f71c9029c2aa3ef0de2

Are you still interested in documentation updates? 
The above commit was done 6 month ago, i'm not really planning on getting back on that (apart for style/typo changes if needed). If you're interested i'll do a pull request

Chris

guitorri, 2017-10-10:
Please send a PR with the documentation update. Thank you.

closed 2021-03-09 by felix-salfelder